espressif/esp-idf #10591: ble_rpa_get_num_peer_dev_records() returned 0 even after bonding

veneno-529, 2023-01-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Development Kit: ESP32 Do it board
Module or chip used: ESP32
IDF version: v4.4.3
Build System: idf.py
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
Operating System: Windows
Using an IDE?: Yes (VS Code withe Platformio extension)
NimBLE version : 1.3.0

Problem Description
I am having problems with BLE bonding on android device.

Basically, what happens is that, after bonding one phone to the same ESP32 ( Android 12 phone). Everything works fine once bonded for first time. When I disconnect and connect it again the bonding fails and The only way to reconnect is to delete the old bond and reconnect. Still, the error will appear again.

The maximum number of bonds is set to 3.

I have noticed that peer device records restored displays zero. Even after bonding it.

Expected Behavior
After bonding was done, even after disconnecting and reconnecting again (Bond info still maintained on android) connection should be successful.

Actual Behavior
After bonding and disconnecting, then trying to reconnect it fails to connect successfully.

### Steps to reproduce
This is easily reproducible.

Bond Android 13 / Iphone
Once connected, Disconnect again. (Do not delete bond information from phone).
Try to reconnect again.
Code to reproduce this issue
I managed to reproduce the problem using the [bleprph ](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/bleprph)example. I have only added RPA address resolution to that example and I have attached the project files:
[bleprph.zip](https://github.com/espressif/esp-idf/files/10367600/bleprph.zip)

### Debug Logs
Logs : After erasing flash, uploading firmware, booting for first time and connecting android phone for first time
D (2601) NIMBLE_NVS: ble_store_config_cccds restored 0 bonds
D (2611) nvs: nvs_open_from_partition nimble_bond 1
D (2611) nvs: nvs_get_str_or_blob p_dev_rec_1
D (2621) nvs: nvs_close 40
D (2621) nvs: nvs_open_from_partition nimble_bond 1
D (2621) nvs: nvs_get_str_or_blob p_dev_rec_2
D (2631) nvs: nvs_close 41
D (2631) nvs: nvs_open_from_partition nimble_bond 1
D (2641) nvs: nvs_get_str_or_blob p_dev_rec_3
D (2641) nvs: nvs_close 42
D (2641) nvs: nvs_open_from_partition nimble_bond 1
D (2651) nvs: nvs_get_str_or_blob p_dev_rec_4
D (2651) nvs: nvs_close 43
D (2661) NIMBLE_NVS: peer_dev_rec restored 0 records
I (2661) NimBLE_BLE_PRPH: BLE Host Task Started
D (2671) NimBLE: registered service 0x1800 with handle=1

D (2671) NimBLE: registering characteristic 0x2a00 with def_handle=2 val_handle=3

D (2681) NimBLE: registering characteristic 0x2a01 with def_handle=4 val_handle=5

D (2691) NimBLE: registered service 0x1801 with handle=6

D (2691) NimBLE: registering characteristic 0x2a05 with def_handle=7 val_handle=8

D (2701) NimBLE: registered service 59462f12-9543-9999-12c8-58b459a2712d with handle=10
[... 280 lines cut ...]

D (2721) NimBLE: registered service 0x1801 with handle=6

D (2731) NimBLE: registering characteristic 0x2a05 with def_handle=7 val_handle=8

D (2741) NimBLE: registered service 59462f12-9543-9999-12c8-58b459a2712d with handle=10

D (2741) NimBLE: registering characteristic 5c3a659e-897e-45e1-b016-007107c96df6 with def_handle=11 val_handle=12

D (2761) NimBLE: registering characteristic 5c3a659e-897e-45e1-b016-007107c96df7 with def_handle=13 val_handle=14

I (2771) uart: queue free spaces: 8
V (2771) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): checking args
V (2781) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): Args okay. Resulting flags 0xE
D (2781) intr_alloc: Connected src 34 to int 3 (cpu 1)
I (2811) NimBLE: Device Address:
I (2811) NimBLE: 5d:4f:bc:40:76:32
I (2811) NimBLE:

I (2821) NimBLE: GAP procedure initiated: advertise;
I (2821) NimBLE: disc_mode=2
I (2821) NimBLE: adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (2831) NimBLE

Notice that even after device was bonded peer records restored is 0.

IshaESP, 2023-02-02:
Hi,
I tried reproducing the issue locally using ESP32 and android 13 phone. But it is working without any error.
Can you send me sdkconfig of the application?

Also I can't find `E (68072) NimBLE_BLE_PRPH: ble_rpa_get_num_peer_dev_records() returned 0` this print in the application shared on github. Is something else being done apart from the code shared?

veneno-529, 2023-02-12:
Hi @IshaESP 
I hope you have used the code from the zip file attached. I have added only RPA feature to the bleprph example which is there on github. I tried it with esp ide as well as platform io too. Can you share your code or maybe details of esp idf version and nimble version?

IshaESP, 2023-02-16:
Hey @veneno-529 
Steps I used to reproduce the issue:

- idf.py erase_flash
- Connect using nrf application
- Bond with esp32 from nrf app (There is a 'bond' option in collapsible tab indicated using 3 vertical dots at top right corner of the application)
- Disconnect
- Reconnect 
- Disconnect again
- Boot esp32
- Connect again

After executing these steps it connects successfully.

IDF version : v4.4.3 (commit - 6407ecb3f8d2cc07c4c230e7e64f2046af5c86f7)
NimBLE version : v1.3.0 (commit - ed8a4b14ad43d88f2ec8913aab5af919208722e5)

I am attaching debug prints patch and output of the same zip file code shared: [IDFGH-10591.zip](https://github.com/espressif/esp-idf/files/10755252/IDFGH-10591.zip)

veneno-529, 2023-02-18:
> 

Hello @IshaESP 
I just upgraded to ESP-IDF 5.0 and Nimble v1.4.0 and everything works fine now.
I will still try your patch and check if I'm getting the same output as yours.
Thanks for your response!

veneno-529, 2023-02-18:
It turns out that the below feature highlighted in menuconfig needs to be enabled.
[image]
Now I got the logs same as yours and device connects and reconnects successfully even after restarting it.

closed 2023-02-18 by veneno-529