espressif/esp-idf #10592: ESP32-LyraT-mini V1.2, ESP-IDF V5.0, esp-idf/examples/storage/sd_card/sdspi/ ERROR

BeijingUncle, 2023-01-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.0

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-LyraT-mini V1.2

### Power Supply used.

USB

### What is the expected behavior?

SD card mounted

### What is the actual behavior?

SD card is not mounted

### Steps to reproduce.

Just build the esp-idf/examples/storage/sd_card/sdspi/ 

### Debug Logs.

```plain
I (329) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (339) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (349) example: Initializing SD card
I (349) example: Using SPI peripheral
I (359) example: Mounting filesystem
I (359) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (389) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
I (389) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (389) example: Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up resistors in place.
```

adokitkat, 2023-04-13:
Hello! Have you checked if the SPI pins are set correctly? Look at the [HW reference](https://espressif-docs.readthedocs-hosted.com/projects/esp-adf/en/latest/design-guide/dev-boards/board-esp32-lyrat-mini-v1.2.html#microsd-card) to see what pins are connected to MicroSD card reader.

adokitkat, 2023-04-21:
Closing this issue for now. You can reopen it if you still have a problem.

closed 2023-04-21 by adokitkat

harsha8370, 2024-07-29:
Hi guys, 
I am using esp32 lyraT mini v1.2 board 

what's the task is we need to configure the board to play the music from the sd card 
can anybody help me with that?

adokitkat, 2024-07-29:
Hi @harsha8370. I can refer you to:

ESP-IoT-Solution
https://github.com/espressif/esp-iot-solution/tree/master/examples/audio/wav_player
https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_audio_player

ESP-ADF
https://github.com/espressif/esp-adf/tree/master/examples

harsha8370, 2024-07-29:
Hi adam,

harsha8370, 2024-07-29:
Hi adam,
I am getting this type of error
and I am not able to fix it by the way, I am using the board ESP32
lyraTmini v1.2

error:
E (629) sdmmc_io: sdmmc_io_reset: unexpected return: 0x109
E (629) vfs_fat_sdmmc: sdmmc_card_init failed (0x109).
E (639) SDCARD: Failed to initialize the card (ESP_ERR_INVALID_CRC). Make
sure S
E (649) SDCARD: SD card initialization failed

rgds,
Harsha  N M

adokitkat, 2024-07-29:
@harsha8370 which example are you using? Which ESP-IDF version are you using? How do you have the card connected? ESP_ERR_INVALID_CRC can for example happen if your cables are too long.

harsha8370, 2024-07-29:
example: a2dp_source and the version is v1.2

harsha8370, 2024-07-29:
[image]
this is the exact error I am getting

adokitkat, 2024-07-29:
As I said, `ESP_ERR_INVALID_CRC` error is most of the times related to a signal integrity issue.

Please check your connection and try to lower the bus frequency (the speed ESP32 is communicating with the SD card, the default is 20MHz so try 10 or 5). if you connect to the card by SD card reader board and cables, try to switch to shorter ones. Also are you sure it does have pull-up resistors in place?

Also can you send a link to your example? I can't find which one you mean.

adokitkat, 2024-07-29:
Try to add this `host.max_freq_khz = 1000;` here:

```c
esp_err_t sd_card_init(void) {
    esp_err_t ret;

    // Configure SD card
    sdmmc_host_t host = SDMMC_HOST_DEFAULT();
    host.flags = SDMMC_HOST_FLAG_1BIT;  // Use 1-bit mode

    host.max_freq_khz = 1000; // Set lower frequency 
    // ...
```

If it works then try  set it higher to e.g. `5000` kHz, then try `10000`  etc. (so you get the highest working frequency). If it doesn't help then just try to run this https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card/sdmmc so we can identify the issue easier.

Also you don't need to set internal pull-ups to D1-D3 because you set your slot width to 1 (only D0 is used).

adokitkat, 2024-07-29:
I am not sure if it true because I don't have the same board as you, but I found supposedly on LyraT mini board you need to set GPIO13 as an output and drive it low. https://esp32.com/viewtopic.php?t=19456#p99457

harsha8370, 2024-07-30:
I made the changes you suggested, but I am still getting the same error.
If there's anything else you'd like to adjust, feel free to let me know!

adokitkat, 2024-07-30:
@harsha8370 Try this:

```c
#define SDCARD_PWR_CTRL GPIO_NUM_13

esp_err_t sd_card_init(void) {
    esp_err_t ret;

    // Configure SD card
    sdmmc_host_t host = SDMMC_HOST_DEFAULT();
    host.max_freq_khz = 5000; // Set lower frequency

    sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();
    slot_config.width = 1; // Set bus width to 1-bit
     
    // Set GPIOs based on the schematic
    slot_config.cd = GPIO_NUM_34;  // Card Detect
    slot_config.flags |= SDMMC_SLOT_FLAG_INTERNAL_PULLUP;  // Enable internal pull-ups

    // Set GPIO_NUM_13 to 0
    gpio_config_t sdcard_pwr_pin_cfg = {
        .pin_bit_mask = 1UL << SDCARD_PWR_CTRL,
        .mode = GPIO_MODE_OUTPUT,
        .pull_up_en = GPIO_PULLUP_DISABLE,
        .pull_down_en = GPIO_PULLDOWN_DISABLE,
        .intr_type = GPIO_INTR_DISABLE,
    };
    gpio_config(&sdcard_pwr_pin_cfg);
    gpio_set_level(SDCARD_PWR_CTRL, 0);

    // Mount the filesystem
    esp_vfs_fat_sdmmc_mount_config_t mount_config = {
        .format_if_mount_failed = false,
        .max_files = 5,
        .allocation_unit_size = 16 * 1024
    };

    // Initialize SD card
    sdmmc_card_t *card;
    ESP_LOGI(TAG, "Initializing SD card...");
    ret = esp_vfs_fat_sdmmc_mount(MOUNT_POINT, &host, &slot_config, &mount_config, &card);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            ESP_LOGE(TAG, "Failed to mount filesystem. If you want the card to be formatted, set format_if_mount_failed = true.");
        } else {
            ESP_LOGE(TAG, "Failed to initialize the card (%s). Make sure SD card lines have pull-up resistors in place.", esp_err_to_name(ret));
        }
        return ret;
    }

    // Card has been initialized, print information
    sdmmc_card_print_info(stdout, card);

    return ESP_OK;
}
```

Please also check if there are any jumpers on your board and if you have them connected properly.

If this still doesn't work, then please update you ESP-ADF version and use the latest examples from https://github.com/espressif/esp-adf/tree/master/examples.

harsha8370, 2024-07-30:
First, thank you so much, @adokitkat. We were able to initialize the SD card.

Could you please help me fetch the songs from the SD card and send them to a connected device?
If you have any code for the same board, please send it to me.

I have found some examples on GitHub, but they have many errors.

adokitkat, 2024-07-30:
No problem, you're welcome.

You should check out these examples https://github.com/espressif/esp-adf/tree/master/examples/player from ESP-ADF. Are you using it, right? The master branch has updated examples with all the code needed (even functioning SD card initialization etc., that's why I already suggested using it).

This is a repository for ESP-IDF (not ADF) and already closed issue, so please if you still have some issues open a new issue in ESP-ADF repo.

harsha8370, 2024-07-30:
Thank you for the suggestion! I’ll try increasing the host.max_freq_khz to 10,000 or 20,000 and see if it improves performance.

I appreciate the link to the ESP-ADF examples. I'll definitely check those out for reference on the SD card initialization and other functionalities. They should be really helpful in getting everything set up correctly.

I understand that the repository I'm working on is for ESP-IDF, but if I encounter further issues, I'll be sure to open a new issue in the ESP-ADF repository as you suggested.

Thanks again for your guidance!

harsha8370, 2024-07-31:
Hi @adokitkat 

Can you suggest how we can send music files from the SD card over Bluetooth? We have already been able to read, write, and create a list of files from the SD card.

harsha8370, 2024-08-01:
Hi @adokitkat ,

I'm stuck at fetching files from the SD card and sending them to connected Bluetooth devices. Can you help me to get this done?

harsha8370, 2024-08-09:
Hi @pacucha42 @BeijingUncle  @pacucha42 @adokitkat  

I am getting this type of error

please check and let me know 
[image]

harsha8370, 2024-08-09:
Hi @adokitkat @pacucha42  

Unable to open the sd card file while transmitting audio data to connected device through bluetooth a2dp 

[image]
[image]

NehalAR, 2025-03-21:
> Unable to open the sd card file

Try enabling the full name from menuconfig as it might be related to formatting, as well as run the example sdcard exmple to list the file names so that you know full path and how names are appearing.