espressif/esp-idf #10594: [Question] Handling multiple HTTP requests at the same time

chipweinberger, 2023-01-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Is there a recommended way for the http server to handle multiple requests at the same time?

is it even possible?

Currently only a single request handler can run at a time, and even if you did spawn a thread the `httpd_req_t` gets deleted when the handler completes, so communication becomes impossible.

How reasonable would it be for me to implement this in ESP-IDF? any pointers?

wuyuanyi135, 2023-01-22:
https://github.com/espressif/esp-idf/blob/49551cc48cb3cdd5563059028749616de313f0ec/examples/protocols/http_server/advanced_tests/main/tests.c#L273

chipweinberger, 2023-01-22:
Ive seen that example before but I don't understand how it would work in a real situation.

`httpd_queue_work` executes code on the same thread as the HTTP Server, and the example just returns a string synchronously. What is the reason for doing it this way?

Perhaps the example could show how to execute a long 30+ second calculation asynchronously and _then_ return it.

Or more important, how to download a large file over HTTP for a few minutes while still executing other requests.

boarchuz, 2023-01-23:
+1 for this feature request

wuyuanyi135, 2023-01-23:
You can save those file descriptor and handle to a structure and pass it to another thread for later use. In this case you cannot use those req-like functions. You have to assemble the http packet yourself. I can show you some c++ code to wrap around these functions when I got home

wuyuanyi135, 2023-01-23:
```cpp
#define HTTP_UTIL_CHECK(x)                                                                              \
  do {                                                                                                  \
    int ret = x;                                                                                        \
    if (ret == HTTPD_SOCK_ERR_INVALID || ret == HTTPD_SOCK_ERR_TIMEOUT || ret == HTTPD_SOCK_ERR_FAIL) { \
      return ret;                                                                                       \
    }                                                                                                   \
  } while (0)

int httpd_socket_send_common_header(httpd_handle_t hd, int fd, const std::string& content_type, size_t content_length, int status = 200) {
  char buf[80];
  size_t sz = snprintf(
      buf,
      sizeof(buf),
      "HTTP/1.1 %d OK\r\n"
      "Content-Type: %s\r\n"
      "Content-Length: %d\r\n",
      status,
      content_type.c_str(),
      content_length);
  return httpd_socket_send(hd, fd, buf, sz, 0);
}

int httpd_socket_header_finishes(httpd_handle_t hd, int fd) {
  const char* sep = "\r\n";
  return httpd_socket_send(hd, fd, sep, strlen(sep), 0);
}

int httpd_send_json(httpd_handle_t hd, int fd, const nlohmann::json& json, int status = 200) {
  auto str = json.dump();
  HTTP_UTIL_CHECK(httpd_socket_send_common_header(hd, fd, HTTPD_TYPE_JSON, str.size(), status));
  HTTP_UTIL_CHECK(httpd_socket_header_finishes(hd, fd));
  HTTP_UTIL_CHECK(httpd_socket_send(hd, fd, str.c_str(), str.size(), 0));
  return 0;
}

void httpd_send_json(httpd_req_t* req, const nlohmann::json& json, const char* status = HTTPD_200) {
  auto str = json.dump();

  CHK_E(httpd_resp_set_type(req, HTTPD_TYPE_JSON), "failed to set content type");
  CHK_E(httpd_resp_set_status(req, status), "failed to set status");
  CHK_E(httpd_resp_sendstr(req, str.c_str()), "failed to send json");
}
```

These are a part of my C++ http utility library. It depends on `nlohmann::json` for json encoding. Those `CHK_E` are macros for error checking. See how `httpd_send_json` works to assemble an asynchronous response.

If you are using C++, you can delegate the response to a thread like this
```cpp
auto hd = req->handle;
auto fd = httpd_req_to_sockfd(req);
std::thread th([hd, fd]{
     json = ....
    httpd_send_json(hd, fd, json);
});
```
If you are using C, you can follow how the example works.

As far as how this works, IIRC, the http socket was kept open if no `httpd_resp_*` was used to send response. So if you do not use `httpd_resp_*` in handler callback and extract the `fd` and `handle` you can use them later in another thread. 

I haven't test if the socket should be closed explicitly after sending all response. But from packet capture it looks this code works fine.

chipweinberger, 2023-01-23:
@wuyuanyi135 

Wow, thanks for the very impressive reply. Very helpful. 

Must have taken some time to figure this out! Very clever!

**Possible Improvements**: Given that the `fd` still works in your approach, it seems possibly easy to hack ESP-IDF into not freeing the `httpd_req_t ` when the handler completes. Perhaps in the req handler we could call a new `httpd_resp_claim_ownership(httpd_req_t* req)` function that makes *us* responsible for freeing it. This would allow us to continue to use it in a new thread.

wuyuanyi135, 2023-01-23:
@chipweinberger You are welcome!

One catch was that `httpd_queue_work` seems should do the work for the asynchronous response but actually it works fine with or without it. I raised a question in the [post](https://esp32.com/viewtopic.php?f=13&t=31446) but no response received. I hope the developer could clarify the use case of `httpd_queue_work`.

For your proposed improvement, I do agree that there is no reason that the asynchronous response should use a different set of API. There should be some APIs to restore the `httpd_req_t` object or prolong its lifetime. Let's see what espressif's opinion on this.

chipweinberger, 2023-01-23:
I'm begining to understand the http server code:

Where the req is fully deleted (not normally called - only called on error):
`esp_err_t httpd_req_delete(struct httpd_data *hd)`

Where the req is reset after a req handler completes, in order to be used again (this is the important one!):
`static void httpd_req_cleanup(httpd_req_t *r)`

Where the req handler is invoked by the server:
`esp_err_t httpd_uri(struct httpd_data *hd)`

The main task loop of the http server (just repeatedly calls `httpd_server(hd)`)
`static void httpd_thread(void *arg)`

chipweinberger, 2023-01-23:
A roadbump, it loops like the http server is designed around a single req object existing at a time.

This is the corresponding internal data for each httpd_handle_t:
```
struct httpd_data {
    ...
    struct httpd_req hd_req;                /*!< The current HTTPD request */
   ....
}
```

chipweinberger, 2023-01-23:
Okay, after looking at more code. A `httpd_req_t` just holds basic info about that request. i.e the uri, the fd, the http req/resp headers.

**It is my current understanding that we could just `memcpy` the  `httpd_req_t req` and the `req->aux` object to somewhere we control, and then continue to use the request object as normal!**

chipweinberger, 2023-01-23:
To alleviate questions about `httpd_queue_work()`, it is used to execute work on the `httpd` thread. i.e. in the `httpd_server` function.

```
static esp_err_t httpd_server(struct httpd_data *hd)
{
        ....
        // queued work will be processed here!
        httpd_process_ctrl_msg(hd);
        ....
}
```

I'm not sure exactly *why* you need to do this. But It seems like a simple way to get "thread safety" for doing various http related things. It lets you do work on a separate (higher priority?) thread, and then come back to the httpd thread to finish sending a reply. It does not allow simultaneous requests in and of itself, as we already know.

boarchuz, 2023-01-23:
I also want to reiterate from the linked forum thread that the request body will be purged upon return from the handler. So while the above may work for requests with no/small body, a firmware upload, for example, where there is a lot of data to receive and a lot of processing time required to do so (ie. writing to flash) requires blocking the httpd task at least until the entire request is received.

chipweinberger, 2023-01-23:
**Open question:** How does the http server handle closing of sockets?  Will multiple async requests properly close sockets? 

I don't see any handling of `Connection: close` in the http server code (only http client). 

And I've yet to find the place where the http server closes existing sockets.

**My current understanding:** the HTTP server will always keep the socket open until the client closes their tcp connection, in which case LWIP will handle the close for us.

wuyuanyi135, 2023-01-23:
As what I heard the esp-idf http server was a bit primitive. The mongoose http server was said to be more full-fledged. Mongoose server might not be cheap for commerial choice though. If working on a hobbist project, maybe check mongoose server and see if they provide the necessary features?

chipweinberger, 2023-01-24:
I've opened a PR to add lru purge support:

https://github.com/espressif/esp-idf/pull/10602

closed 2023-05-24 by espressif-bot

chipweinberger, 2023-05-24:
This feature has been merged.