espressif/esp-idf #10596: [RMT] Make rmt_channel_t public or add getters for 'basic' settings

higaski, 2023-01-22:
### Is your feature request related to a problem?

It might be a good idea to add simple getters for some parts of RMT channels. The struct is currently defined as follows

```c
struct rmt_channel_t {
    int channel_id;         // channel ID, index from 0
    int gpio_num;           // GPIO number used by RMT RX channel
    uint32_t channel_mask;  // mask of the memory blocks that occupied by the channel
    size_t mem_block_num;   // number of occupied RMT memory blocks
    rmt_group_t *group;     // which group the channel belongs to
    portMUX_TYPE spinlock;  // prevent channel resource accessing by user and interrupt concurrently
    uint32_t resolution_hz; // channel clock resolution
    intr_handle_t intr;     // allocated interrupt handle for each channel
    rmt_fsm_t fsm;          // channel life cycle specific FSM
    rmt_channel_direction_t direction; // channel direction
    rmt_symbol_word_t *hw_mem_base;    // base address of RMT channel hardware memory
    rmt_symbol_word_t *dma_mem_base;   // base address of RMT channel DMA buffer
    gdma_channel_handle_t dma_chan;    // DMA channel
    esp_pm_lock_handle_t pm_lock;      // power management lock
#if CONFIG_PM_ENABLE
    char pm_lock_name[RMT_PM_LOCK_NAME_LEN_MAX]; // pm lock name
#endif
    // RMT channel common interface
    // The following IO functions will have per-implementation for TX and RX channel
    esp_err_t (*del)(rmt_channel_t *channel);
    esp_err_t (*set_carrier_action)(rmt_channel_t *channel, const rmt_carrier_config_t *config);
    esp_err_t (*enable)(rmt_channel_t *channel);
    esp_err_t (*disable)(rmt_channel_t *channel);
};
```

While I agree that most of those members should not concern users some information, e.g. the channel_id, might be pretty important. For example, consider some code which might use RMT transmission and then change to some GPIO bitbanging for a wired-or response which needs to read the pin level.

Currently this requires users to delete the whole channel configuration and reconfigure the pin as GPIO although a simple call to the GPIO matrix could be sufficient as well. The problem is though, users have pretty much no way of telling which RMT channel they are even using... I was pretty surprised that some of my code makes use of channel 0+2 and not 0+1.

So code like the the one in `rmt_new_tx_channel` to configure the GPIO matrix is simply not possible without hardcoding the group and channel id:
```c
esp_rom_gpio_connect_out_signal(config->gpio_num,
                                rmt_periph_signals.groups[group_id].channels[channel_id + RMT_TX_CHANNEL_OFFSET_IN_GROUP].tx_sig,
                                config->flags.invert_out, false);
```

suda-morris, 2023-01-28:
Hi @higaski Thank you for the feed back. Yes, we can expose more getters if necessary. Firstly, I'd like to check if the current API can satisfy your requirement.

> consider some code which might use RMT transmission and then change to some GPIO bitbanging for a wired-or response which needs to read the pin level.

Have you tried this [io_loop_back](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/rmt.html#_CPPv4N23rmt_tx_channel_config_t12io_loop_backE) ?

higaski, 2023-01-28:
I don't think it helps in my particular case. I'd still need to manually toggle the GPIO pin bound to the RMT channel and I don't think that's possible without re-configuring the GPIO matrix?

suda-morris, 2023-01-29:
> I don't think it helps in my particular case. I'd still need to manually toggle the GPIO pin bound to the RMT channel and I don't think that's possible without re-configuring the GPIO matrix?

Yes, if that's the case, we need to reconfigure the GPIO matrix. Otherwise, you will have to delete the RMT channel and reinstall it later.

closed 2023-02-17 by espressif-bot