espressif/esp-idf #10598: C++20 Semaphore suffers deadlock but the FreeRTOS counting semaphore works fine

wuyuanyi135, 2023-01-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-3030-g09aeffbe9b

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-WROOM-1-N16R2

### Power Supply used.

USB

### What is the expected behavior?

C++20 Semaphore should work like the FreeRTOS Semaphore

### What is the actual behavior?

The C++20 Semaphore Producer-Consumer model will be stuck at deadlock but the equivalent FreeRTOS counting semaphore works fine.

### Steps to reproduce.

# Working FreeRTOS code 
The following code will simulate the Producer-consumer problem (produce in a thread and consume in another, synchronized by two semaphores).
The program will generate infinite output without halting.
```cpp
#include <esp_log.h>
#include <freertos/FreeRTOS.h>
#include <freertos/semphr.h>

#include <thread>

SemaphoreHandle_t n_empty_sem;
SemaphoreHandle_t n_filled_sem;

static const char* TAG = "MAIN";
size_t produced = 0;
size_t consumed = 0;
extern "C" void app_main() {
  n_empty_sem = xSemaphoreCreateCounting(3, 2);
  n_filled_sem = xSemaphoreCreateCounting(3, 0);
  auto producer = std::thread([] {
    while (true) {
      xQueueSemaphoreTake(n_empty_sem, portMAX_DELAY);
      produced++;
      if (produced % 1000 == 0) {
        ESP_LOGI(TAG, "Produced %d", produced);
      }
      xSemaphoreGive(n_filled_sem);
    }
  });
  auto consumer = std::thread([] {
    while (true) {
      xQueueSemaphoreTake(n_filled_sem, portMAX_DELAY);
      consumed++;
      if (consumed % 1000 == 0) {
        ESP_LOGW(TAG, "Consumed %d", consumed);
      }
      xSemaphoreGive(n_empty_sem);
    }
  });

  while (true) {}
}
```

# Equivalent C++20 Semaphore Implementation
This program is almost identical to the FreeRTOS version, I only replaced the FreeRTOS semaphore with the C++ counting semaphore. This program will run for a short period (a few seconds). Then, it no longer output anything. **I still see watchdog output so I believe it was deadlock.**
```cpp
#include <esp_log.h>

#include <semaphore>
#include <thread>

std::counting_semaphore<3> n_filled_sem{0};
std::counting_semaphore<3> n_empty_sem{2};
static const char* TAG = "MAIN";
size_t produced = 0;
size_t consumed = 0;
extern "C" void app_main() {
  auto producer = std::thread([] {
    while (true) {
      n_empty_sem.acquire();
      produced++;
      if (produced % 1000 == 0) {
        ESP_LOGI(TAG, "Produced %d", produced);
      }
      n_filled_sem.release();
    }
  });
  auto consumer = std::thread([] {
    while (true) {
      n_filled_sem.acquire();
      consumed++;
      if (consumed % 1000 == 0) {
        ESP_LOGW(TAG, "Consumed %d", consumed);
      }
      n_empty_sem.release();
    }
  });

  while (true) {}
}
```

wuyuanyi135, 2023-01-22:
Possibly related big https://gcc.gnu.org/bugzilla/show_bug.cgi?id=104928

wuyuanyi135, 2023-01-31:
Hi @0xjakob. Is this issue reproducible?

0xjakob, 2023-02-13:
@wuyuanyi135 Sorry for the long waiting time! The issue is not reproducible for me. I tried it a couple of times. Do you have an sdkconfig file? Do you also have an IDF commit ID that reproduces the issue and that is public (I can't reach the `09aeffbe9b` from your git describe)?

wuyuanyi135, 2023-02-13:
@0xjakob Thanks for your response! I may need sometime to find the project files. I will update here. Meanwhile did you keep the program running for a while? For me it stucked randomly, from a few seconds to minutes.

0xjakob, 2023-02-14:
@wuyuanyi135 It seems that I now encountered the issue (twice, after a few minutes). We'll have a look.

wuyuanyi135, 2023-02-14:
@0xjakob Thanks for the update!

0xjakob, 2023-02-28:
@wuyuanyi135 The fix needs a toolchain upgrade and we are working on it. But but toolchain updates take some time until they are ready. We will keep you updated here.

wuyuanyi135, 2023-02-28:
@0xjakob Nice work! Thanks for addressing these fundamental issues!

AxelLin, 2023-03-19:
> @wuyuanyi135 The fix needs a toolchain upgrade and we are working on it. But but toolchain updates take some time until they are ready. We will keep you updated here.

@0xjakob 
Do you have target timeline for the next toolchain upgrade?

0xjakob, 2023-03-21:
@AxelLin We might release another toolchain in Q2. Please excuse the long waiting time, however, toolchain upgrades have a big impact and can't be done as easily as a new commit on IDF.

closed 2023-05-21 by espressif-bot

wuyuanyi135, 2023-05-21:
Thank you for fixing this issue! does the latest master (efe9192) already contain the fix?

Lapshin, 2023-05-21:
@wuyuanyi135 , yes, all changes for the fix are in master branch!