espressif/esp-idf #10599: Unable to override managed component with mock

ljden, 2023-01-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I am trying to build a linux based `host_test` project to unit test a component. This component has a dependency on `espressif/esp-idf-cxx`. I am unable to override the dependency so that the build uses the mock. I'm not sure if I have implemented it correctly, the documentation for host testing, mocking and component manager are a bit lacking.

### What is the actual behavior?

It errors when adding the mock via any one of component manager, extra components dir, components dir.
```make
-- building ESP EVENT MOCKS
-- building FREERTOS MOCKS (only task, event-groups and queue)
-- building esp-idf-cxx MOCKS (I2C only)
CMake Error at /home/vbox/esp/esp-idf/tools/cmake/component.cmake:249 (message):
  ERROR: Cannot process component requirements.  Multiple candidates to
  satisfy project requirements:

    requirement: "esp-idf-cxx" candidates: "esp-idf-cxx, espressif__esp-idf-cxx"

Call Stack (most recent call first):
  /home/vbox/esp/esp-idf/tools/cmake/build.cmake:553 (__component_get_requirements)
  /home/vbox/esp/esp-idf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:15 (project)
```

### Steps to reproduce.

```bash
vbox@vbox:~/comp/host_test$ cat comp/host_test/CMakeLists.txt
cmake_minimum_required(VERSION 3.16)
set(CMAKE_CXX_STANDARD 20)
set(CMAKE_CXX_STANDARD_REQUIRED ON)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
set(COMPONENTS main)

idf_build_set_property(COMPILE_DEFINITIONS "-DconfigTICK_RATE_HZ=1000" APPEND)

list(APPEND EXTRA_COMPONENT_DIRS "$ENV{IDF_PATH}/tools/mocks/freertos/")
list(APPEND EXTRA_COMPONENT_DIRS "$ENV{IDF_PATH}/tools/mocks/esp_event/")
...
list(APPEND EXTRA_COMPONENT_DIRS "./mock_comps/mocks/esp-idf-cxx")

project(sensor_manager_host_test)
vbox@vbox:~/comp/host_test$ cat main/CMakeLists.txt
idf_component_register(SRCS "test_comp.cpp"
    WHOLE_ARCHIVE
    PRIV_REQUIRES comp cmock
    )

find_package(Catch2 3 REQUIRED)
find_package(trompeloeil REQUIRED)

target_link_libraries(${COMPONENT_LIB}
  PRIVATE
  Catch2::Catch2WithMain
  trompeloeil::trompeloeil
  )
vbox@vbox:~/comp/host_test$ cat main/idf_component.yml
dependencies:
  idf: "5.0.*"
  comp:
    path: "../../"
vbox@vbox:~/comp/host_test$ cat ../idf_component.yml
version: "0.0.1"
description: component
dependencies:
  idf: "5.0.*"
  espressif/esp-idf-cxx: "~1.0.0-beta"
...
vbox@vbox:~/sensor_manager/host_test$ tree mock_comps/
mock_comps/
└── mocks
    ├── esp-idf-cxx
    │   ├── CMakeLists.txt
    │   └── include
    │       └── i2c_cxx.hpp
    └── README.md

```
```make
vbox@vbox:~/comp/host_test$ rm -rf dependencies.lock managed_components/ sdkconfig build/ && idf.py build
...
-- building ESP EVENT MOCKS
-- building FREERTOS MOCKS (only task, event-groups and queue)
-- building esp-idf-cxx MOCKS (I2C only)
CMake Error at /home/vbox/esp/esp-idf/tools/cmake/component.cmake:249 (message):
  ERROR: Cannot process component requirements.  Multiple candidates to
  satisfy project requirements:

    requirement: "esp-idf-cxx" candidates: "espressif__esp-idf-cxx, esp-idf-cxx"

Call Stack (most recent call first):
  /home/vbox/esp/esp-idf/tools/cmake/build.cmake:553 (__component_get_requirements)
  /home/vbox/esp/esp-idf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:15 (project)
```

### More Information.

This may have something to do with the espressif version being namespaced? If that's the case, is there a workaround?
According to the docs, I would have expected to be able to override https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#multiple-components-with-the-same-name

0xjakob, 2023-02-14:
@ljden Sorry for the long waiting time! Thanks for taking action and posting this issue at `idf-component-manager`. Let's wait what the developers there will say.

0xjakob, 2023-04-11:
@ljden BTW, if you are trying to mock [ESP-IDF-CXX](https://github.com/espressif/esp-idf-cxx), it might not work as easily as mocking C-APIs since CMock cannot easily mock C++ classes.

Alvin1Zhang, 2023-06-13:
Thanks for reporting, let us track in the component manager issue tracker, https://github.com/espressif/idf-component-manager/issues/23,  thanks.

closed 2023-06-13 by Alvin1Zhang