espressif/esp-idf #10600: Compile failure in esp_sntp.h when included from cpp after commit a71fa821772

gtjoseph, 2023-01-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-3025-g49551cc48c

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Get a successful compile

### What is the actual behavior?

```
In file included from .../esp-idf/examples/protocols/sntp/main/sntp_example_main.cpp:21:
.../esp-idf/components/lwip/include/apps/esp_sntp.h: In function 'void sntp_setoperatingmode(u8_t)':
.../esp-idf/components/lwip/include/apps/esp_sntp.h:231:31: error: invalid conversion from 'u8_t' {aka 'unsigned char'} to 'esp_sntp_operatingmode_t' [-fpermissive]
  231 |     esp_sntp_setoperatingmode(operating_mode);
      |                               ^~~~~~~~~~~~~~
      |                               |
      |                               u8_t {aka unsigned char}
.../esp-idf/components/lwip/include/apps/esp_sntp.h:174:57: note:   initializing argument 1 of 'void esp_sntp_setoperatingmode(esp_sntp_operatingmode_t)'
  174 | void esp_sntp_setoperatingmode(esp_sntp_operatingmode_t operating_mode);
      |                                ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
```

### Steps to reproduce.

1. Change examples/protocols/sntp to a cpp program instead of c
2. idf.py build
...

### More Information.

esp_sntp.h line 231 needs a cast added to the call to esp_sntp_setoperatingmode...
```patch
diff --git a/components/lwip/include/apps/esp_sntp.h b/components/lwip/include/apps/esp_sntp.h
index 4086bb7825..ce68896f1e 100644
--- a/components/lwip/include/apps/esp_sntp.h
+++ b/components/lwip/include/apps/esp_sntp.h
@@ -221,21 +221,21 @@ void esp_sntp_servermode_dhcp(bool enable);
 #endif /* LWIP_DHCP_GET_NTP_SRV */
 
 #if !defined(ESP_LWIP_COMPONENT_BUILD) && !defined(ESP_NETIF_COMPONENT_BUILD)
 /**
  * @brief if not build within lwip, provide translating inlines,
  * that will warn about thread safety
  */
 static inline __attribute__((deprecated("use esp_sntp_setoperatingmode() instead")))
 void sntp_setoperatingmode(u8_t operating_mode)
 {
-    esp_sntp_setoperatingmode(operating_mode);
+    esp_sntp_setoperatingmode((esp_sntp_operatingmode_t)operating_mode);
 }
 
 static inline __attribute__((deprecated("use esp_sntp_servermode_dhcp() instead")))
 void sntp_servermode_dhcp(int set_servers_from_dhcp)
 {
 #if LWIP_DHCP_GET_NTP_SRV
     esp_sntp_servermode_dhcp(set_servers_from_dhcp);
 #endif
 }
 
```

AxelLin, 2023-03-18:
Fixed by: c8f0ec7fa4f9555db2e085977dc502689cad71fd

david-cermak, 2023-06-13:
https://github.com/espressif/esp-idf/pull/10625 Accepted and closed by https://github.com/espressif/esp-idf/commit/c8f0ec7fa4f9555db2e085977dc502689cad71fd

Thanks AxelLin for pointing this out!