espressif/esp-idf #10605: May be an error about SPI RAM of  ESP32-S3-WROOM-1

952736147, 2023-01-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

In [ESP-IDF Programming Guide]
(https://docs.espressif.com/projects/esp-idf/zh_CN/v4.4.3/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html?highlight=37) it says ：
“板载 ESP32-S3-WROOM-**2** 模组系列的开发板中，管脚 GPIO35、GPIO36 和 GPIO37 已用于内部 ESP32-S3 芯片与 SPI flash/PSRAM 之间的通信，外部不可使用。”
it is easy to make me believe that GPIO 35-37 of ESP32-S3-WROOM-**1** are free to use, however, this is not the truth. GPIO 35-37 of ESP32-S3-WROOM-**1** with OSPI could not be used either.
**SO I THINK THE SENTENCE IN PROGRAMMING GUIDE MAY BE MISLEADING!**

Icarus113, 2023-01-30:
@952736147  Hello, thanks for the report, we will improve the doc accordingly.

952736147, 2023-01-30:
> @952736147 Hello, thanks for the report, we will improve the doc accordingly.

Thanks , i am really happy for that i seemed to make contribute to ESP-IDF. should i close the issue now?

Icarus113, 2023-01-30:
@952736147 , there will be a commit to improve the doc you mentioned. After that commit is merged, this issue will be closed at the same time. So you can just leave this issue as it is~

closed 2023-01-31 by espressif-bot