espressif/esp-idf #10606: I2C master crash

952736147, 2023-01-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V4.4.3 ESP32-S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

i2c_master_start(cmd_link);                                                            
//省略其它I2C通信内容
i2c_master_read(cmd_link,&data ,1,I2C_MASTER_ACK);                                                                         
i2c_master_cmd_begin(I2C_NUM,cmd_link,pdMS_TO_TICKS(1000));

从机通过I2C发来若干字节，需要读取第一字节判断后续内容长度；
在上述代码中，期望表现是：
调用i2c_master_read读取一字节，然后根据读取的内容，再读取若干字节

### What is the actual behavior?

当I2C实际接收到的内容长度大于i2c_master_read函数试图读取的长度时，会导致i2c_master_read函数传入的读缓冲区（函数的第二个参数）越界。报错内容为：
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
 
Core  0 register dump:
PC      : 0x40377024  PS      : 0x00060031  A0      : 0x803772ed  A1      : 0x3fc92880
0x40377024: i2c_master_cmd_begin_static at E:/ESP32_ENV/esp-idf/components/driver/i2c.c:1289
 
A2      : 0x00000000  A3      : 0x3fce0734  A4      : 0x00000000  A5      : 0x3fcf46a0
A6      : 0x3fc946b4  A7      : 0x3fcf46c0  A8      : 0x00000000  A9      : 0x3fc94b64  
A10     : 0x00000093  A11     : 0x3fce09b8  A12     : 0x00000001  A13     : 0x3fc94b64
A14     : 0x60013000  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000008  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
 
 
Backtrace:0x40377021:0x3fc928800x403772ea:0x3fc928b0 0x40376916:0x3fc928f0 0x42018a97:0x3fcf46f0 0x42001f33:0x3fcf4710 0x4037b841:0x3fcf4730 0x4037cd71:0x3fcf4750
0x40377021: i2c_master_cmd_begin_static at E:/ESP32_ENV/esp-idf/components/driver/i2c.c:1288
 
0x403772ea: i2c_isr_handler_default at E:/ESP32_ENV/esp-idf/components/driver/i2c.c:497
 
0x40376916: _xt_lowint1 at E:/ESP32_ENV/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1111
 
0x42018a97: cpu_ll_waiti at E:/ESP32_ENV/esp-idf/components/hal/esp32s3/include/hal/cpu_ll.h:182
 (inlined by) esp_pm_impl_waiti at E:/ESP32_ENV/esp-idf/components/esp_pm/pm_impl.c:837
 
0x42001f33: esp_vApplicationIdleHook at E:/ESP32_ENV/esp-idf/components/esp_system/freertos_hooks.c:63
 
0x4037b841: prvIdleTask at E:/ESP32_ENV/esp-idf/components/freertos/tasks.c:3973 (discriminator 1)
 
0x4037cd71: vPortTaskWrapper at E:/ESP32_ENV/esp-idf/components/freertos/port/xtensa/port.c:131

### Steps to reproduce.

1. Step init i2c,and type code as "What is the expected behavior?" shown
2. Step slave send more bytes than the second parameter of "i2c_master_read"(read length)
3. Step ESP32 crashes
...

### More Information.

the error occurs when you make "i2c_master_read" as the last command while receive more data than you read
e.g.
i2c receive 3 bytes data and you set "i2c_master_read" as the last command to read 2 bytes
what you expect is read two bytes first and there remains one byte to read next time, but it will crash because of array bounds write

the solution is to provide a large enough buff for "i2c_master_read" if the bug is not fixed

AxelLin, 2023-04-25:
@o-marshmallow Any comment for this issue?

o-marshmallow, 2023-04-25:
Hi @952736147 ,

I  am not sure to understand what you are doing exactly. You say "Step slave send more bytes than the second parameter of "i2c_master_read"(read length)", this is not possible, the ESP32 is the master, the slave cannot send data without being clocked by the master.

Can you show exactly how you are preparing and starting the transfer? From what I see in the small snippet you gave, you have not added any `stop` to your transaction before calling `i2c_master_cmd_begin`.

I would adivse you to use `i2c_master_read_from_device` function instead:
```
i2c_master_read_from_device(I2C_NUM, device_address, &data, 1, pdMS_TO_TICKS(1000))
```

mythbuster5, 2025-01-13:
Any update for this thread?

mythbuster5, 2025-03-06:
Seems no respond any more. I wil close this issue, if there are any more update, feel free to reopen it again.

closed 2025-03-06 by mythbuster5