espressif/esp-idf #10609: esp_http_client: allow no url and expose function to set user data

MartyMcFlyInTheSky, 2023-01-24:
This is a feature request:

I'm implementing a c++ wrapper around esp_http_client that is loosely leaned on javascripts fetch APII. For that it would be very handy if the esp_http_client_init() would allow for _no url_ to be set (currently there's an error showing when the user attempts this). The point is I want to be able to set the url in the method performing the actual http request, e.g client.GET(std::string url) so the url is reset anyway.

Second, as the my GET function invokes esp_http_client_perform(), it would be handy if you could somehow reset the user_data paremeter on the fly. This way I could pass a pointer to a stack-allocated string which is then filled in the event handler (events get the user_data parameter passed) and returned to the caller which would possibly allow for copy elision optimizations. The way I do it right now is I just pass the wrapper object as user_data at initialization, but that means I have to keep a member string object which is then moved out of the GET() function every time it is called, which also increases the object size unnecessarily.

I hope it was somehow understandable what I'm getting at, otherwise please feel free to ask!

hmalpani, 2023-03-14:
Hello @MartyMcFlyInTheSky 
I'm sorry for not getting back to you sooner.
For the first requirement, I think you can set the URL as '0.0.0.0' initially. This way you will not get the errors related to the URL not being set.
For the second requirement, there is a [PR](https://github.com/espressif/esp-idf/pull/10715). Please check if that fulfills your requirements.

hmalpani, 2023-03-27:
Hello @MartyMcFlyInTheSky 
Support to access the user_data has been added https://github.com/espressif/esp-idf/commit/82cdcc5de5f12befe7fb8f959e410d59c95a63cf
Closing this issue. Please feel free to reopen.

closed 2023-03-27 by hmalpani