espressif/esp-idf #10610: Use single button to allow or reject a bond / pairi

bheestandLMT, 2023-01-25:
### Is your feature request related to a problem?

I am porting a product over from NRF52 to ESP32-S3.
I have a single button on the BLE peripheral the while pressed and released gives 30 seconds for a phone (central) to pair.
Unless that button is pressed, the pairing should be rejected.
On the NFR52 that was done by simply setting the` pm_sec_params_set(NULL)` to not allow bonding, etc.
To allow pairing for 30 seconds, i changed the security parameters to allow bonding, etc
`pm_sec_params_set(&sec_param);`
(my &sec_param is:
`   memset(&sec_param, 0, sizeof(ble_gap_sec_params_t));
    sec_param.bond           = SEC_PARAM_BOND;
    sec_param.mitm           = SEC_PARAM_MITM;
    sec_param.io_caps        = SEC_PARAM_IO_CAPABILITIES;
    sec_param.oob            = SEC_PARAM_OOB;
    sec_param.min_key_size   = SEC_PARAM_MIN_KEY_SIZE;
    sec_param.max_key_size   = SEC_PARAM_MAX_KEY_SIZE;
    sec_param.kdist_own.enc  = 1;
    sec_param.kdist_own.id   = 1;
    sec_param.kdist_peer.enc = 1;
    sec_param.kdist_peer.id  = 1;`

### Describe the solution you'd like.

A way to turn pairing / bonding on for a certain duration of time and then also turn off the ability to pair / bond.

### Additional context.

We use this as a security feature.

rahult-github, 2023-01-29:
Hi @bheestandLMT , 
 
As i understand, there is no such API in nimble to directly block pairing/ bonding. But one suggestion is that  BLE_GAP_EVENT_PASSKEY_ACTION event would be generated from stack to application to get user's input during pairing. May be you can start a timer of your preferred duration and only if GAP layer receives the event, then application can approve the request, or else reject it .

rahult-github, 2023-04-25:
Closing this due to no activity. Please feel free to reopen in case of any other issue

closed 2023-04-25 by rahult-github