espressif/esp-idf #10611: sntp is broken in current master tree

AxelLin, 2023-01-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-3025-g49551cc48c

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32C3DevKit-M1

### Power Supply used.

USB

### What is the expected behavior?

ntp sync should work as it was working before pull recent esp-idf update.

### What is the actual behavior?

sntp_setservername / esp_sntp_setservername does not work.

Hi @david-cermak 
sntp is broken after commit a71fa82177214657a62dda77d148631227f68e9e ("esp_netif/lwip: Fix core-locking config")
I think you cannot pass the address of a **local variable** as an argument for tcpip_callback().
e.g.
https://github.com/espressif/esp-idf/blob/master/components/lwip/apps/sntp/sntp.c#L219-L223

Needs to review all the argument pass to tcpip_callback().

### Steps to reproduce.

I notice the NTP time sync fail with below code:
    esp_sntp_setoperatingmode(ESP_SNTP_OPMODE_POLL);
    esp_sntp_setservername(0, ntp_srv);
    esp_sntp_set_sync_interval(60 * 1000);
    esp_sntp_set_time_sync_notification_cb(my_sntp_time_sync_notification_cb);
    esp_sntp_init();

### Debug Logs.

```plain
Debug message shows:
sntp_request: Invalid server address, trying next server.
```

david-cermak, 2023-01-25:
Hi @AxelLin 

Thanks for the quick report and sorry for the regression.

> I think you cannot pass the address of a local variable as an argument for tcpip_callback().

That is correct, need to use `tcpip_api_call()` instead. Quick fix is attached:
[lwip-sntp-Fix-esp_sntp_.-races.patch.txt](https://github.com/espressif/esp-idf/files/10497258/lwip-sntp-Fix-esp_sntp_.-races.patch.txt)

As a workaround, you can the esp-netif API, which is now *the recommended interface* to lwip.

https://github.com/espressif/esp-idf/blob/49551cc48cb3cdd5563059028749616de313f0ec/examples/protocols/http2_request/main/http2_request_example_main.c#L92-L94

AxelLin, 2023-01-30:
> Hi @AxelLin
>
> Thanks for the quick report and sorry for the regression.
>
>
> That is correct, need to use `tcpip_api_call()` instead. Quick fix is attached: [...]

Thanks for the prompt response with quick fix.
Hope to get the fix merged soon since existing application code stop working now.

> 
> As a workaround, you can the esp-netif API, which is now _the recommended interface_ to lwip.

I might move to new interface latter but not now.
For now, I prefer minimal changes to allow my code works with stable branches and master.

closed 2023-02-01 by espressif-bot