espressif/esp-idf #10613: [ESP32S3] sysview_tracing permanently crashing in v4.4

KonssnoK, 2023-01-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3-354-gb3e8121fb0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32S3-WROOM N8R2

### Power Supply used.

USB

### What is the expected behavior?

Run sysview

### What is the actual behavior?

continuous crash on timer installation

### Steps to reproduce.

1. Build example
2. flash
3. monitor
...

### Debug Logs.

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403755a8
0x403755a8: esp_restart_noos at C:/src/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:151 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1670
load:0x403c9700,len:0xb4c
load:0x403cc700,len:0x2efc
SHA-256 comparison failed:
Calculated: 9b3f828a1a05a4639ba64223a7cc8f08bde10f5091df78db944cf848dc129fa6
Expected: c8905b56e0c8552ec0d8b388daaa12e1295f3a6f559f762113f9ebfe525d71d6
Attempting to boot anyway...
entry 0x403c9934
I (47) boot: ESP-IDF v4.4.3-354-gb3e8121fb0 2nd stage bootloader
I (48) boot: compile time 13:31:45
I (48) boot: chip revision: 0
I (51) boot.esp32s3: Boot SPI Speed : 80MHz
I (55) boot.esp32s3: SPI Mode       : DIO
I (60) boot.esp32s3: SPI Flash Size : 8MB
[... 410 lines cut ...]
ELF file SHA256: bdb65e11a6718537

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403755a8
0x403755a8: esp_restart_noos at C:/src/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:151 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1670
load:0x403c9700,len:0xb4c
load:0x403cc700,len:0x2efc
SHA-256 comparison failed:
Calculated: 9b3f828a1a05a4639ba64223a7cc8f08bde10f5091df78db944cf848dc129fa6
Expected: c8905b56e0c8552ec0d8b388daaa12e1295f3a6f559f762113f9ebfe525d71d6
Attempting to boot anyway...
entry 0x403c9934
```

### More Information.

on master branch it works, some backport is missing.
I remind you v4.4 is still an active stable branch supporting esp32s3.

gerekon, 2023-01-26:
Can reproduce this problem. It happens due to stack overflow. In v5.0 the stack size for the tasks was increased https://github.com/espressif/esp-idf/commit/02f6b83f4708f68bf8727e595af372060cf29b5d#diff-5e35b069c93e046316c229df63fa26c51b822f6ab98598808f090e39ffee2298R159

KonssnoK, 2023-01-26:
@gerekon my current crash is related to the interrupts not being taken correctly by CPU, but i see also that part was changed in 5.0 without backporting

```
if !CONFIG_APPTRACE_SV_TS_SOURCE_TIMER_10
    example_timer_init(TIMER_GROUP_1, TIMER_0, 2000);
#else
#warning "Timer (Group 1, Timer 0) is used by sysview module itself!"
#endif

#if !CONFIG_APPTRACE_SV_TS_SOURCE_TIMER_00
    example_timer_init(TIMER_GROUP_0, TIMER_0, 4000);
#else
#warning "Timer (Group 0, Timer 0) is used by sysview module itself!"
#endif
    for (int i = 0; i < portNUM_PROCESSORS; i++) {
        gptimer_config_t timer_config = {
            .clk_src = GPTIMER_CLK_SRC_APB,
            .direction = GPTIMER_COUNT_UP,
            .resolution_hz = 1000000,
        };
        ESP_ERROR_CHECK(gptimer_new_timer(&timer_config, &event_data[i].gptimer));
        event_data[i].period = 1000000 * (i + 1);
    }
```

erhankur, 2024-12-06:
I am closing this issue as the support period for v4.4 has ended. @KonssnoK Feel free to reopen it if you switch to v5.x and encounter any issues.

closed 2024-12-06 by erhankur