espressif/esp-idf #10614: HEAP_CAPS_MALLOC lost values

MolnarM22, 2023-01-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

EPS32 WROVER

### Power Supply used.

USB

### What is the expected behavior?

I use LVGL and i want to read BMP file from SDCARD. When i load bitmaps my bmp file to an array, I get back random values.

### What is the actual behavior?

Malloc my array, and lost data.

### Steps to reproduce.

img_datas =(uint8_t *) heap_caps_malloc (Header.biWidth * Header.biHeight * 4 * sizeof(pixelValue), MALLOC_CAP_SPIRAM);
         if(!img_datas){
        	 ESP_LOGE(TAG, "Hiba a memora foglalas kozben!");
        	 return;
         }

   	    printf("------------------------------------------------------------\n");
   	    printf(" 32 bit: %d\n", heap_caps_get_free_size(MALLOC_CAP_32BIT));
   	    printf(" 8 bit: %d\n", heap_caps_get_free_size(MALLOC_CAP_8BIT));
   	    printf(" SPI RAM: %d\n", heap_caps_get_free_size(MALLOC_CAP_SPIRAM));
   	    printf(" Largest free block SPIRAM: %d\n", heap_caps_get_largest_free_block(MALLOC_CAP_SPIRAM));
   	    printf("------------------------------------------------------------\n");
         int i=-1;
         for(int k = 0; k < Header.biWidth * Header.biHeight * 4 ; k++){
       	    i++;
            fread(&pixelValue, sizeof(pixelValue), 1, BMPFile);
            //memcpy(&img_datas[i], &pixelValue, sizeof(pixelValue));
            img_datas[i]=pixelValue;
            if(k < 70){
            	printf(" %x " , img_datas[k]);
            }

         }
         fclose(BMPFile);

         for(int j=0; j < 70; j++){
        	ESP_LOGE(TAG, " %x ", img_datas[j]);
         }

### Debug Logs.

```plain
Output: 

0  0  ff  0  0  ff  0  0  ff  0  0  ff  0  0  ff  0  0  ff  0  0  ff  0  0  ff  0  0  ff  0  0  ff  0  0  ff  0  0 

14  0  80  0  14  ff  80  0  e2  0  b2  ff  a2  0  e8  0  e1  ff  e2  0  ae  0  b6  ff  c0  0  aa  0  a6  ff  8b  0  a2  0  f8
```

igrr, 2023-02-02:
@MolnarM22 Could you please post the complete code snippet we could compile and run? The code you have posted has some undefined variables, like `Header`, `pixelValue`, `BMPFile`, so we couldn't reproduce the issue.

MolnarM22, 2023-02-02:
Full code: 
main -> https://pastebin.com/jAWi9gBK
mount -> https://pastebin.com/bg01tq14

adokitkat, 2023-04-21:
Hello, could you please share your whole project as a zip file here? Content on your SD card (the BMP file, etc.) included as well. We will try to reproduce your issue.

adokitkat, 2023-04-26:
I have created [lvgl_esp32-wrover-kit_sdcard_example](https://github.com/adokitkat/lvgl_esp32-wrover-kit_sdcard_example) repo where I try to reproduce your issue. It uses the latest LVGL with the latest ESP-IDF. 

This is `sdkconfig` I used for 16bit BMP picture:

```
CONFIG_SPIRAM=y

# Force malloc to use  SPIRAM above allocation size of 100 bytes
CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=100

# LVGL configuration
CONFIG_LV_COLOR_16_SWAP=y

# Force LVGL to use ESP-IDFs malloc
CONFIG_LV_MEM_CUSTOM=y
CONFIG_LV_MEMCPY_MEMSET_STD=y

# Set LVGL to use posix API for filesystem (read from SD card)
CONFIG_LV_USE_FS_POSIX=y
CONFIG_LV_FS_POSIX_LETTER=65

# Turn on BMP decoder
CONFIG_LV_USE_BMP=y
```

16 bit BMP works as expected, however 32 bit one does not. I don't know where's the issue yet.

```
CONFIG_LV_COLOR_DEPTH_32=y
CONFIG_LV_COLOR_DEPTH=32
```

This is how it looks:
[image]

adokitkat, 2023-05-02:
Apparently the display on ESP32-WROVER-KiT cannot display 32bit data - 18bit (RGB666) at most however it isn't supported by LVGL, so 16bit (RGB565) in LVGL is used.

closed 2023-05-09 by adokitkat