espressif/esp-idf #10615: DPP Auth Failed on dpp-enrollee example

knight-ryu12, 2023-01-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-WROVER-DevkitC

### Power Supply used.

USB

### What is the expected behavior?

I expected it to correctly authenticate with Accesspoint. 

### What is the actual behavior?

It does not authenticate with accesspoint at all;
Seems that it fails with WPA being `D (66148) wpa: Mgmt Tx Status - 1, Cookie - 0x400e036c`, and returns ESP_ERR_DPP_TX_FAILURE no matter what accesspoint it uses.

### Steps to reproduce.

[Code](https://gist.github.com/knight-ryu12/9d7d811984ce6f3090ec5d6714e2ac77)
It is 1:1 to the example, expect I removed the key part. which should just work without it.

### Debug Logs.

```plain
D (64538) wpa: DPP: Rx Public Action frame: action - 9
I (64538) wpa: DPP: Authentication Request from ac:3e:b1:02:80:e0
D (64538) event: running post WIFI_EVENT:20 with handler 0x400d6adc and context 0x3ffbdcfc on loop 0x3ffbc9f0
0x400d6adc: event_handler(void*, char const*, int, void*) at C:/Users/Alma/sensor-data-update/main/main.cpp:55

E (64548) wpa: curve: secp256r1

E (64558) wpa: prime len is 32

E (64558) wpa: Q_x: - hexdump(len=32):
E (64568) wpa: 2e cc 68 18 67 c2 7a 7f d2 59 20 1f 89 c5 99 28
E (64568) wpa: d2 59 3b 12 b9 07 13 4d 04 02 9a 80 0e 57 69 5c
E (64578) wpa: Q_y: - hexdump(len=32):
E (64578) wpa: 52 00 62 4e d9 33 98 b6 79 86 f6 ba 1c 20 a0 34
E (64588) wpa: 6f ec 12 e8 8d 28 56 47 7e 6b 09 00 48 04 84 29 
E (64598) wpa: d:      - hexdump(len=32):
E (64598) wpa: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
E (64608) wpa: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
D (64558) event: running post WIFI_EVENT:20 with handler 0x400e0420 and context 0x3ffcf050 on loop 0x3ffbc9f0
0x400e0420: offchan_event_handler at C:/Users/Alma/esp443/esp-idf/components/wpa_supplicant/esp_supplicant/src/esp_dpp.c:449

D (64858) wpa: DPP: ECDH shared secret (M.x) - hexdump(len=32):
D (64858) wpa: f0 8c 6a ab fc 7b b8 4e 1b 6f f0 ff b7 2d a7 51
[... 128 lines cut ...]
D (66028) wpa: 2a 26 5b f9 8f 6a cd a1 18 3b 6f 8b 4f ed e4 15
D (66038) wpa: f8 25 b0 c9 35 7f c6 15 bb d2 77 dc b9 33 9b 5b
D (66038) wpa: 73 5b 8c 81 8e 96 45 c9 08 d6 39 7c 20 90 4c 70
D (66048) wpa: 14 06 87 41 e0 96 3d 9e b7 5a 9e b1 4e 7e
D (66048) wpa: DPP: Mgmt Tx - MAC:ac:3e:b1:02:80:e0, Channel-6, WaitT-500        
D (66138) event: running post WIFI_EVENT:19 with handler 0x400d6adc and context 0x3ffbdcfc on loop 0x3ffbc9f0
0x400d6adc: event_handler(void*, char const*, int, void*) at C:/Users/Alma/sensor-data-update/main/main.cpp:55

D (66138) event: running post WIFI_EVENT:19 with handler 0x400e0420 and context 0x3ffcf020 on loop 0x3ffbc9f0
0x400e0420: offchan_event_handler at C:/Users/Alma/esp443/esp-idf/components/wpa_supplicant/esp_supplicant/src/esp_dpp.c:449

D (66148) wpa: Mgmt Tx Status - 1, Cookie - 0x400e036c
0x400e036c: esp_supp_rx_action at C:/Users/Alma/esp443/esp-idf/components/wpa_supplicant/esp_supplicant/src/esp_dpp.c:413

I (66148) wifi dpp-enrollee: DPP Auth failed (Reason: ESP_ERR_DPP_TX_FAILURE), retry...
D (66658) event: running post WIFI_EVENT:20 with handler 0x400d6adc and context 0x3ffbdcfc on loop 0x3ffbc9f0
0x400d6adc: event_handler(void*, char const*, int, void*) at C:/Users/Alma/sensor-data-update/main/main.cpp:55

D (66658) event: running post WIFI_EVENT:20 with handler 0x400e0420 and context 0x3ffcf050 on loop 0x3ffbc9f0
0x400e0420: offchan_event_handler at C:/Users/Alma/esp443/esp-idf/components/wpa_supplicant/esp_supplicant/src/esp_dpp.c:449
```

### More Information.

Seems afterwards it crashes Wi-Fi module.

knight-ryu12, 2023-01-26:
Update on this:
if, LWIP `Enable reassembly incoming fragmented IP4 packets` are enabled, above example will not work. if disabled, it suddenly work now.

Hmm.

jasta, 2023-02-21:
I'm having the same issue (dpp-enrollee yields ESP_ERR_DPP_TX_FAILURE).  For me it is strangely persistent but unpredictable.  That is, I've had it work correctly in the past, but somehow once a given device gets into this state it will yield this error over and over no matter what I do, even if I reset the flash.   I believe @knight-ryu12 is describing something similar where a seemingly unrelated change "fixed it".  Looks like others have also experienced this: https://esp32.com/viewtopic.php?t=28573.

Needless to say this unpredictability makes it impossible to actually ship something that uses DPP, so we need to find some predictable way of detecting and working around this condition.

kapilkedawat, 2023-02-21:
Hi @jasta , we are trying to reproduce it locally. Are you using the example? 

Please share your sdkconfig as well as IDF version commit.

jasta, 2023-02-21:
Yes, I'm able to reproduce consistently with the unmodified example on the current v4.4.4 tag of esp-idf.  I am using the default settings (channel 6, no provided key/device info).  Here's the sdkconfig: https://gist.github.com/42ef0f07990ca812bba8b541685ef798

jasta, 2023-02-21:
The bug really smells like a race condition IMO.  I had it all working perfectly and changed nothing of significance about my app and it just started giving my ESP_ERR_DPP_TX_FAILURE over and over.  Then no amount of reverting my code could fix it, including reverting all the way back to the unmodified example which is where I started.  It of course used to work with the sample and even my considerably more robust full app.  I suspect that the "condition" that changes to cause it to become persistently broken is literally in the air -- something about my Wi-Fi setup must be able to consistently reproduce an "unlikely" race condition outcome.

jasta, 2023-02-21:
Well that's interesting, even though I get ESP_ERR_DPP_TX_FAILURE pretty much every time, I now just got ESP_ERR_DPP_INVALID_ATTR.  Adds quite a bit of evidence to my theory that this is a race :)

jasta, 2023-02-21:
As an aside, I am the one working on adding DPP support to Rust and ran into this issue maturing the implementation even though I didn't change anything functionally interesting with respect to esp_* calls: https://github.com/esp-rs/esp-idf-svc/pull/228 .

jasta, 2023-02-21:
Here's a log with all the debugging turned up:

https://gist.github.com/8e142d289b991838c56694c54bb20889

kapilkedawat, 2023-02-21:
> I have logs with the full debugging turned up but I'm not posting here as I believe they will contain my Wi-Fi creds. Lmk and I can share them privately or reproduce with a dummy network (though I suspect changing my network around will change the results)

Please black out wifi cred from the logs and share the rest.

jasta, 2023-02-21:
I can't actually tell what is and isn't sensitive about this, but I'm interested enough in getting this solved that I'll risk it hehe: https://gist.github.com/09fe320e7b549967b37088170e59c5cb

I confirmed again this morning it still repros.  I seemingly have a 100% reliable repro (it's happened at least the last 20 or 30 times I've tried) on a devkit module that once worked just fine.

knight-ryu12, 2023-02-21:
I don't know for sure, but it crashes Wi-Fi event handler hardly afterwards. I cannot receive any events after DPP had failed. I assume that might have to do with how LWIP is handling packets? (Or is it just my condition being bad...)

jasta, 2023-02-21:
> I don't know for sure, but it crashes Wi-Fi event handler hardly afterwards. I cannot receive any events after DPP had failed. I assume that might have to do with how LWIP is handling packets? (Or is it just my condition being bad...)

I see the same thing re the hard crash, but I don't believe it is related to LWIP packet fragmentation.  These aren't even IP packets AFAICT, they're action frame packets in 802.11.  Further, I'm seeing essentially identical behavior to you with the unmodified example, which I'm guessing is the same thing our friends at Espressif are testing with but seeing different results.  Highly likely an environmental condition causing the difference.

gayafhannah, 2023-02-21:
> I see the same thing re the hard crash, but I don't believe it is related to LWIP packet fragmentation. These aren't even IP packets AFAICT, they're action frame packets in 802.11. Further, I'm [...]

Afaik it doesnt actually crash anything, it just sets `s_dpp_stop_listening` to true after getting the first packet, which stops it from sending more ROC events recursively, with the intention of setting `s_dpp_stop_listening` to false upon a successful dpp configure, and then sending a ROC event to start the recursive loop of them using `esp_supp_dpp_start_listen()`

From my testing, the wifi driver and event loop are all still fully functioning even when that error happens, its just there's no events actually going on to show up in the log but if you set `s_dpp_stop_listening` to false and call the start_listen function again, or if you manually post events, they'll show up normally

jasta, 2023-02-21:
> Afaik it doesnt actually crash anything, it just sets `s_dpp_stop_listening` to true after getting the first packet, which stops it from sending more ROC events recursively, with the intention of [...]

Ahh good catch, however what I'm seeing is that `esp_supp_dpp_start_listen()` does not actually once again start listening.  That is, if you hit Retry in the Android Wi-Fi UI it will fail without printing any more debug output on the esp32.  Looking deeper from your analysis, it looks like the reason is because `esp_supp_dpp_start_listen()` doesn't clear the `s_dpp_stop_listening` flag so even if you call `esp_supp_dpp_start_listen()` again the call to `offchan_event_handler` trying to trigger `SIG_DPP_LISTEN_NEXT_CHANNEL` will be silently ignored (because `s_dpp_stop_listening` is still true).  The code is subtle in that no matter what it'll get one `SIG_DPP_LISTEN_NEXT_CHANNEL` event handled after `esp_supp_dpp_start_listen()`, but the timeout is 500ms and it will not try again even on the same channel.

So at the very least we've identified one clear bug that needs fixing: `esp_supp_dpp_start_listen()` needs to call  `s_dpp_stop_listening = false`.

I'm going to go try your MBO workaround now and report back

gayafhannah, 2023-02-21:
@jasta I've also just noticed that since i'm not testing on the master branch and am instead on, in the master branch it seems to be called `ESP_WIFI_MBO_SUPPORT` in Kconfig, but `WPA_MBO_SUPPORT` in `5.0.0`
I'll have to try grab myself the master branch and see how that goes

jasta, 2023-02-21:
@gayafhannah, MBO support with v4.4.4 at least didn't seem to affect the results, though I still think you're onto something with your analysis.  My network set-up is that my phone is connected to a Google Wi-Fi network on 5GHz ch 149, but the 2.4GHz network is on channel 1.  I've tried a bunch of configurations of different channels for DPP to use though and still can't get it to work.  I'm a bit stumped still.

gayafhannah, 2023-02-21:
> @gayafhannah, MBO support with v4.4.4 at least didn't seem to affect the results, though I still think you're onto something with your analysis. My network set-up is that my phone is connected to a Google Wi-Fi network on 5GHz ch 149, but the 2.4GHz network is on channel 1. I've tried a bunch of configurations of different channels for DPP to use though and still can't get it to work. I'm a bit stumped still.
> 
> Note: I used idf.py menuconfig to enable MBO support so the settings should match my local checkout. I'm using the v4.4.4 tag in esp-idf.

I know that without MBO support, i was able to get it to be more consistent at not having the error(although not 100%) if the only channel dpp was configured to listen on was the channel of the wifi network, and with my phone also connected to that network, and therefore on that same channel, which is what led me to mbo support when snooping around menuconfig.

jasta, 2023-02-21:
Disabling MBO and setting channels to 1,6 worked once, but then I rebooted the device and tried one more time and it failed.  Then rebooted and tried again several times until it finally worked.  How bizarre.

gayafhannah, 2023-02-21:
Have thrown together a pull request #10812 that should work for the latest version of espidf with the changes i made to get fully functional dpp
I should note though that i haven't tried using the example in the repo and instead have been testing with my own heavily modified version, although there shouldn't be much difference in how they work in the end.
Also @jasta there may be a chance that on v4.4.4 MBO support may not work quite the same, especially considering its an entire major release difference.
99% of my testing has been done on v5.0.0

jasta, 2023-02-21:
> Have thrown together a pull request #10812 that should work for the latest version of espidf with the changes i made to get fully functional dpp

Looks good, you might want to also include my fix to esp_supp_dsp_start_listen which resets the s_dpp_stop_listening flag.  This is what enables retries to work as was intended by the example (i.e. not needing deinit/init before it can be retried).

> I should note though that i haven't tried using the example in the repo and instead have been testing with my own heavily modified version, although there shouldn't be much difference in how they [...]

Planning to retest all this on v5.0.1 but was sticking to v4.4.4 because that's where it once worked then stopped so I figured we could learn the most about what exact environmental difference actually matters.

gayafhannah, 2023-02-21:
> Looks good, you might want to also include my fix to esp_supp_dsp_start_listen which resets the s_dpp_stop_listening flag. This is what enables retries to work as was intended by the example (i.e. not needing deinit/init before it can be retried).

I'll have to give your fix a shot in the morning
That fix was actually the first thing I tried, and while it did allow me to retry even after significant errors, it had some... strange side effects. I can't explain what caused them exactly but that could just be because I was really fucking with esp_dpp.c to figure out just where was it all going wrong and somehow because of that little fix I was getting too many ROC events every second to even bother counting.
Hopefully that change works better with a cleaner and working esp_dpp.c than it did this morning.

Also my guess for environmental differences causing the issues is just the phone/AP switching channels. It's the only thing I can come up with since like I said in earlier comments, setting the listen channel to be the same as the ap and phone channel, maybe 80-90% success rate atleast. Instead of the not even 1% chance if it wasn't the same. A fun way to test it was to in the channel listen list that you put into one of the functions, just up all channels 1 through to 13, I also added a little thing in esp_dpp to show the current channel as it was cycling through. If I managed to get the timing pressing the button on my phone just right to land on channel 1 on the esp(channel 1 also what the ap and my phone were on) then it'd almost always connect fine, but if I missed that channel(difficult to get it right when the timeout is 500ms) then it'd fail to connect except for maybe exactly one time.

Also if I remember correctly, setting the channel to be near the same as the ap, if the channel width for the ap is 40MHz, I noticed even being +-2 channels from the centre channel, it had moderate success rates. Definitely not the same as setting the listen channel directly ontop of the centre channel though. Also 40MHz wide ones were slightly less successful on the centre channel than 20MHz wide(probably because 20MHz is literally just 1 channel wide).
I have no explanation for this stuff to do with channel width,  it's just something I noticed during testing since the extender here is 40MHz wide and the main ap is 20MHz wide and I was constantly switching networks. I feel like I'm going crazy because I have no solid evidence and it could all just be a coincidence and there was some other environmental stuff going on, but I spent alot of time testing things and that pattern of width playing a small part(and being directly ontop of the right channel being an even bigger part) have been constants the entire time.

jasta, 2023-02-21:
> I'll have to give your fix a shot in the morning That fix was actually the first thing I tried, and while it did allow me to retry even after significant errors, it had some... strange side effects. [...]

I'm not observing any strange side effects from the fix FWIW.  And more importantly I'm noticing that my phone will actually automatically retry now with this setting on, indicating that the DPP standard actually expects some intermittent failures (at least due to RF interference would be my guess, but maybe even this channel weirdness).  That suggests to me that the lack of working retry is a big part of the root cause here.  Imagine what would happen to our global TCP infrastructure if you disabled packet retransmits, for example...

> Also my guess for environmental differences causing the issues is just the phone/AP switching channels. It's the only thing I can come up with since like I said in earlier comments, setting the [...]

Definitely plausible and consistent with my experience.  If I just sit there and hit Retry on my Android phone over and over with my fix it will eventually work.  Not ideal UX by any stretch, though, and I'd be quite surprised if we (or maybe just Espressif) couldn't come up with a more reliable way to fix this.

> Also if I remember correctly, setting the channel to be near the same as the ap, if the channel width for the ap is 40MHz, I noticed even being +-2 channels from the centre channel, it had moderate success rates. Definitely not the same as setting the listen channel directly ontop of the centre channel though. Also 40MHz wide ones were slightly less successful on the centre channel than 20MHz wide(probably because 20MHz is literally just 1 channel wide). I have no explanation for this stuff to do with channel width, it's just something I noticed during testing since the extender here is 40MHz wide and the main ap is 20MHz wide and I was constantly switching networks. I feel like I'm going crazy because I have no solid evidence and it could all just be a coincidence and there was some other environmental stuff going on, but I spent alot of time testing things and that pattern of width playing a small part(and being directly ontop of the right channel being an even bigger part) have been constants the entire time. This would also likely explain why it sounds like others have been having much more inconsistent errors, since if the target ap has a wide channel and being close to a wide one increases chances but only a bit, whereas for me having a network where if I'm not right on it, it doesn't work, until I use the extender but then it works only sometimes

I can't really say for sure, I'd probably need to dive deep into how this stuff is even supposed to function at the standard level before I could speculate why that would play a role.  One curious thing I did notice however was that the QR code does in fact include the channel that the enrollee (the esp32) is listening on, so it doesn't make any sense why the configurator (the phone) would have any troubles sending/receiving messages on the correct channel.

gayafhannah, 2023-02-22:
> I can't really say for sure, I'd probably need to dive deep into how this stuff is even supposed to function at the standard level before I could speculate why that would play a role. One curious [...]

I noticed the channels being listed in the qr code aswell, and your exact reasons are why I'm hesitant that my guess is actually correct. It's just that pattern being there made it really seem like it has to have something to do with the channel being wrong. Especially because the error afaik is caused by the esp failing to send the packet "offchannel" and not in receiving any packets, as even with the error happening, it still successful receives the first packet/frame from the phone just phone.
I'd love to give it a try with my phone not connected to any network while doing dpp so it's not bound to a channel(although when my phone can handle 2 wifi connections like it's nothing that's hard to believe it's an issue) except thanks to Samsung's Oneui being bored, I can barely access the dpp provisioning to begin with, and I can't select any network other than the current one for provisioning, so I can't test that.

jasta, 2023-02-28:
The plot thickens.  It looks like if you try to work around this issue by calling esp_supp_dpp_deinit/init again and just starting all over, you will quickly run into a race condition in esp_dpp.c in which this assert fails:

https://gist.github.com/a067c88e511dfff52e7f704d469e157f

I am using unmodified ESP-IDF 4.4.4 to cause this behaviour.

The reason appears to be because of an inherent race condition in which esp_supp_dpp_init makes no effort to check if the previous s_dpp_evt_queue and associated task is drained and cleaned up before proceeding to start up esp_dpp_task again.  Illustrated as such:

| Caller Task | dppT Task #1 | dppT Task #2 |
| ----- | ----- | -----|
| dpp_init() | | |
| | xQueueReceive (blocked) | |
| dpp_deinit() | | |
| | Handling SIG_DPP_DEL_TASK | |
| dpp_init() | |
| | | xQueueReceive (blocked) | |
| | s_dpp_evt_queue = NULL; | |
| | | xQueueReceive crashes: s_dpp_evt_queue is NULL) |

It seems to me like the fix would be to have a condition variable and associated state that deinit sets true, SIG_DPP_DEL_TASK sets false, and init checks for / waits until it is false.

jasta, 2023-03-01:
@gayafhannah Did you ever test on the master branch?  I started prepping my separate s_dpp_stop_listening fix and noticed that while v5.0.1 still has issues, the master branch is actually working correctly!  I checked in wireshark on a promiscuous third party device and discovered that in fact the OTA packets are distinct between v4.4.4/v5.0.1 and master.  For the former, there are several action frame transmits in response to the initial auth request from my phone but they are all ignored by my phone.  The phone eventually times out and gives up, despite in theory the packets being made available.  For master, there is exactly one transmit of auth response from the esp32 and the phone picks it up right away and the whole process completes successfully really fast.

The only difference I can think of here is if the wifi libs got updated in master and fixed some subtle but important bug in the action frame tx path.

Next step I was going to try setting up wpa_supplicant on my PC so I can increase the debug logs and understand why the packets I can see sent from the esp32 are not being accepted/received by the peer, but this is slow going :)

shreyassheth1, 2023-03-16:
Hey @gayafhannah @knight-ryu12 @jasta we are actively looking into this issue. For us, it is little bit hard to stimulate one consistent behavior . Also @gayafhannah regarding your PR we are trying to make more changes and make the state machine cleaner. Meanwhile, you can try out these changes 
- Configure `BOOTSTRAP_ROC_WAIT_TIME` in `components/wpa_supplicant/esp_supplicant/src/esp_dpp_i.h` to **650**
- Please make sure you have disabled wpa_supplicant logs

jasta, 2023-03-16:
Note that I was able to make things work with https://github.com/espressif/esp-idf/pull/10865.  I also dug into Wireshark a bit and saw some fairly surprising results (packets going out but not picked up by the configurator phone) which made me think this is why they changed the way the initial listening phase works in DPP V2.  I think the right path to go down now is to upgrade wpa_supplicant.

ragnarmargus, 2023-05-03:
Hey, I am experiencing the same issue. DPP fails to connect. Is there a workaround without problems?

jasta, 2023-07-24:
> Hey, I am experiencing the same issue. DPP fails to connect. Is there a workaround without problems?

Can you try with a version that has my patches applied (looks like espressif backported this to v4.3, v4.4 and v5.0 branches, but I don't think any versioned releases were made with the fix)?  What you should see is that there are many automatic retries if you test with the dpp-enrollee sample and it will eventually work.  In either case, can you report your findings here?

ragnarmargus, 2023-08-01:
> Can you try with a version that has my patches applied (looks like espressif backported this to v4.3, v4.4 and v5.0 branches, but I don't think any versioned releases were made with the fix)? What [...]

Thanks for pointing out that you patched this. We decided to go with smartConfig instead. Maybe I will return to this in the future.

Stencht, 2023-08-26:
Hmm I'm also experiencing the same issue when running the dpp-enrollee sample, which has the fix, using:

- ESP-IDF V5.0/release (V5.0.3)
- ESP32-S3

I got it properly working on the master branch (V5.2), but I had to return to V5.0 because of incompatibility of other dependencies.
Now I get inconsistent results. I got it to work once, and other times I get errors like:

```
I (891) phy_init: phy_version 610,2bff4c8,Jul 27 2023,20:22:14
I (931) wifi:mode : sta (f4:12:fa:db:91:d8)
I (931) wifi:enable tsf
I (941) wifi dpp-enrollee: Started listening for DPP Authentication
I (24841) wifi dpp-enrollee: DPP Auth failed (Reason: ESP_ERR_DPP_INVALID_ATTR), retry...
I (24851) wifi dpp-enrollee: DPP Auth failed (Reason: ESP_ERR_DPP_INVALID_ATTR), retry...
I (24871) wifi dpp-enrollee: DPP Auth failed (Reason: ESP_ERR_DPP_INVALID_ATTR), retry...
I (24871) wifi dpp-enrollee: DPP Auth failed (Reason: ESP_ERR_DPP_INVALID_ATTR), retry...
I (24891) wifi dpp-enrollee: DPP Auth failed (Reason: ESP_ERR_DPP_INVALID_ATTR), retry...
I (24911) wifi dpp-enrollee: DPP Authentication failed after 5 retries
I (24911) main_task: Returned from app_main()
```

And
```
I (888) phy_init: phy_version 610,2bff4c8,Jul 27 2023,20:22:14
I (928) wifi:mode : sta (f4:12:fa:db:91:d8)
I (928) wifi:enable tsf
I (938) wifi dpp-enrollee: Started listening for DPP Authentication
I (16848) wifi dpp-enrollee: DPP Auth failed (Reason: ESP_ERR_DPP_FAILURE), retry...
```

I also tried to set the config options:
```
CONFIG_WPA_11KV_SUPPORT=y
CONFIG_WPA_SCAN_CACHE=y
CONFIG_WPA_MBO_SUPPORT=y
CONFIG_WPA_DPP_SUPPORT=y
```
But that didn't resolve it either.

Any ideas?

Shreyas0-7, 2024-07-25:
Hey @Stencht @jasta @gayafhannah 
We have prepared a patch which we think will fix this issue and need your help to test it. This patch is based on v5.0.3
- IDF patch [dpp_off_tx.zip](https://github.com/user-attachments/files/16376887/dpp_off_tx.zip)
 you can unzip and apply this patch in `esp-idf` using command `git am dpp_off_tx.patch`
- Lib patch [dpp_off_tx_lib.zip](https://github.com/user-attachments/files/16376900/dpp_off_tx_lib.zip)
 you can unzip and apply this patch in `esp-idf/components/esp_wifi/lib` using command `git am --binary dpp_off_tx_lib.patch`

michael-betz, 2024-07-31:
@Shreyas0-7 I tried your patch with an ESP32 + Android 14 (Lineage OS) phone.

Half the time it works. The other half, nothing is printed on the serial console, Android times out with `Couldn't add device` and offers me to Retry. Clicking the Retry button always leads to another timeout. The only way to get a success in this case is to reboot the ESP32 and scan the QR code again.

Shreyas0-7, 2024-07-31:
@michael-betz Can you please set ESP into debug mode and attach ESP serial logs if possible

AxelLin, 2024-08-06:
@michael-betz Any feedback about comment https://github.com/espressif/esp-idf/issues/10615#issuecomment-2261070551 ?

michael-betz, 2024-08-06:
Hi, sorry for the delay.

Here's the complete debug log of dpp failing 5 times (I push the retry button on my phone 5 times)

[failed_after_5_retries.txt](https://github.com/user-attachments/files/16507389/failed_after_5_retries.txt)

kapilkedawat, 2024-08-06:
Hi @michael-betz, thanks for the logs. Was the mobile connected to the AP during this time, and was the AP operating on another channel? From the logs, it looks like the mobile didn't receive the frames sent by the device (it didn't wait for sufficient time). T

his issue was later realized, and DPP R2 has provisions to mitigate this type of scenario.

Could you also try setting the ESP listen channel to be the same as the AP's channel and test it again?

AxelLin, 2024-08-14:
@michael-betz Any update regarding the comment in https://github.com/espressif/esp-idf/issues/10615#issuecomment-2271641093 ?

Zmmfly, 2024-08-28:
IDF version: `5.3`

Same problem, but when i read code, i found that caused by hash compare of keys:
https://github.com/espressif/esp-idf/blob/0bbd72819602b4b8cf786a8833d0ee9c922268bf/components/wpa_supplicant/esp_supplicant/src/esp_dpp.c#L165-L172

The log start with DPP key pair generation until share AP secret failed from Android with QR code scan.
```
D (8736) wpa: DPP: Generating a keypair
E (8736) wpa: curve: secp256r1
E (8736) wpa: prime len is 32
E (8736) wpa: Q_x: - hexdump(len=32):
E (8737) wpa: da 13 d2 0d 14 a4 bc c4 4c 60 52 1b bb e2 99 d4
E (8738) wpa: 33 4f 6f 05 fa c5 84 0d a7 78 83 f7 01 dc 8e 5b
E (8738) wpa: Q_y: - hexdump(len=32):
E (8739) wpa: 23 59 c0 83 8a 32 57 a5 f1 e2 cb ca 4f 97 09 c5
E (8740) wpa: 2e 40 0b 22 24 1b b2 f1 87 80 a4 c8 37 21 d1 32
E (8740) wpa: d:      - hexdump(len=32):
E (8741) wpa: 6b e5 79 52 ce bc d4 83 2f ba 29 39 43 3c d8 94
E (8742) wpa: 59 e7 ad 8e d2 a7 ab 17 cd e9 37 57 a1 41 7a 01
D (8743) wpa: DPP: Compressed public key (DER) - hexdump(len=59):
D (8744) wpa: 30 39 30 13 06 07 2a 86 48 ce 3d 02 01 06 08 2a
D (8744) wpa: 86 48 ce 3d 03 01 07 03 22 00 02 da 13 d2 0d 14
D (8745) wpa: a4 bc c4 4c 60 52 1b bb e2 99 d4 33 4f 6f 05 fa
D (8746) wpa: c5 84 0d a7 78 83 f7 01 dc 8e 5b
D (8746) wpa: DPP: Public key hash - hexdump(len=32):
D (8747) wpa: 77 e7 21 a5 b5 a4 e4 ab 94 cc d9 cb 29 29 99 24
D (8748) wpa: a0 60 20 7a 37 6d 64 77 72 b7 ac 52 41 71 18 eb
D (8748) wpa: DPP: URI mac: 30:30:f9:5a:8a:54
D (9251) wpa: DPP: Sent event 3 to DPP task
~~ Repeat ~~
D (11258) wpa: DPP: Sent event 3 to DPP task
I (01/01/70 00:00:10.598) [  scr:wifi_sc  ] update qrcode done
D (11760) wpa: DPP: Sent event 3 to DPP task
~~ Repeat ~~
D (15772) wpa: DPP: Sent event 3 to DPP task
D (15788) wpa: DPP: Sent event 2 to DPP task
D (15788) wpa: DPP: Rx Public Action frame: action - 9
I (15789) wpa: DPP: Authentication Request from b6:c0:af:07:30:b6
V (15790) wpa: DPP: Responder Bootstrapping Key Hash - hexdump(len=32):
V (15790) wpa: 09 bb 6f cc 65 d0 90 e7 61 e5 00 d5 a7 91 8e f3
V (15791) wpa: 38 19 16 07 d0 5b a4 28 b5 a7 1b ff 1d 48 94 4a
V (15791) wpa: DPP: Initiator Bootstrapping Key Hash - hexdump(len=32):
V (15792) wpa: a9 18 43 12 b6 8b fa 58 b9 f8 3c 00 b7 6f 99 b2
V (15793) wpa: 42 a1 4c d2 da 56 d8 55 46 68 fb 6c 38 4f 92 07
I (15794) wpa: DPP: No matching own bootstrapping key found as responder - ignore message
D (15795) wpa: DPP: Sent event 3 to DPP task
E (01/01/70 00:00:14.657) [ antx:idf:wifi ] DPP auth fail, reason: ESP_ERR_DPP_INVALID_ATTR, data: 0x3099
```

Kaden-mpc, 2024-09-05:
I have experienced issues with DPP running on IDF v5.2.1, out of 47 attempts I experienced a failure rate of 51%. 40 attempts were made with verbose logging enabled, and WPA logs disabled, and 7 attempts were made with both verbose logging and WPA logs enabled.

- IDF version: `v5.2.1`
- Test device: `ESP32-S3-DevKitC-1`
- Mobile device used: `Motorola Edge 2021`
- Mobile device OS: `Android 13`

[Logs of all failures can be found here](https://github.com/tovala/ESP-IDF-DPP-Fail-Logs)

----

> Then I enabled `WPA_DEBUG_PRINT` in menuconfig and re-compiled.
> 
> Surprisingly, the next 15 attempts **all** succeeded. Even after changing `EXAMPLE_DPP_LISTEN_CHANNEL_LIST` to 12 and letting the phone connect to the wifi.
> 
> Then I disabled `WPA_DEBUG_PRINT`, re-compiled and only 4 out of 8 attempts succeeded.
> 
> What a strange result.
> 

> 
> With `WPA_DEBUG_PRINT` enabled: 7 / 8 attempts succeeded
> 
> Same setup but `WPA_DEBUG_PRINT` disabled: 2 / 8 attempts succeeded.

As @michael-betz has [also experienced](https://github.com/espressif/esp-idf/issues/10615#issuecomment-2288387428), changing the value of `WPA_DEBUG_PRINT` changes the results a lot, without any other code changed.

Aditi-Lonkar, 2024-09-12:
Hi @Kaden-mpc , I am sharing a debug build for dpp_enrollee example on v5.2.1 here with some extra logs and few fixes. 
Will you please test with this build and share the logs and sniffer capture?

[dpp_test_build_s3.zip](https://github.com/user-attachments/files/16977874/dpp_test_build_s3.zip)

The file flash_cmd contains command for flashing. Please use that to flash the device. Thanks.

Kaden-mpc, 2024-09-12:
@Aditi-Lonkar I have collected 20 logs of the build you shared, 8/20 failed. I have not yet collected any network packets. I will get back to you with those, and additional logs when I get them. [The logs can be found here](https://github.com/tovala/ESP-IDF-DPP-Fail-Logs/tree/master/09122024_debug_logs). Thank you for jumping on this issue!

Aditi-Lonkar, 2024-09-13:
@Kaden-mpc Thanks for trying the earlier build and sharing the logs. We are actively working on the issue, however we are not able to reproduce exact same results in our setup. 

I am attaching one more build here with some more fixes added.  Will you please test and share the logs and sniffer capture with this too? 

The fixes includes
1) retrying sending auth response for 5 times on same channel in 1second interval if we didn't get auth confirm.
2) starting timer of 1second for getting auth confirm after getting ack for auth_response
3) staying on current channel until the whole auth process finishes
4) stopping listen when dpp returns failure and some reason code related changes.
5) keep rf enable until the offchannel transmission is done
6) starting a timer to fail if config not received in 120 secs

[dpp_test_build_521_2.zip](https://github.com/user-attachments/files/16994349/dpp_test_build_521_2.zip)
The file flash_cmd contains command for flashing. Please use that to flash the device.

Thanks.

ryanrampage1, 2024-09-13:
@Aditi-Lonkar thank you for the responsiveness, @Kaden-mpc and I have the ESP logs from the new version but do not have the sniffer capture we are able to share yet due to our company data security policy. We are working on an alternative way to get those logs and will get that over to you early next week.

 Thank you again for the help!

Kaden-mpc, 2024-09-16:
@Aditi-Lonkar we have acquired logs alongside network traffic, however the access point used did not support any form of pcap, ssh, or port mirroring, and an external device was used to capture the packets using wireshark. Let us know if you require any additional logs, or if we need to acquire a AP that supports proper pcap. Additionally let us know if you have any recommendations for capturing more thorough debug information to aid with resolving issues. The logs and packet captures can be [found here](https://github.com/tovala/ESP-IDF-DPP-Fail-Logs/tree/master/09162024_debug_logs). Thank you for the ongoing support!

Kaden-mpc, 2024-09-17:
@Aditi-Lonkar we have captured more logs, with router network sniffing as well. Out of 25 attempts, we had 2 failures. The logs can be [found here](https://github.com/tovala/ESP-IDF-DPP-Fail-Logs/tree/master/09172024_debug_logs). Thank you for the help!

kapilkedawat, 2024-09-19:
Hi @Kaden-mpc,

Thank you for sharing the logs and helping us debug this issue further. What we need is a wireless capture where we can verify whether the mobile device is acknowledging the packets. You can see how to capture wireless packets [here](https://github.com/espressif/esp-hosted/issues/387#issuecomment-2185987082).

In our internal testing, we have observed that sometimes the station stops receiving DPP frames altogether from the responder (ESP) device. We believe this is an issue with the mobile device, as the ESP responds to the frames within the stipulated timeframe. This behavior seems inconsistent, which explains why the connection sometimes works flawlessly and at other times fails.

If you are able to consistently reproduce the issue, could you try the same test with a different responder device (not ESP) to see if the problem persists?

Kaden-mpc, 2024-09-19:
@kapilkedawat I have collected more logs, and wireless packet captures using iw monitor mode, which can be [found here](https://github.com/tovala/ESP-IDF-DPP-Fail-Logs/tree/master/09192024_debug_logs). We will send logs and sniffs of non ESP preforming DPP, tomorrow or early next week. Let us know if there is anything else we can provide to aid in debugging. Thanks for the support!

Kaden-mpc, 2024-09-20:
@kapilkedawat We have tested a non ESP device (Google Chromecast) using the same AP, phone, sniffer, and same relative device position. We continued retrying the same experiment until we received a failure, which took 50 attempts. Whereas with the ESP we are able to consistently reproduce roughly every 10 attempts. Additionally, the Chromecast DPP appears much faster, with the success message appearing on the phone almost instantaneously, and never shows the retry screen, whereas with the ESP DPP it almost always take a few seconds, and every few attempts shows the retry screen (as shown in [09122024](https://github.com/tovala/ESP-IDF-DPP-Fail-Logs/tree/master/09122024_debug_logs), attempts 3, 8, and 15. [09132024](https://github.com/tovala/ESP-IDF-DPP-Fail-Logs/tree/master/09132024_debug_logs) attempt 11. [09172024](https://github.com/tovala/ESP-IDF-DPP-Fail-Logs/tree/master/09172024_debug_logs) attempt 12. And potentially other runs where I marked it as successful without noting retries).

The Chromecast sniffs [can be found here](https://github.com/tovala/ESP-IDF-DPP-Fail-Logs/tree/master/09202024_chromecast_sniffs). Thank you for the support!

ryanrampage1, 2024-09-25:
@Aditi-Lonkar @kapilkedawat wanted to follow up and see what other information we can provide to assist. I also tested on the Chromecast and was able to get 50 consecutive successes on the Chromecast showing that It is not an issue with the mobile device. We look forward to hearing from you!

kapilkedawat, 2024-09-26:
Hi @ragnarmargus @Kaden-mpc, apologies for the delayed response. Yes, Chromecast sessions are faster since Chromecast has significantly more computing power compared to the ESP32. However, we are looking into optimizing the code further to improve performance, which may, in turn, reduce the error rates. This is still just a hunch, but we will try to provide another build tomorrow.

kapilkedawat, 2024-09-30:
HI @ragnarmargus @Kaden-mpc apologies for the delay. Could you please try with the attached experimental build, this reduces response time of DPP frames processing.
[test_build_30_sept.zip](https://github.com/user-attachments/files/17193278/test_build_30_sept.zip)

Kaden-mpc, 2024-09-30:
@kapilkedawat We have collected 50 runs from the build you shared, 6 required pressing the retry button, and 1 failed. There is a noticeable increase in speed between scanning the QR code and receiving the success status. The failure that we experienced did not succeed on pressing the retry button, or rescanning the QR code; is this a fatal failure, or would rerunning the full DPP process enable another attempt at preforming DPP, similarly to how Chromecast handled its failure? The logs can be [found here](https://github.com/tovala/ESP-IDF-DPP-Fail-Logs/tree/master/09302024_debug_logs#readme). Thank you for the ongoing support!

kapilkedawat, 2024-10-01:
Hi @Kaden-mpc,

Thank you very much for your continued testing and providing the results, really appreciate the same.

The failure in attempt 46 is recoverable, and I have made some changes to handle that. Unfortunately, we cannot make the response faster than this in current CPU freq, as we are limited by the computational power available in the chip. DPP request frame parsing and the creation of the response frame involve two ECDH operations, each taking more than ~150 ms.

I have created another build that includes a fix for the issue mentioned above. This build also increases the CPU frequency to 240MHz to make the operations a bit faster.

Please help to test and provide the results.

[test_build_1_oct.zip](https://github.com/user-attachments/files/17204735/test_build_1_oct.zip)

Kaden-mpc, 2024-10-01:
@kapilkedawat We have collected 50 more logs using your latest build. 5 of the attempts required retries, and 2 were failures. For 1 of the 2 failures, the ESP connected to the wifi successfully, but the phone showed a failure. And the other stopped responding, with no errors shown in the logs, and the sniff showing the ESP not responding. 

[Logs can be found here](https://github.com/tovala/ESP-IDF-DPP-Fail-Logs/tree/master/10012024_debug_logs#readme)

Thanks for the quick response, and for the support!

kapilkedawat, 2024-10-03:
Hi @Kaden-mpc,

Thank you for your invaluable testing efforts. I think we can now focus only on the non-recoverable errors.

Instance 49 is recoverable, but the timeout for that was 120 seconds. We can add some further optimizations there. I have made some adjustments to the timeout in the gas query. Please try the attached build.

[test_build_3oct.zip](https://github.com/user-attachments/files/17239822/test_build_3oct.zip)

Kaden-mpc, 2024-10-03:
@kapilkedawat We have collected 50 more logs. 9 of the attempts required retires (2 of which required retrying twice). And we experienced another false failure, where the phone displayed failure but it connected to the wifi. Once non recoverable errors are resolved, will progress be made on making the retries less frequent, or are we locked at the current 10-22% requiring retries?

[Logs can be found here](https://github.com/tovala/ESP-IDF-DPP-Fail-Logs/tree/master/10032024_debug_logs#readme)

Thank you for the support!

kapilkedawat, 2024-10-07:
Hi @Kaden-mpc, thanks again for your testing efforts and for providing the logs. All the non-recoverable errors have been resolved, and the errors we are currently encountering are due to the mobile device either not replying in time or not receiving the frames (likely going off-channel). Unfortunately, there isn't much more that can be done from an optimization perspective, as we are processing the frames as fast as possible.

I want to emphasize that the ESP device is following the protocol correctly, and our behavior aligns with the current upstream wpa_supplicant codebase. You will observe different behavior with various configurations, depending on how much the driver honors wpa_supplicant's requests. For example, in our internal testing with a b43 driver based chip, we encountered 0 errors out of 10.

ryanrampage1, 2024-10-08:
@kapilkedawat a sample size of 10 attempts is likely insufficient to demonstrate that the code works consistently. We are expecting this to be utilized by hundreds of customers per day.  We would like to see at least 25 attempts to have a statistically relevant sample size. @Kaden-mpc has been doing 50 for even more consistent and repeatable data.

The consistent 10-15% failure rate we have seen on our 50 attempt test runs does not meet the quality standards our end customers expect and is not shippable. Anything more than 1-2% failure will result in a poor experience given that the DPP sharing experience is driven by the Android Operating system and we have no way to control what that 10-15% of users experience during failure.

Additionally we are having a hard time understanding how the chip is insufficient to support closer to a 98% success rate without retry as has been shown with the Chromecast. Additionally  @Kaden-mpc has utilized the same phone for all testing, both with the Chromecast and ESP device showing that is not the issue.

kapilkedawat, 2024-10-10:
Hi @ryanrampage1,

We conducted around 40 additional tests, and the error occurred 3 times during those retries.

I would like to reiterate that the issues we faced are not due to the Espressif chip but rather because the initiator is not following the protocol correctly.

For example, in the [shard logs](https://github.com/tovala/ESP-IDF-DPP-Fail-Logs/tree/master/10032024_debug_logs#readme):

* Run 3: The confirm frame was sent after ~1.5 seconds. We waited for 600ms on the channel for the response (note: the upstream supplicant waits for 500ms).
* Run 13: The initiator did not receive an authentication response, even after sending the packet 5 times. Each of these 5 retries would have included 32 hardware retries. The response was sent within 300ms, which should have been acceptable.
* Run 21: The configuration response was sent after ~1.5 seconds from the initiator.
* Run 28: No response was received for the configuration request in the first instance. In the second instance, the configuration response was sent after ~1.5 seconds.
* Run 29: The authentication confirm frame was sent after ~1.5 seconds.
* Run 32: The authentication confirm frame was sent after ~1.4 seconds.
* Run 35: The authentication confirm frame was sent after ~1.5 seconds.
* Run 36: The authentication confirm frame was sent after ~1.4 seconds.
* Run 40: The authentication confirm frame was sent after ~1.5 seconds.
* Run 47: The authentication confirm frame was sent after ~1.5 seconds.

I’ve created a [new build](https://github.com/user-attachments/files/17326654/test_10oct.zip) where the device waits for 1.6 seconds for the frame instead. Could you please try that? 

Also, do you have any ESP32C2 or ESP32C6 devkits available? We would like to check the behavior on those as well, as they have ECC hardware, which should allow the response to be sent much faster and see if that makes any difference.

[9 more replies]

closed 2025-05-07 by Alvin1Zhang