espressif/esp-idf #10619: mqtt/ssl_ds build error in ESP32S3

Jinjungdong, 2023-01-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v4.4.3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

I run following procedure with "https://github.com/espressif/esp-idf/blob/master/examples/protocols/mqtt/ssl_ds/README.md"
But build error is happend.

### What is the actual behavior?

PS D:\Project\AVR\VSC\esp_mqtt_ssl_ds_00> idf.py -p COM7 flash 
Executing action: flash
Running cmake in directory d:\project\avr\vsc\esp_mqtt_ssl_ds_00\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 d:\project\avr\vsc\esp_mqtt_ssl_ds_00"...
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.39.1.windows.1") 
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/Project/AVR/VSC/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: D:/Project/AVR/VSC/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: D:/Project/AVR/VSC/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
-- Project sdkconfig file D:/Project/AVR/VSC/esp_mqtt_ssl_ds_00/sdkconfig
Loading defaults file D:/Project/AVR/VSC/esp_mqtt_ssl_ds_00/sdkconfig.defaults...
-- Found Python3: D:/Project/AVR/VSC/.espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe (found version "3.8.7") found components: Interpreter
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "esp_mqtt_ssl_ds_00" version: 1
-- Adding linker script D:/Project/AVR/VSC/esp_mqtt_ssl_ds_00/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script D:/Project/AVR/VSC/esp-idf/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script D:/Project/AVR/VSC/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script D:/Project/AVR/VSC/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script D:/Project/AVR/VSC/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script D:/Project/AVR/VSC/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script D:/Project/AVR/VSC/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script D:/Project/AVR/VSC/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib-time.ld
-- Adding linker script D:/Project/AVR/VSC/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32s3 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test ieee802154 jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl openthread partition_table perfmon protobuf-c protocol_examples_common protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: D:/Project/AVR/VSC/esp-idf/components/app_trace D:/Project/AVR/VSC/esp-idf/components/app_update D:/Project/AVR/VSC/esp-idf/components/asio D:/Project/AVR/VSC/esp-idf/components/bootloader D:/Project/AVR/VSC/esp-idf/components/bootloader_support D:/Project/AVR/VSC/esp-idf/components/bt D:/Project/AVR/VSC/esp-idf/components/cbor D:/Project/AVR/VSC/esp-idf/components/cmock D:/Project/AVR/VSC/esp-idf/components/coap D:/Project/AVR/VSC/esp-idf/components/console D:/Project/AVR/VSC/esp-idf/components/cxx D:/Project/AVR/VSC/esp-idf/components/driver D:/Project/AVR/VSC/esp-idf/components/efuse D:/Project/AVR/VSC/esp-idf/components/esp-tls D:/Project/AVR/VSC/esp-idf/components/esp32s3 D:/Project/AVR/VSC/esp-idf/components/esp_adc_cal D:/Project/AVR/VSC/esp-idf/components/esp_common D:/Project/AVR/VSC/esp-idf/components/esp_eth D:/Project/AVR/VSC/esp-idf/components/esp_event D:/Project/AVR/VSC/esp-idf/components/esp_gdbstub D:/Project/AVR/VSC/esp-idf/components/esp_hid D:/Project/AVR/VSC/esp-idf/components/esp_http_client D:/Project/AVR/VSC/esp-idf/components/esp_http_server D:/Project/AVR/VSC/esp-idf/components/esp_https_ota D:/Project/AVR/VSC/esp-idf/components/esp_https_server D:/Project/AVR/VSC/esp-idf/components/esp_hw_support D:/Project/AVR/VSC/esp-idf/components/esp_ipc D:/Project/AVR/VSC/esp-idf/components/esp_lcd D:/Project/AVR/VSC/esp-idf/components/esp_local_ctrl D:/Project/AVR/VSC/esp-idf/components/esp_netif D:/Project/AVR/VSC/esp-idf/components/esp_phy D:/Project/AVR/VSC/esp-idf/components/esp_pm D:/Project/AVR/VSC/esp-idf/components/esp_ringbuf D:/Project/AVR/VSC/esp-idf/components/esp_rom D:/Project/AVR/VSC/esp-idf/components/esp_serial_slave_link D:/Project/AVR/VSC/esp-idf/components/esp_system D:/Project/AVR/VSC/esp-idf/components/esp_timer D:/Project/AVR/VSC/esp-idf/components/esp_websocket_client D:/Project/AVR/VSC/esp-idf/components/esp_wifi D:/Project/AVR/VSC/esp-idf/components/espcoredump D:/Project/AVR/VSC/esp-idf/components/esptool_py D:/Project/AVR/VSC/esp-idf/components/expat D:/Project/AVR/VSC/esp-idf/components/fatfs D:/Project/AVR/VSC/esp-idf/components/freemodbus D:/Project/AVR/VSC/esp-idf/components/freertos D:/Project/AVR/VSC/esp-idf/components/hal D:/Project/AVR/VSC/esp-idf/components/heap D:/Project/AVR/VSC/esp-idf/components/idf_test D:/Project/AVR/VSC/esp-idf/components/ieee802154 D:/Project/AVR/VSC/esp-idf/components/jsmn D:/Project/AVR/VSC/esp-idf/components/json D:/Project/AVR/VSC/esp-idf/components/libsodium D:/Project/AVR/VSC/esp-idf/components/log D:/Project/AVR/VSC/esp-idf/components/lwip D:/Project/AVR/VSC/esp_mqtt_ssl_ds_00/main D:/Project/AVR/VSC/esp-idf/components/mbedtls D:/Project/AVR/VSC/esp-idf/components/mdns D:/Project/AVR/VSC/esp-idf/components/mqtt D:/Project/AVR/VSC/esp-idf/components/newlib D:/Project/AVR/VSC/esp-idf/components/nghttp D:/Project/AVR/VSC/esp-idf/components/nvs_flash D:/Project/AVR/VSC/esp-idf/components/openssl D:/Project/AVR/VSC/esp-idf/components/openthread D:/Project/AVR/VSC/esp-idf/components/partition_table D:/Project/AVR/VSC/esp-idf/components/perfmon D:/Project/AVR/VSC/esp-idf/components/protobuf-c D:/Project/AVR/VSC/esp-idf/examples/common_components/protocol_examples_common D:/Project/AVR/VSC/esp-idf/components/protocomm D:/Project/AVR/VSC/esp-idf/components/pthread D:/Project/AVR/VSC/esp-idf/components/sdmmc D:/Project/AVR/VSC/esp-idf/components/soc D:/Project/AVR/VSC/esp-idf/components/spi_flash D:/Project/AVR/VSC/esp-idf/components/spiffs D:/Project/AVR/VSC/esp-idf/components/tcp_transport D:/Project/AVR/VSC/esp-idf/components/tcpip_adapter D:/Project/AVR/VSC/esp-idf/components/tinyusb D:/Project/AVR/VSC/esp-idf/components/ulp D:/Project/AVR/VSC/esp-idf/components/unity D:/Project/AVR/VSC/esp-idf/components/usb D:/Project/AVR/VSC/esp-idf/components/vfs D:/Project/AVR/VSC/esp-idf/components/wear_levelling D:/Project/AVR/VSC/esp-idf/components/wifi_provisioning D:/Project/AVR/VSC/esp-idf/components [... cut ...] rite_flash @flash_args" -D WORKING_DIRECTORY="D:/Project/AVR/VSC/esp_mqtt_ssl_ds_00/build" -P D:/Project/AVR/VSC/esp-idf/components/esptool_py/run_serial_tool.cmake"
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
PS D:\Project\AVR\VSC\esp_mqtt_ssl_ds_00>
```

### More Information.

I use the ESP-IDF with VSCode extention on Window11.
And i recieve the solution of previous problem with "mqtt/ssl_ds/configure_esp_secure_cert.py excution error in ESP32S3 (IDFGH-9116) #10517"

AdityaHPatwardhan, 2023-02-07:
Hi @Jinjungdong,
Have you first configured the DS peripheral as mentioned [here](https://github.com/espressif/esp-idf/tree/release/v4.3/examples/protocols/mqtt/ssl_ds#3-configure-the-ds-peripheral)

As I see in the above case, the application image has been built successfully.
After that, the build system tries to flash the `pre_prov.bin` which is the binary for the pre_provisioning partition (Now renamed as esp_secure_cert partition with the latest IDF)
I see that the build system is unable to find the pre_prov.bin which is generated in the configuration step that I have mentioned above. Can you please try configuring the DS peripheral and then try to build and flash the example?

Jinjungdong, 2023-02-08:
Thank you for your response. I will check it!

Jinjungdong, 2023-02-09:
Hi @AdityaHPatwardhan
 
I have tried that as you emntioned. But i am very confusing now!
There is no support of esp32s3 in the configure_ds.py.
What type of python script can i use? configure_ds.py or configure_esp_secure_cert.py?
Please teach me detail and correct procedure to solve this problem.
Thank you.

AdityaHPatwardhan, 2023-02-09:
Hi @Jinjungdong The `configure_ds.py` script supports the esp32s3 chip [(reference)](https://github.com/espressif/esp-idf/blob/dab3f38f0f966437c95e35f2c27e20d9a2a18fe7/examples/protocols/mqtt/ssl_ds/configure_ds.py#L39).

Please ensure that you have set the appropriate target for the project before executing this script.
`idf.py set-target esp32s3`

The steps that are required to followed in the exact sequence are listed [here](https://github.com/espressif/esp-idf/tree/release/v4.4/examples/protocols/mqtt/ssl_ds#configure-the-project)

Jinjungdong, 2023-02-09:
Hi, @AdityaHPatwardhan 

Actually, i added the code and then run script 'configure_ds.py'
But, below meessges are printed.

 PS D:\Project\AVR\VSC\esp_mqtt_ssl_ds_00> idf.py set-target esp32s3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Executing action: set-target
Set Target to: esp32s3, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory d:\project\avr\vsc\esp_mqtt_ssl_ds_00\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=1 d:\project\avr\vsc\esp_mqtt_ssl_ds_00"...
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.39.1.windows.1") 
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/Project/AVR/VSC/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: D:/Project/AVR/VSC/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: D:/Project/AVR/VSC/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
-- Project sdkconfig file D:/Project/AVR/VSC/esp_mqtt_ssl_ds_00/sdkconfig
Loading defaults file D:/Project/AVR/VSC/esp_mqtt_ssl_ds_00/sdkconfig.defaults...
-- Found Python3: D:/Project/AVR/VSC/.espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe (found version "3.8.7") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "esp_mqtt_ssl_ds_00" version: 1
-- Adding linker script D:/Project/AVR/VSC/esp_mqtt_ssl_ds_00/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script D:/Project/AVR/VSC/esp-idf/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script D:/Project/AVR/VSC/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script D:/Project/AVR/VSC/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script D:/Project/AVR/VSC/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script D:/Project/AVR/VSC/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script D:/Project/AVR/VSC/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script D:/Project/AVR/VSC/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib-time.ld
-- Adding linker script D:/Project/AVR/VSC/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32s3 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test ieee802154 jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl openthread partition_table perfmon protobuf-c protocol_examples_common protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: D:/Project/AVR/VSC/esp-idf/components/app_trace D:/Project/AVR/VSC/esp-idf/components/app_update D:/Project/AVR/VSC/esp-idf/components/asio D:/Project/AVR/VSC/esp-idf/components/bootloader D:/Project/AVR/VSC/esp-idf/components/bootloader_support D:/Project/AVR/VSC/esp-idf/components/bt D:/Project/AVR/VSC/esp-idf/components/cbor D:/Project/AVR/VSC/esp-idf/components/cmock D:/Project/AVR/VSC/esp-idf/components/coap D:/Project/AVR/VSC/esp-idf/components/console D:/Project/AVR/VSC/esp-idf/components/cxx D:/Project/AVR/VSC/esp-idf/components/driver D:/Project/AVR/VSC/esp-idf/components/efuse D:/Project/AVR/VSC/esp-idf/components/esp-tls D:/Project/AVR/VSC/esp-idf/components/esp32s3 D:/Project/AVR/VSC/esp-idf/components/esp_adc_cal D:/Project/AVR/VSC/esp-idf/components/esp_common D:/Project/AVR/VSC/esp-idf/components/esp_eth D:/Project/AVR/VSC/esp-idf/components/esp_event D:/Project/AVR/VSC/esp-idf/components/esp_gdbstub D:/Project/AVR/VSC/esp-idf/components/esp_hid D:/Project/AVR/VSC/esp-idf/components/esp_http_client D:/Project/AVR/VSC/esp-idf/components/esp_http_server D:/Project/AVR/VSC/esp-idf/components/esp_https_ota D:/Project/AVR/VSC/esp-idf/components/esp_https_server D:/Project/AVR/VSC/esp-idf/components/esp_hw_support D:/Project/AVR/VSC/esp-idf/components/esp_ipc D:/Project/AVR/VSC/esp-idf/components/esp_lcd D:/Project/AVR/VSC/esp-idf/components/esp_local_ctrl D: [... cut ...]
-- Build files have been written to: D:/Project/AVR/VSC/esp_mqtt_ssl_ds_00/build

PS D:\Project\AVR\VSC\esp_mqtt_ssl_ds_00> python .\configure_ds.py --port COM7 --private-key .\main\client.key
d:\Project\AVR\VSC\.espressif\tools\idf-python\3.8.7\python.exe: can't open file '$IDF_PATH/components/esptool_py/esptool/espefuse.py': [Errno 2] No such file or directory
PS D:\Project\AVR\VSC\esp_mqtt_ssl_ds_00> 

After above messages, i checked the path of espefuse.py. But path is correct.
I do not know why this message is printed.
Thank you.

AdityaHPatwardhan, 2023-02-09:
Have you ensured that the submodules are updated in the esp-idf directory?
I am not sure about windows, is there any additional permission requirement? is it necessary that the esp-idf should be present in the same Drive?

Jinjungdong, 2023-02-09:
Thank you. I will check it.
Have a good night!

AdityaHPatwardhan, 2023-02-13:
@Jinjungdong Was the issue resolved ?
Thanks,
Aditya

Jinjungdong, 2023-02-13:
Hi @AdityaHPatwardhan 
Maybe, i will try to check end of this week.
At that time, i will talk you for result.
Thank you for your concern!

Jinjungdong, 2023-02-15:
Hi @AdityaHPatwardhan

I have checked this problem as you mentioned. But, same problem is happened.
So i thought about the situration step deeply.
As you know, i made the esp_secure_cert.bin file using configure_esp_secure_cert.py with previous my question "mqtt/ssl_ds/configure_esp_secure_cert.py excution error in ESP32S3 (IDFGH-9116) #10517".
Thus, i think that your last comment is not suitable.
Just i think that this problem is involved in during updated procedure.
Please check again that detaily.
Thank you.

AdityaHPatwardhan, 2023-02-15:
Hi @Jinjungdong
Please dont use configure_esp_secure_cert.py as it is, it is not compatible with IDF v4.4.

As for this issue
```
'$IDF_PATH/components/esptool_py/esptool/espefuse.py': [Errno 2] No such file or directory
PS D:\Project\AVR\VSC\esp_mqtt_ssl_ds_00>
```
As I have mentioned earlier, please try to check if all the submodules are properly updated and are pointing to the correct SHA in the esp-idf.
I think the issue is related to your environment. I can see that the espefuse.py is still present in the [esptool_py component](https://github.com/espressif/esp-idf/tree/release/v4.4/components/esptool_py) in IDFv4.4. so the command should be able to execute.

If you want to use the `configure_esp_secure_cert.py` utility
then please use the ssl_ds [example in IDFv5.0](https://github.com/espressif/esp-idf/tree/release/v5.0/examples/protocols/mqtt/ssl_ds)

AdityaHPatwardhan, 2023-02-24:
Hi @Jinjungdong Any update?
Thanks

Jinjungdong, 2023-02-24:
Hi @AdityaHPatwardhan 
Actually i do not have been solving this problem.
So, i will skip its situration, and i will find another solution.
Thank you.

AdityaHPatwardhan, 2023-02-24:
@Jinjungdong I have tested the configure_ds.py script on v4.4 locally and it works as expected.
I think the issue that you are facing seems to be related to your setup.
Please use esp-idf v4.4 default setup and environment for executing the script.

Below is output that I obtained for configure_ds.py script:

```
python configure_ds.py --private-key ~/espressif/esp_secure_cert_mgr/tools/client.key -p /dev/cu.SLAB_USBtoUART --efuse_key_id 2

Provided key block (KEY BLOCK 2) is writable
 Generating a new key and burning it in the efuse..

espefuse.py v3.3.2
Connecting....

=== Run "burn_key" command ===
Burn keys to blocks:
 - BLOCK_KEY2 -> [b0 ac 74 4f 20 a4 da e5 99 db ea c1 8a cd 71 45 a8 ee 3e 63 ad dd 74 d9 5e 71 08 1a 1b 2d 01 2d]
	'KEY_PURPOSE_2': 'USER' -> 'HMAC_DOWN_DIGITAL_SIGNATURE'.
	Disabling write to 'KEY_PURPOSE_2'.
	Disabling write to key block

Keys will remain readable (due to --no-read-protect)

Check all blocks for burn...
idx, BLOCK_NAME,          Conclusion
[00] BLOCK0               is not empty
	(written ): 0x000000000000000000000000780000000000000101800300
	(to write): 0x000000000000000000000007000000000000000002000400
	(coding scheme = NONE)
[06] BLOCK_KEY2           is empty, will burn the new value
.
This is an irreversible operation!
Type 'BURN' (all capitals) to continue.
BURN
BURN BLOCK6  - OK (write block == read block)
BURN BLOCK0  - OK (all write block bits are set)
Reading updated efuses...
Successful
b0 ac 74 4f 20 a4 da e5 99 db ea c1 8a cd 71 45 a8 ee 3e 63 ad dd 74 d9 5e 71 08 1a 1b 2d 01 2d
Key was successfully written to the efuse (KEY BLOCK 2)

Creating NVS binary with version: V2 - Multipage Blob Support Enabled

Created NVS binary: ===> /Users/flying_raijin/esp/esp-idf/examples/protocols/mqtt/ssl_ds/esp_ds_data/pre_prov.bin
```

Jinjungdong, 2023-02-24:
Thank you. I will check it more detaily.
Have a nice weekend!

AdityaHPatwardhan, 2023-02-27:
Hi @Jinjungdong If possible, Can you please try doing a fresh installation of esp-idf v4.4
I think any setup-related issue may get resolved with that.
Thanks,
Aditya

AdityaHPatwardhan, 2023-03-13:
@Jinjungdong I am closing this issue as 
Please feel free to re-open if the above solutions are not able to fix it for you.
Thanks,
Aditya

closed 2023-03-13 by AdityaHPatwardhan