espressif/esp-idf #10620: a quick question about emmc supported capacity

FlynnMa, 2023-01-26:
I am evaluating the new esp32-s3, can I connect multiple emmc chips to one slot?
what is the maximum size of each and total?

thanks

igrr, 2023-01-26:
Hi @FlynnMa, we haven't supported multiple SD or eMMC devices on the same bus in the driver, so for now you can only connect one chip. There isn't a limit on the capacity, though — anything supported by SD 2.0 or eMMC 4.5 protocol should work. Typically you will meet the limit on the capacity of a particular filesystem first, such as in the case of FAT32.

If you have a use case which requires support for more than one eMMC on the same bus, please explain it. Then we can consider adding support for this.

FlynnMa, 2023-01-26:
Thanks @igrr 

So I understand the driver actually don't limit the capacity but the file system runs on it.

I had a quick search, just got that FAT32 support up to 4GB file size and 8TB maximum partition size?

Then that's beyond what I need.

igrr, 2023-01-26:
Yes, the driver doesn't limit the capacity, there is only the limit of SD protocol itself (which is 2TB, if I'm not mistaken).
You are correct about FAT32 FS and file size limits.

FlynnMa, 2023-01-26:
BTW, since esp32 s3 is very powerful, I am sure the use cases are going to increase fast.

If it is possible to drive multiple emmc, I think it will be a good thing to benefit design flexibility.😊

FlynnMa, 2023-01-26:
You mean SDXC? I thought eMMC is higher version than SDXC. Or does it also have 2TB limits?

igrr, 2023-01-26:
eMMC 5.1 indeed has support for 4kB block size, which allows addressing up to 16TB of storage. But that isn't supported in IDF, since we haven't seen use cases where ESP chips would need access to terabytes of storage.
We use the same 512 block size for SD and eMMC, which means that a 2TB region is addressable.

FlynnMa, 2023-01-26:
So I understand the limit is from software side?

Both multiple eMMC and 16TB, am I right?

igrr, 2023-01-26:
> So I the limit is from software side, both multiple eMMC and 16TB, am I right?

For the 4kB support, this is a software limitation. Regarding multiple eMMC, I haven't studied the HW design requirements so I'm not 100% sure. Maybe there is some hardware feature necessary for this as well.

> But if software hard coded the block size as 512bytes, what if I use a eMMC with 2k sized block?

The block size should be set up by the host, so i think everything will still work, just only 2TB would be available. (I haven't seen eMMC chips that large, so this is a mostly theoretical question for now.)

closed 2023-01-28 by FlynnMa