espressif/esp-idf #10621: ESP-IDF Partition error with binary file compiled from Arduino framework

SalmanMalik7183, 2023-01-26:
Hi,
I am working on a project where I need to flash the firmware file into the ESP-32 C3 devkit using Esp idf v5.0 but the signed binary file i am getting from Arduino framework seems to be creating some issue with the partition table. I am attaching some snaps of the issue.

P.S:  I am using secure boot v2 . 

`E (383) partition: load_partitions returned 0x105

assert failed: esp_ota_get_running_partition esp_ota_ops.c:553 (it != NULL)
Stack dump detected
Core  0 register dump:
MEPC    : 0x40381df6  RA      : 0x4038789c  SP      : 0x3fc9c6d0  GP      : 0x3fc8ee00  
0x40381df6: bootloader_flash_execute_command_common at C:/Espressif/frameworks/esp-idf-v5.0/components/bootloader_support/bootloader_flash/src/bootloader_flash.c:524

0x4038789c: systimer_hal_set_alarm_period at C:/Espressif/frameworks/esp-idf-v5.0/components/hal/systimer_hal.c:102

TP      : 0x3fc86c28  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x3fc9c84b  S1      : 0x0000007c  A0      : 0x3fc9c734  A1      : 0x3fc913a5
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fc98000
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fc9c728  S3      : 0x00000001
S4      : 0x3fc9c728  S5      : 0x42009716  S6      : 0x00000000  S7      : 0x00000000
0x42009716: esp_flash_app_enable_os_functions at C:/Espressif/frameworks/esp-idf-v5.0/components/spi_flash/spi_flash_os_func_app.c:310

S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000

Failed to run gdb_panic_server.py script: Command '['riscv32-esp-elf-gdb', '--batch', '-n', 'c:\\users\\mst-ws-ee-008\\esp\\esp-idf\\examples\\get-started\\blink\\build\\blink.elf', '-ex', 'target remote | "C:\\Espressif\\python_env\\idf5.0_py3.10_env\\Scripts\\python.exe" "C:\\Espressif\\frameworks\\esp-idf-v5.0\\tools\\idf_monitor_base\\..\\gdb_panic_server.py" --target esp32c3 "C:\\Users\\MST-WS~1\\AppData\\Local\\Temp\\tmp8uxrjcpj"', '-ex', 'bt']' returned non-zero exit status 1.
b"error starting child process 'C:\\Espressif\\python_env\\idf5.0_py3.10_env\\Scripts\\python.exe C:\\Espressif\\frameworks\\esp-idf-v5.0\\tools\\idf_monitor_base\\..\\gdb_panic_server.py --target esp32c3 C:\\Users\\MST-WS~1\\AppData\\Local\\Temp\\tmp8uxrjcpj': CreateProcess: No such file or directory\r\nNo stack.\r\n"

`

igrr, 2023-01-28:
The error "partition: No MD5 found in partition table" indicates that:
- The app is expecting the partition table to have an MD5 hash appended. (Arduino-esp32 project [is configured](https://github.com/espressif/arduino-esp32/blob/23d715af1b185fac21558c907af4ff5b86526aa4/tools/sdk/esp32/sdkconfig#L134) to produce apps which verify the partition table MD5 hash.)
- The partition table is built without an MD5 hash appended.

The solution should be either:
- Flash the device with the partition table and the bootloader built with CONFIG_PARTITION_TABLE_MD5=y.
- If you can't reflash the partition table and the bootloader (e.g. due to Secure Boot), then [recompile the Arduino framework](https://github.com/espressif/esp32-arduino-lib-builder) with CONFIG_PARTITION_TABLE_MD5=n, or [use it as an IDF component](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html).

SalmanMalik7183, 2023-01-30:
I have enabled the MD5 for the current flash. It still gives the issue of MD5 not present.
CONFIG_PARTITION_TABLE_MD5=y is set.
And yes I am using the secure boot and the boot was secured by a RSA based key. Now when I tried to implement Arduino as a IDF component I had to switch to an older version of IDF(v.4.4.3) because the Arduino as ESP-IDF component is not supported for IDF v 5.0. 
And in IDF v4.4.3 the secure key is generated using ECDSA scheme. Is there any way I can change my RSA secure key into the one that'll be used in IDF v4..4.3

Alvin1Zhang, 2025-03-06:
Thanks for reporting, will close, feel free to create a new ticket in [arduino-esp32 issue tracker](https://github.com/espressif/arduino-esp32/issues).

closed 2025-03-06 by Alvin1Zhang