espressif/esp-idf #10622: ESP32-S3 RGB LCD example does not run well, always restart.

hunasdf, 2023-01-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-WROOM-1

### Power Supply used.

External 5V

### What is the expected behavior?

Run the example code and see something on the display

### What is the actual behavior?

After i setup the rgb lcd example (https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/rgb_panel), the ESP always restart by the Task Watchdog.

### Steps to reproduce.

1. Install VSC
2. Download Espressif IDF extension
3. Install the latest (v5.0) ESP-IDF with the extension
4. Open rgb lcd example with the extension (ESP-IDF: Show examples)
5. Modify the pinouts and LCD specific settings
6. Build & flash & monitor

### Debug Logs.

```plain
I (29) boot: ESP-IDF v5.0 2nd stage bootloader
I (29) boot: compile time 10:35:07
I (29) boot: chip revision: v0.1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (38) boot.esp32s3: Boot SPI Speed : 80MHz
I (43) boot.esp32s3: SPI Mode       : DIO
I (47) boot.esp32s3: SPI Flash Size : 2MB
W (52) boot.esp32s3: PRO CPU has been reset by WDT.
W (58) boot.esp32s3: APP CPU has been reset by WDT.
I (63) boot: Enabling RNG early entropy source...
I (69) boot: Partition Table:
I (72) boot: ## Label            Usage          Type ST Offset   Length
I (79) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (87) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (94) boot:  2 factory          factory app      00 00 00010000 00100000
I (102) boot: End of partition table
I (106) boot_comm: chip revision: 1, min. application chip revision: 0
I (113) esp_image: segment 0: paddr=00010020 vaddr=3c050020 size=0ea9ch ( 60060) map
I (133) esp_image: segment 1: paddr=0001eac4 vaddr=3fc93300 size=01554h (  5460) load
I (134) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=4625ch (287324) map
I (190) esp_image: segment 3: paddr=00066284 vaddr=3fc94854 size=01f88h (  8072) load
I (192) esp_image: segment 4: paddr=00068214 vaddr=40374000 size=0f224h ( 61988) load
I (210) esp_image: segment 5: paddr=00077440 vaddr=50000000 size=00010h (    16) load
I (216) boot: Loaded app from partition at offset 0x10000
I (217) boot: Disabling RNG early entropy source...
I (230) octal_psram: vendor id    : 0x0d (AP)
I (230) octal_psram: dev id       : 0x02 (generation 3)
I (230) octal_psram: density      : 0x03 (64 Mbit)
I (235) octal_psram: good-die     : 0x01 (Pass)
I (240) octal_psram: Latency      : 0x01 (Fixed)
I (246) octal_psram: VCC          : 0x01 (3V)
I (251) octal_psram: SRF          : 0x01 (Fast Refresh)
I (257) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (262) octal_psram: BurstLen     : 0x01 (32 Byte)
I (268) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (274) octal_psram: DriveStrength: 0x00 (1/1)
W (279) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
I (290) esp_psram: Found 8MB PSRAM device
I (294) esp_psram: Speed: 80MHz
I (331) mmu_psram: Instructions copied and mapped to SPIRAM
I (338) mmu_psram: Read only data copied and mapped to SPIRAM
I (339) cpu_start: Pro cpu up.
I (339) cpu_start: Starting app cpu, entry point is 0x403753d4
0x403753d4: call_start_cpu1 at C:/Users/Felhaszlo/esp/esp-idf/components/esp_system/port/cpu_start.c:142

I (0) cpu_start: App cpu up.
I (769) esp_psram: SPI SRAM memory test OK
I (777) cpu_start: Pro cpu start user code
I (777) cpu_start: cpu freq: 160000000 Hz
I (777) cpu_start: Application information:
I (780) cpu_start: Project name:     rgb_panel
I (785) cpu_start: App version:      1
I (790) cpu_start: Compile time:     Jan 26 2023 10:57:19
I (796) cpu_start: ELF file SHA256:  bbc521bd54dd7282...
I (802) cpu_start: ESP-IDF:          v5.0
I (807) heap_init: Initializing. RAM available for dynamic allocation:
I (814) heap_init: At 3FC97750 len 00051FC0 (327 KiB): D/IRAM
I (820) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (827) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (833) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (840) esp_psram: Adding pool of 7808K of PSRAM memory to heap allocator
I (847) spi_flash: detected chip: gd
I (851) spi_flash: flash io: dio
W (855) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (869) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (889) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (889) example: Turn off LCD backlight
I (899) gpio: GPIO[6]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (909) example: Install RGB LCD panel driver
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375961
0x40375961: panicHandler at C:/Users/Felhaszlo/esp/esp-idf/components/esp_system/port/panic_handler.c:217

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x167c
load:0x403c9700,len:0xbe8
load:0x403cc700,len:0x2e98
entry 0x403c9904
```

### More Information.

Tried on Windows 10 and Linux (Ubuntu20.04) too, same issue

suda-morris, 2023-01-28:
> Modify the pinouts and LCD specific settings

Please post your pin connection

YoonTaiJun, 2023-02-02:
I had the same problem.
Are you using LVGL?
Proceed without using style
Style is duplicated and restarted.

JiunnYang, 2023-12-17:
I also meet same problem:
message as the below:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x4200345f
0x4200345f: panic_handler at C:/ESP32/esp-idf/components/esp_system/port/panic_handler.c:145 (discriminator 3)

and usage same as @YoonTaiJun, it always happen when I lv_obj_add_style(...)