espressif/esp-idf #10626: Intermitent crash during IDF initialisation

KaeLL, 2023-01-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

9ee3c8337d3c4f7914f62527e7f7c78d7167be95

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

External 5V

### What is the expected behavior?

To not crash, especially before getting to my code.

### What is the actual behavior?

It crashes

### Steps to reproduce.

<details>
<summary>defconfig</summary>

```
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF=y
CONFIG_BOOTLOADER_LOG_LEVEL_NONE=y
CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE=y
CONFIG_BOOTLOADER_SKIP_VALIDATE_ALWAYS=y
CONFIG_ESPTOOLPY_FLASHMODE_QIO=y
CONFIG_ESPTOOLPY_FLASHFREQ_80M=y
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
# CONFIG_ESPTOOLPY_FLASHSIZE_DETECT is not set
CONFIG_ESPTOOLPY_AFTER_NORESET=y
CONFIG_ESPTOOLPY_MONITOR_BAUD_CONSOLE=y
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions_ota.csv"
CONFIG_PARTITION_TABLE_OFFSET=0xC000
CONFIG_COMPILER_WARN_WRITE_STRINGS=y
CONFIG_BT_ENABLED=y
CONFIG_BTDM_CTRL_BLE_MAX_CONN=1
# CONFIG_BTDM_CTRL_MODEM_SLEEP is not set
# CONFIG_BTDM_BLE_SCAN_DUPL is not set
# CONFIG_BTDM_BLE_ADV_REPORT_FLOW_CTRL_SUPP is not set
# CONFIG_BT_BLE_SMP_ENABLE is not set
CONFIG_BT_STACK_NO_LOG=y
CONFIG_BT_ACL_CONNECTIONS=1
# CONFIG_BT_MULTI_CONNECTION_ENBALE is not set
CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY=y
CONFIG_BT_BLE_HOST_QUEUE_CONG_CHECK=y
CONFIG_SPI_MASTER_IN_IRAM=y
[... 180 lines cut ...]
0x40093765: prvCopyDataToQueue at /esp-idf/components/freertos/queue.c:2245

0x40093ce8: xQueueGenericSend at /esp-idf/components/freertos/queue.c:854

0x40099d96: esp_log_impl_unlock at /esp-idf/components/log/log_freertos.c:56

0x401ce778: s_log_level_get_and_unlock at /esp-idf/components/log/log.c:162

0x401ce8ce: esp_log_writev at /esp-idf/components/log/log.c:195

0x40099cf5: esp_log_write at /esp-idf/components/log/log.c:210

0x400d806e: initialization at project/main/main.c:161

0x400d836b: app_main at project/main/main.c:359

0x401ce5d7: main_task at /esp-idf/components/freertos/port/port_common.c:141 (discriminator 2)
```

</details>
```

### More Information.

Running on a vanilla ESP32

igrr, 2023-01-27:
> To not crash, especially before getting to my code.

> 0x40099cf5: esp_log_write at /esp-idf/components/log/log.c:210
> 0x400d806e: initialization at project/main/main.c:161
> 0x400d836b: app_main at project/main/main.c:359
> 0x401ce5d7: main_task at /esp-idf/components/freertos/port/port_common.c:141 (discriminator 2)

This might totally be a bug in IDF, but from the backtrace it does seem like it gets to `initalization` in `main.c:161`. Do you think you could try to narrow down, what particular steps during the initialization correlate with the crash occurring?

KaeLL, 2023-01-27:
[image]

KaeLL, 2023-01-27:
[image]

ESP-Marius, 2023-02-02:
@KaeLL did the above mentioned PR fix your issue?

KaeLL, 2023-02-02:
Given the intermittent and trivial nature of the issue, it's too soon to say. Meanwhile, some context like the why and what regarding this issue would be appreciated and educative.

closed 2023-02-08 by espressif-bot

ESP-Marius, 2023-02-09:
@KaeLL at the moment we are not sure why the errors are occurring, but the PR do fix something in our newlib component that was inconsistent with the newlib documentation. The feedback so far is that it seems to fix these priority/freertos assert that several users saw.

We have an internal ticket to investigate this deeper, and if we discover something I'll make sure we report back in this issue.