espressif/esp-idf #10628: Unable to upgrade ESP-IDF v4.4.3 to latest OpenOCD v0.11.0-esp32-20221026 tool

JBHunterI, 2023-01-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

Trying to resolve JTAG debug problems with blink example code from ESP-IDF v4.4.3 (I can't upgrade to IDF v5.0 because one of the libraries my code uses doesn't support the latter's API changes) on ESP32-C3 DevKitM-1. Per the [openocd-esp32](https://github.com/espressif/openocd-esp32) Releases page, [v0.11.0-esp32-20221026](https://github.com/espressif/openocd-esp32/releases/tag/v0.11.0-esp32-20221026) seems to fix some, and my Windows10 installation of IDF v4.4.3 only offers [v0.11.0-esp32-20220706](https://github.com/espressif/openocd-esp32/releases/tag/v0.11.0-esp32-20220706).  According to the [idf_tools.py script](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-tools.html#idf-tools-py-script) section of the ESP-IDF Programming Guide » API Guides » Tools » [Downloadable Tools](https://docs.espressif.com/projects/esp-idf/en/v4.4.3/esp32c3/api-guides/tools/idf-tools.html#downloadable-tools) page, it appears idf_tools.py is the most flexible way to do that, whether idf_tools.py is used the CMD shell or the Eclipse Plugin's Tools Manager Wizard. I would expect this tool to show both the recommended as wells the available versions, as I've seen it do before for cmake. 

C:\App\ESP\IDF\esp-idf-v4.4.3\tools> python idf_tools.py list
 ...
 openocd-esp32: OpenOCD for ESP32
   v0.11.0-esp32-20220706 (recommended, installed)
...

C:\App\ESP\IDF> python esp-idf-v4.4.3\tools\idf_tools.py install openocd-esp32@v0.11.0-esp32-20221026
<whatever a successful installation would should show>

### What is the actual behavior?

C:\App\ESP\IDF> python esp-idf-v4.4.3\tools\idf_tools.py install openocd-esp32@v0.11.0-esp32-20221026
Current system platform: win64
**ERROR: unknown version for tool openocd-esp32: v0.11.0-esp32-20221026**

### Steps to reproduce.

See "What is the actual behavior?" section above.

### More Information.

I don't see any restriction on the platform (win64, in my case), but when I look at %IDF_PATH%\tools\tools.json line 530, it seems to indicate that it's only for the Linux platform:
     ...
    {
      "description": "OpenOCD for ESP32",
      ...
      "info_url": "https://github.com/espressif/openocd-esp32",
      "install": **"always"**,
      "license": "GPL-2.0-only",
      "name": "openocd-esp32",
      "platform_overrides": [
        {
          "install": **"on_request"**,
          **"platforms": [
            "linux-i686"
          ]**
        }
      ],
      "supported_targets": [
        **"all"**
      ],
      ...

There's nothing on the [openocd-esp32 Releases page](https://github.com/espressif/openocd-esp32/releases) saying this version is only for 32-bit Linux & not for Windows.

I tried downloading [openocd-esp32-win32-0.11.0-esp32-20221026.zip](https://github.com/espressif/openocd-esp32/releases/download/v0.11.0-esp32-20221026/openocd-esp32-win32-0.11.0-esp32-20221026.zip) into my %IDF_TOOLS_PATH%\dist folder, and even extracting it to a new %IDF_TOOLS_PATH%\tools\openocd-esp32\v0.11.0-esp32-20221026 but that doesn't seem to make the new version appear in my PATH. There's gotta be a better way to get idf_tools.py to manage this tool's version than manually maintaining the right path in my PATH environment variable. I just can't find a description to do that anywhere.

dobairoland, 2023-01-27:
Hi @JBHunterI. The version you are looking for will be available in the v4.4.4 bugfix release. This will be available soon - regression tests have already passed. 

The version is already available in the release branch: https://github.com/espressif/esp-idf/blob/release/v4.4/tools/tools.json#L733

BTW, `idf_tools.py` does not support of installing an arbitrary version. For each release there are compatible versions added there which pass all of our tests.

JBHunterI, 2023-01-27:
I tried  saving that page to a text file then cutting & pasting lines 723 ("name": "v0.11.0-esp32-20221026")to 733 ("url": "https://.../openocd-esp32-win32-0.11.0-esp32-20221026.zip") into my tools.json. I was then able to install it:
C:\App\ESP\IDF
> python esp-idf-v4.4.3\tools\idf_tools.py install openocd-esp32
Current system platform: win64
Installing openocd-esp32@v0.11.0-esp32-20221026
file openocd-esp32-win32-0.11.0-esp32-20221026.zip is already downloaded
Extracting C:\App\ESP\IDF\dist\openocd-esp32-win32-0.11.0-esp32-20221026.zip to C:\App\ESP\IDF\tools\openocd-esp32\v0.11.0-esp32-20221026

C:\App\ESP\IDF
> python esp-idf-v4.4.3\tools\idf_tools.py check openocd-esp32
usage: idf_tools.py [-h] [--quiet] [--non-interactive] [--tools-json TOOLS_JSON] [--idf-path IDF_PATH] {list,check,export,install,download,install-python-env} ...
idf_tools.py: error: unrecognized arguments: openocd-esp32

C:\App\ESP\IDF> python esp-idf-v4.4.3\tools\idf_tools.py check
Checking for installed tools...
...
Checking tool openocd-esp32
    version found in PATH: v0.11.0-esp32-20220706
    version installed in tools directory: v0.11.0-esp32-20221026
...

Then opened a new CMD shell to make sure my PATH was update with ...;C:\App\ESP\IDF\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin;...  but now I can't even run the blink example:

C:\...\test\test_PC-WiFi_ESP\blink
> idf.py flash monitor
Executing action: flash
Running ninja in directory c:\...\test\blink\build
Executing "ninja flash"...
[1/5] cmd.exe /C "cd /D C:\...\test\test_PC-WiFi_ESP\blink\build\esp-idf\esptool_py && python C:/.../test/blink/build/partition_table/partition-table.bin C:/.../test/blink/build/blink.bin"
blink.bin binary size 0x22dd0 bytes. Smallest app partition is 0x100000 bytes. 0xdd230 bytes (86%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\...\test\blink\build\bootloader\esp-idf\esptool_py && python C:/App/ESP/IDF/esp-idf-v4.4.3/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 C:/.../test/blink/build/bootloader/bootloader.bin"
Bootloader binary size 0x4d30 bytes. 0x32d0 bytes (40%) free.
[2/3] cmd.exe /C "cd /D C:\App\ESP\IDF\esp-idf-v4.4.3\components\esptool_py && C:\App\ESP\IDF\tools\cmake\3.23.1\bin.../test/blink/build" -P C:/App/ESP/IDF/esp-idf-v4.4.3/components/esptool_py/run_serial_tool.cmake"
esptool.py esp32c3 -p COM8 -b 115200 --before=default_reset --after=hard_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 blink.bin 0x8000 partition_table/partition-table.bin
esptool.py v3.3.2
Serial port COM8
Connecting...
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 58:cf:79:e3:64:c8
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00004fff...
Flash will be erased from 0x00010000 to 0x00032fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Erasing flash...
Took 0.26s to erase flash block
...
Writing at 0x00004c00... (100 %)
Wrote 20480 bytes at 0x00000000 in 0.4 seconds (453.9 kbit/s)...
Hash of data verified.
Erasing flash...
Took 1.19s to erase flash block
Writing at 0x00010000... (0 %)
Writing at 0x00010400... (1 %)
Writing at 0x00010800... (2 %)
...
Writing at 0x00032c00... (100 %)
Wrote 143360 bytes at 0x00010000 in 2.5 seconds (455.1 kbit/s)...
Hash of data verified.
Erasing flash...
Took 0.05s to erase flash block
Writing at 0x00008000... (33 %)
Writing at 0x00008400... (66 %)
Writing at 0x00008800... (100 %)
Wrote 3072 bytes at 0x00008000 in 0.0 seconds (513.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Executing action: monitor
Running idf_monitor in directory c:\...\blink
Executing "C:\App\ESP\IDF\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\App\ESP\IDF\esp-idf-v4.4.3\tools/idf_monitor.py -p COM8 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32c3 --decode-panic backtrace c:\...\test\blink\build\blink.elf -m 'C:\App\ESP\IDF\python_env\idf4.4_py3.8_env\Scripts\python.exe' 'C:\App\ESP\IDF\esp-idf-v4.4.3\tools\idf.py'"...
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM8 instead...
--- idf_monitor on \\.\COM8 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x4 (DOWNLOAD(USB/UART0/1))
Saved PC:0x400462e0
waiting for download
                   <------------ stuck here

Guess I have to wait for the IDF v4.4.4 release, then?

dobairoland, 2023-01-30:
The consequent issue you are facing now is something entirely different. This was not happening before?

Could you try to reset the chip manually after flashing? Or if you are at that point of "waiting for download", try to reset the chip by CTRL+T, CTRL+R?

JBHunterI, 2023-01-30:
That was simply my attempt to see if adding OpenOCD 20221026 to IDF 4.4.3 would help, but that doesn't seem to work at all even with the blink example.

FYI, I went back to the original openocd version in IDF 4.4.3 and without even involving the Eclipse Plugin, i.e. simlpy with the IDF 4.4.3 CMD shell, I get this:
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
adapter speed: 40000 kHz

Warn : Interface already configured, ignoring
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
**Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED**
Info : esp_usb_jtag: serial (58:CF:79:E3:64:C8)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : starting gdb server for esp32c3 on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Memory protection is enabled. Reset target to disable it...
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Reset cause (3) - (Software core reset)
Warn : No symbols for FreeRTOS!
Info : [esp32c3] Found 8 triggers
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 28 KB
Info : Flash mapping 1: 0x20020 -> 0x3c0e0020, 152 KB
Info : Auto-detected flash bank 'esp32c3.flash' size 4096 KB
Info : Using flash bank 'esp32c3.flash' size 4096 KB
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 28 KB
Info : Flash mapping 1: 0x20020 -> 0x3c0e0020, 152 KB
Info : Using flash bank 'esp32c3.irom' size 0 KB
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 28 KB
Info : Flash mapping 1: 0x20020 -> 0x3c0e0020, 152 KB
Info : Using flash bank 'esp32c3.drom' size 156 KB
Warn : negative reply, retrying
Warn : negative reply, retrying
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Reset cause (3) - (Software core reset)
Info : dropped 'gdb' connection
Info : accepting 'gdb' connection on tcp/3333
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Reset cause (3) - (Software core reset)
Error: Duplicate Breakpoint address: 0x420064d4 (BP 3)
Error: Duplicate Breakpoint address: 0x420064d4 (BP 3)
Error: Duplicate Breakpoint address: 0x420064d4 (BP 3)
Error: Duplicate Breakpoint address: 0x420064d4 (BP 3)
Error: Duplicate Breakpoint address: 0x420064d4 (BP 3)
Error: Duplicate Breakpoint address: 0x420064d4 (BP 3)
Error: Duplicate Breakpoint address: 0x420064d4 (BP 3)
...

which matches GDB in another window, saying:
C:\...\test\myApp> riscv32-esp-elf-gdb.exe -x gdbinit build\myApp.elf
**C:\App\ESP\IDF\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin\riscv32-esp-elf-gdb.exe: warning: Couldn't determine a path for the index cache directory.**   <-- ???
GNU gdb (crosstool-NG esp-2021r2-patch5) 9.2.90.20200913-git
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
[... 42 lines cut ...]
...
Duplicate Breakpoint address: 0x420064d4 (BP 3)
Hart 0 unexpectedly reset!
Timed out after 5s waiting for busy to go low (abstractcs=0x10001002). Increase the timeout with riscv set_command_timeout_sec.
Abstract command ended in error 'busy' (abstractcs=0x10001102)
Timed out after 5s waiting for busy to go low (abstractcs=0x10001102). Increase the timeout with riscv set_command_timeout_sec.
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
...

at which point my terminal shows:

Setting breakpoint at 0x420064d4 and returning...
Guru Meditation Error: Core  0 panic'ed (Cache error).
ibus tried to write cache

Setting breakpoint at 0x420064d4 and returning...
Guru Meditation Error: Core  0 panic'ed (Cache error).
ibus tried to write cache

Setting breakpoint at 0x420064d4 and returning...
Guru Meditation Error: Core  0 panic'ed (Cache error).
ibus tried to write cache
...

dobairoland, 2023-01-31:
Oh, this is another unrelated issue. Could you please try the Windows tool installer? It is pretty handy in setting up a working environment. It is available here: https://dl.espressif.com/dl/esp-idf/

Setting up an environment manually and solving all the possible issues could take a lot of time.

JBHunterI, 2023-01-31:
The Universal Online Installer v2.18 for Windows **_is_** the way I installed this IDF v4.4.3 & I _**always used prior versions of it**_ (way back to v2.16) to install all versions of IDF.  Actually, I did try using the Eclipse Plugin to install IDF in the hardcoded %USERPROFILE%\\.esspresif folder, but stopped when I had all sorts of problems with that Eclipse Plugin. Now, it's all CMD shell, at least until I can get a reasonably stable development environment for more than just the blink example. 

The only reason I tried swapping in OpenOCD 20221026 into IDF 4.4.3 was to see if it might resolved my issue without having to way for it in IDF v4.4.4. When I saw it didn't, I reverted the latter back to using its original OpenOCD 20220706, which as you say resulted in "another unrelated issue". 

So, is IDF v4.4.4 any closer to becoming available?  Will this "other unrelated issue" also be fixed in the upcoming IDF v4.4.4?

dobairoland, 2023-02-01:
Ok, please wait for v4.4.4. The release should happen any day now. And if you still encounter these issues then please report it separately so each could be handled by the right team/person.

closed 2023-02-01 by dobairoland

JBHunterI, 2023-02-01:
I noticed IDF v5.0 uses OpenOCD v0.11.0-esp32-20221026. I'll try to see if building my project in IDF 4.4.3 (because of its incompatibility with IDF 5.0) & copying it to an IDF v5.0 project to debug it works.

dobairoland, 2023-02-03:
Installer with v4.4.4 support is available now at https://dl.espressif.com/dl/esp-idf/

JBHunterI, 2023-02-03:
Thanks, this new release v4.4.4 fixed my debug issues.  Thanks for your help. Feel free to close this issue.

FYI though, I did get this incorrect QR-code when I rebuilt my project straight out of the CMD window that open at the end of the Universal Online Installer v2.18 installation:

...
I (470) wifi_prov_mgr: Provisioning started with service name : PROV_e364c8
I (470) app_wifi: Provisioning started
I (470) app_wifi: Scan this QR code from the phone app for Provisioning.

  ��������������������� ������ ������ ��������������������������������� ���������������������
  ��� ��������� ��� ������������ ��� ������������ ������ ������ ��� ��������� ���
  ��� ��������� ��� ��� ��� ������ ��������� ������������������ ��� ��������� ���
  ��������������������� ��� ��� ��� ��� ��� ��� ��������� ��� ���������������������
  ��������������� ��������������� ��������� ������������������������������������ ������������
  ��� ������  ������������������������ ���  ������������������ ���������  ������
  ��� ��� ������������ ��������������� ������������������ ������ ������������ ���������
  ��������������������� ��� ��� ������ ��������������� ��� ��������� ������ ���������
  ��� ��������������������������� ������������������������ ���    ������������ ������
  ��������������������� ������ ��������� ������ ������ ��� ���  ��������� ���������
  ��������� ��������� ��������������������� ���������������  ������������������ ��� ���
  ��� ��� ���������  ������ ��� ������������������ ������������ ���������������������
  ���  ���  ��� ��������� ������������������������ ��� ������������������������������
  ��������������������� ��� ��������� ������  ���  ��� ������ ��� ��� ������
  ��� ��������� ��� ������ ��������� ������������������������ ������������������������
  ��� ��������� ��� ���������  ��� ��������������������� ������ ������������������
  ��������������������� ���  ��� ��������������� ���  ��������� ������   ���

I (530) app_wifi: If QR code is not visible, copy paste the below URL in a browser.
https://espressif.github.io/esp-jumpstart/qrcode.html?data={"ver":"v1","name":"PROV_e364c8","pop":"79e364c8","transport":"ble"}
I (530) app_wifi: Provisioning Started. Name : PROV_e364c8, POP : 79e364c8
...

Or, if you prefer:
[image]

I had to open a new CMD window to get a usable QR-code. Just FYI.

JBHunterI, 2023-03-01:
Looks like any other app less trivial than the blink example still has the same problem, i.e. the app runs right past both the automatic temporary breakpoint at the beginning of app_main() and the breakpoint I added at the first statement in my infinite loop. Even with IDF 4.4.4, the debugger stops if I hit the Pause tool as 'idy.py monitor' shows it's looping infinitely, but if I try to allow it to run again with the Play tool, I just get this on the 'idf.py monitor':

...
I (22637) coreMQTT: Ack packet deserialized with result: MQTTSuccess.
I (22637) coreMQTT: State record updated. New state=MQTTPublishDone.
I (22637) temp_sub_pub_and_led_control_demo: Rx'ed ack for QoS1 publish from Tx to /filter/TempSubPubLED (P5:F0).
I (22657) coreMQTT: De-serialized incoming PUBLISH packet: DeserializerResult=MQTTSuccess.
I (22657) coreMQTT: State record updated. New state=MQTTPubAckSend.
I (22657) temp_sub_pub_and_led_control_demo: Received incoming publish message {"temperatureSensor":{ "taskName": "TempSubPubLED", "temperatureValue": 30.887400, "iteration": 4}}
                             <--- hit Pause then Play a second later here
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

Setting breakpoint at 0x4204825e and returning...
0x4204825e: esp_vApplicationIdleHook at C:/App/ESP/IDF/esp-idf-v4.4.4/components/esp_system/freertos_hooks.c:50 (discriminator 2)

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

Setting breakpoint at 0x4204825e and returning...
0x4204825e: esp_vApplicationIdleHook at C:/App/ESP/IDF/esp-idf-v4.4.4/components/esp_system/freertos_hooks.c:50 (discriminator 2)

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).
...

Anyone else see this?  Are there any other updates I need to apply to IDF v4.4.4?