espressif/esp-idf #10629: ESP HTTP Client: disable_auto_redirect causes infinite loop

jgillick, 2023-01-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

XIAO_ESP32C3

### Power Supply used.

USB

### What is the expected behavior?

When using `.disable_auto_redirect = true`, I'd expect the HTTP request to end when it get's to a redirect. At that point, the firmware can manually get the `location` header and decide what to do with the redirect or handle it in the event handler.

```c
esp_http_client_config_t config = {
    .url = "http://httpbin.org/absolute-redirect/3",
    .event_handler = _http_event_handler,
    .disable_auto_redirect = true,
};
```

### What is the actual behavior?

The HTTP client goes into an infinite loop until `redirect_counter` reaches the maximum count.

This is caused by the following [lines of code](https://github.com/espressif/esp-idf/blob/release/v5.0/components/esp_http_client/esp_http_client.c#L847-L855):

```c
if (client->disable_auto_redirect) {
    http_dispatch_event(client, HTTP_EVENT_REDIRECT, NULL, 0);
} else {
    ESP_ERROR_CHECK(esp_http_client_set_redirection(client));
}
client->redirect_counter ++;
client->process_again = 1;
```

The block of code prevents the redirect from happening, but then sets `process_again` to `1`, which causes the response to be processed again.

I believe the solution should be to put the `redirect_counter` and `process_again` assignments in the `else` statement.

```c
if (client->disable_auto_redirect) {
    ESP_LOGI(TAG, "REDIRECT!");
    http_dispatch_event(client, HTTP_EVENT_REDIRECT, NULL, 0);
} else {
    ESP_ERROR_CHECK(esp_http_client_set_redirection(client));
    client->redirect_counter ++;
    client->process_again = 1;
}
```

### Steps to reproduce.

1. Create a simple HTTP client that points to a URL with a redirect and set `disable_auto_redirect` to `false`.
2. Do NOT handle the redirect in the event handler
3. Run the program, and observe that the request hits the maximum redirects error and fails.

AxelLin, 2023-02-14:
master: 5feb4696be0f3987eaeba20a170645f8a8895930
v5.0: 9463a66fc0e28dd3710e5bd0e43121d1fd611f99

Harshal5, 2023-02-16:
Hello @jgillick ,
Sorry for the delayed response. 

> The HTTP client goes into an infinite loop until redirect_counter reaches the maximum count.

Thank you for reporting this issue. Rather than putting the `redirect_counter` and `process_again` assignments in the else statement, we should add them in the `esp_http_client_set_redirection` function, because it will also handle the case when  `disable_auto_redirect = true` and the user calls `esp_http_client_set_redirection` in the event handler.

Attached is the patch for same. Could you please check if it satisfies your requirement?
[0001-http_client-fixed-looping-caused-when-disable_auto_r.patch](https://github.com/espressif/esp-idf/files/10754404/0001-http_client-fixed-looping-caused-when-disable_auto_r.patch)

jgillick, 2023-02-18:
@Harshal5 I haven't had a chance to try this patch directly, but the change makes sense and looks good.

closed 2023-02-27 by espressif-bot