espressif/esp-idf #10631: [ESP-IDF] Masked tasks priorities

KonssnoK, 2023-01-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

In our development cycle, we perform scheduling planning based on assigned tasks priorities.

Currently IDF is spawning tasks that have unknown priorities and name.
The only way to obtain it is via runtime calls.

Examples:
[image]
Each red ? means that task has unknown priority.

Could you provide a table of priorities and spawned tasks for each component?
(we are interested in BT, wifi, LWIP + TCPIP)

KonssnoK, 2023-01-27:
by looking in BT i could find
```
BTC -> BTC_TASK, 19
BTU -> BTU_TASK, 20
HCI -> hciT, 22
```
in esp_task.h we have
```
TCPIP -> 18
MAIN -> 1
TIMER -> 22
EVENT -> 20
```

xueyunfei998, 2023-01-31:
hi @KonssnoK 
You can use this function to print out the priority and stack of each task.
`void vTaskList( char * pcWriteBuffer )`

KonssnoK, 2023-01-31:
@xueyunfei998 
i was looking for some documentation about threads prioritization.
I guess there's none, so we need to take the info from runtime.
Thanks.

closed 2023-02-03 by KonssnoK