espressif/esp-idf #10633: esp32-s3 jump to bootloader from tinyusb device

korneliuszo, 2023-01-27:
I try to do SW reset from tinyusb code to bootloader for development cycle.

There is:
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-tinyusb.c
which looks like it's doing things right, but after small modifications to build under IDF v5.0 it doesn't work.

korneliuszo, 2023-01-29:
https://gist.github.com/korneliuszo/90b4c935955a44c989fee77f21d5d08d - arduino part doesn't work either

korneliuszo, 2023-01-29:
I have:
`DIS_USB_OTG_DOWNLOAD_MODE (BLOCK0)                 Disables USB-OTG download feature in UART download = True R/W (0b1)
                                                    boot mode  `
Is this cause of my problem?

igrr, 2023-01-30:
If this eFuse is burned, indeed entering USB download mode is not possible.

korneliuszo, 2023-01-31:
checked DFU mode don't work either.

closed 2023-01-31 by korneliuszo