espressif/esp-idf #10634: Allow bootloader to use original ESP-IDF components instead of mocked components

robbawebba, 2023-01-27:
### Is your feature request related to a problem?

I have mocked the `efuse` component in my unit test project using CMock, and most features are working well for my application. However, The bootloader that is built for my unit test project also uses the mocked `efuse` component, resulting in some undesirable behavior. On boot, the bootloader tries to access some efuse block using the mocked function, and the bootloader seems to restart shortly after the mocked function is called (this could possibly be due to some Unity test behavior built into the mocked function). On the second boot attempt, the issue does not happen again and the bootloader is able to run my application.

Here are the logs for this scenario:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x164c
load:0x403b6000,len:0xb7c
load:0x403ba000,len:0x2f74
entry 0x403b6248
I (25) boot: ESP-IDF v4.4.1 2nd stage bootloader
I (25) boot: compile time 17:34:53
I (25) boot: chip revision: 0
I (26) boot.esp32s3: Boot SPI Speed : 80MHz
I (31) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 8MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00113000
I (79) boot:  3 user_data        Unknown data     01 82 00123000 006d7000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0ff0ch ( 65292) map
I (111) esp_image: segment 1: paddr=0001ff34 vaddr=3fc91fd0 size=000e4h (   228) load
I (112) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=2d5d8h (185816) map
I (150) esp_image: segment 3: paddr=0004d600 vaddr=3fc920b4 size=02e7ch ( 11900) load
I (153) esp_image: segment 4: paddr=00050484 vaddr=40374000 size=0dfc8h ( 57288) load
I (169) esp_image: segment 5: paddr=0005e454 vaddr=50000000 size=00010h (    16) load
I (174) boot: Loaded app from partition at offset 0x10000
I (175) boot: Disabling RNG early entropy source...
I (189) opi psram: vendor id : 0x0d (AP)
I (189) opi psram: dev id    : 0x02 (generation 3)
I (189) opi psram: density   : 0x03 (64 Mbit)
I (193) opi psram: good-die  : 0x01 (Pass)
I (198) opi psram: Latency   : 0x01 (Fixed)
I (202) opi psram: VCC       : 0x01 (3V)
I (207) opi psram: SRF       : 0x01 (Fast Refresh)
I (212) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (218) opi psram: BurstLen  : 0x01 (32 Byte)
I (223) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (229) opi psram: DriveStrength: 0x00 (1/1)
I (234) spiram: Found 64MBit SPI RAM device
I (239) spiram: SPI RAM mode: sram 40m
I (243) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (250) cpu_start: Pro cpu up.
I (254) cpu_start: Starting app cpu, entry point is 0x403752d8
0x403752d8: call_start_cpu1 at /home/rob/esp/esp-idf/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
I (991) spiram: SPI SRAM memory test OK
:0::FAIL:Function esp_efuse_read_field_blob.  Called more times than expected.Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4202b962  PS      : 0x00060f30  A0      : 0x00000000  A1      : 0x3fceb210
0x4202b962: __wrap_longjmp at /home/rob/esp/esp-idf/components/esp_rom/patches/esp_rom_longjmp.S:135

A2      : 0x3fc9f138  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x40035880  A7      : 0xfffffff0  A8      : 0x820039f0  A9      : 0x3fceb1e0
A10     : 0x6000006c  A11     : 0x00000020  A12     : 0x00000000  A13     : 0x00970000
A14     : 0x3fcef3d4  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x0000001d
EXCVADDR: 0xfffffff0  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff

Backtrace:0x4202b95f:0x3fceb210
0x4202b95f: __wrap_longjmp at /home/rob/esp/esp-idf/components/esp_rom/patches/esp_rom_longjmp.S:133

ELF file SHA256: 73c366b300085ad4

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x403757a8
0x403757a8: esp_restart_noos at /home/rob/esp/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:143 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x164c
load:0x403b6000,len:0xb7c
load:0x403ba000,len:0x2f74
entry 0x403b6248
I (29) boot: ESP-IDF v4.4.1 2nd stage bootloader
I (29) boot: compile time 17:34:53
I (29) boot: chip revision: 0
I (31) boot.esp32s3: Boot SPI Speed : 80MHz
I (36) boot.esp32s3: SPI Mode       : DIO
I (40) boot.esp32s3: SPI Flash Size : 8MB
I (45) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 00113000
I (84) boot:  3 user_data        Unknown data     01 82 00123000 006d7000
I (91) boot: End of partition table
I (95) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0ff0ch ( 65292) map
I (116) esp_image: segment 1: paddr=0001ff34 vaddr=3fc91fd0 size=000e4h (   228) load
I (116) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=2d5d8h (185816) map
I (155) esp_image: segment 3: paddr=0004d600 vaddr=3fc920b4 size=02e7ch ( 11900) load
I (157) esp_image: segment 4: paddr=00050484 vaddr=40374000 size=0dfc8h ( 57288) load
I (173) esp_image: segment 5: paddr=0005e454 vaddr=50000000 size=00010h (    16) load
I (179) boot: Loaded app from partition at offset 0x10000
I (179) boot: Disabling RNG early entropy source...
I (193) opi psram: vendor id : 0x0d (AP)
I (193) opi psram: dev id    : 0x02 (generation 3)
I (194) opi psram: density   : 0x03 (64 Mbit)
I (197) opi psram: good-die  : 0x01 (Pass)
I (202) opi psram: Latency   : 0x01 (Fixed)
I (207) opi psram: VCC       : 0x01 (3V)
I (211) opi psram: SRF       : 0x01 (Fast Refresh)
I (217) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (222) opi psram: BurstLen  : 0x01 (32 Byte)
I (227) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (233) opi psram: DriveStrength: 0x00 (1/1)
I (238) spiram: Found 64MBit SPI RAM device
I (243) spiram: SPI RAM mode: sram 40m
I (247) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (255) cpu_start: Pro cpu up.
I (258) cpu_start: Starting app cpu, entry point is 0x403752d8
0x403752d8: call_start_cpu1 at /home/rob/esp/esp-idf/components/esp_system/port/cpu_start.c:160

I (243) cpu_start: App cpu up.
I (996) spiram: SPI SRAM memory test OK
I (1004) cpu_start: Pro cpu start user code
I (1004) cpu_start: cpu freq: 160000000
I (1004) cpu_start: Application information:
```

### Describe the solution you'd like.

I would like for the bootloader to use the original `efuse` component implementation provided by IDF, instead of the mocked `efuse` component that is defined in my project.

### Describe alternatives you've considered.

I have tried ignoring mocked function calls by default setting the CMock option `fail_on_unexpected_calls` to false, but this has no effect.

I am also open to implementing some other workaround that does not involve registering a mocked component that overrides the `efuse` name. Maybe I can make my own component called `mock_efuse` and link my app component against my custom mocked component instead?

chipweinberger, 2023-01-28:
The bootloader has it's own `components/bootloader/CMakeLists.txt`, so I'd think this is possible somehow.

`in build/bootloader/project_description.json` you can see the efuse component path used.

chipweinberger, 2023-01-28:
Look relevant:

**components/bootloader/subproject/CMakeLists.txt**

```
# A number of these components are implemented as config-only when built in the bootloader
set(COMPONENTS
    bootloader
    esptool_py
    esp_hw_support
    esp_system
    freertos
    hal
    partition_table
    soc
    bootloader_support
    log
    spi_flash
    micro-ecc
    main
    efuse
    esp_system
    newlib)

# Make EXTRA_COMPONENT_DIRS variable to point to the bootloader_components directory
# of the project being compiled
set(PROJECT_EXTRA_COMPONENTS "${PROJECT_SOURCE_DIR}/bootloader_components")
list(APPEND EXTRA_COMPONENT_DIRS "${PROJECT_EXTRA_COMPONENTS}")
```

0xjakob, 2023-02-01:
@robbawebba Thanks for the investigation and the problem report! We'll take a look. It seems that the code calling the mocked functions is actually not from the bootloader. Note the `I (250) cpu_start: Pro cpu up.` message. It's [part of the esp_system code](https://github.com/espressif/esp-idf/blob/master/components/esp_system/port/cpu_start.c#L415). Components overridden are not automatically overridden in the bootloader as well. Instead, there is a separate mechanism to override components in the bootloader.

If the startup code (after the bootloader finishes, before `app_main()` is entered) calls functions from mocked components, then mocking is tricky at best, at least as it looks to me now. So far, we haven't encountered this problem, so we will need to discuss this internally as well...

One possibility to workaround would be to add some hooks in the early startup code to prepare the mocks before anything happens. But this sounds quite hacky.

Another possibility is: Have you considered testing your application of parts of it on the host? If possible, you won't have this problem and additionally the testing might run much faster (no upload clycles). In the docs, You can find documentation about [running applications on host](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/host-apps.html). Further down, there is also a little table with the current status of which IDF components are available as mocks or implementations (or both).

robbawebba, 2023-02-01:
> It's part of the esp_system code. Components overridden are not automatically overridden in the bootloader as well. Instead, there is a separate mechanism to override components in the bootloader.

Okay, that makes sense. Thanks for clarifying!

> Another possibility is: Have you considered testing your application of parts of it on the host?

I considered this possibility awhile ago when I first began implementing unit tests. My project still uses v4.4.1, and the linked document doesn't seem to exist for this version. When I started adding unit tests it looked like on-target tests were the more mature solution at the time. However, I seem to be reinventing the wheel with all of my mocking. After reviewing the linked documentation, it looks like host testing is pretty well supported at this point and IDF will probably have most or all of the mocked components that I need! I'll try this out soon and report back.

0xjakob, 2023-02-02:
@robbawebba Thanks for the feedback! I would suggest to switch to 5.0 or above for host testing, as far as you can achieve that with reasonable costs. From 5.0, the linux support is much better.

In general, the idea was to also introduce mocks for target tests (as you have demonstrated), but we haven't thought about the consequences and feasibility yet. At the moment, we rather focus on moving more tests in IDF to host tests and to QEMU, so quite the opposite.

robbawebba, 2023-02-02:
> At the moment, we rather focus on moving more tests in IDF to host tests and to QEMU, so quite the opposite.

Totally understand, this sounds like the better approach for unit testing. Feel free to close this issue if you think there are no further actions.

I'll try to migrate our unit tests to host tests soon. I think a 5.0 migration is feasible, except for maybe some extra libraries like esp-aws-iot and esp-adf.

0xjakob, 2023-02-22:
@robbawebba As discussed, I'm closing this issue. But please let us know (in a new issue) if you encounter any problems with migrating tests to the host.

closed 2023-02-22 by 0xjakob