espressif/esp-idf #10636: adc Samples delivered in wrong order / dropped on ESP32 / continuous read with more than 2 channels

Kurs270, 2023-01-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-3035-gd825753387

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Esp32-wrover

### Power Supply used.

USB

### What is the expected behavior?

I changed examples/peripherals/adc/continuous_read/main/continuous_read_main.c to use 4 instead of only 2 adc channels.

//static adc_channel_t channel[2] = {ADC_CHANNEL_6, ADC_CHANNEL_7};
static adc_channel_t channel[] = {ADC_CHANNEL_4, ADC_CHANNEL_5, ADC_CHANNEL_6, ADC_CHANNEL_7};

I would expect the program output to be:
I (396) TASK: ret is 0, ret_num is 256
I (396) EXAMPLE: Unit: ADC_UNIT_1, Channel: 4, Value: c07
I (396) EXAMPLE: Unit: ADC_UNIT_1, Channel: 5, Value: 8e6
I (416) EXAMPLE: Unit: ADC_UNIT_1, Channel: 6, Value: 28b
I (406) EXAMPLE: Unit: ADC_UNIT_1, Channel: 7, Value: fff
I (426) EXAMPLE: Unit: ADC_UNIT_1, Channel: 4, Value: ede
I (416) EXAMPLE: Unit: ADC_UNIT_1, Channel: 5, Value: 98b
I (436) EXAMPLE: Unit: ADC_UNIT_1, Channel: 6, Value: 390
I (436) EXAMPLE: Unit: ADC_UNIT_1, Channel: 7, Value: fff
...and so on.

### What is the actual behavior?

However, the samples are delivered in arbitrary, non-constant order like this. (Samples are ordered in a non-reproducible manner, samples are left out/dropped etc..)

I (396) TASK: ret is 0, ret_num is 256
I (396) EXAMPLE: Unit: ADC_UNIT_1, Channel: 5, Value: 8e6
I (396) EXAMPLE: Unit: ADC_UNIT_1, Channel: 4, Value: c07
I (406) EXAMPLE: Unit: ADC_UNIT_1, Channel: 7, Value: fff
I (416) EXAMPLE: Unit: ADC_UNIT_1, Channel: 6, Value: 28b
I (416) EXAMPLE: Unit: ADC_UNIT_1, Channel: 5, Value: 98b
I (426) EXAMPLE: Unit: ADC_UNIT_1, Channel: 4, Value: ede
I (436) EXAMPLE: Unit: ADC_UNIT_1, Channel: 7, Value: fff
I (436) EXAMPLE: Unit: ADC_UNIT_1, Channel: 6, Value: 390
I (446) EXAMPLE: Unit: ADC_UNIT_1, Channel: 7, Value: fff
I (446) EXAMPLE: Unit: ADC_UNIT_1, Channel: 4, Value: fff
I (456) EXAMPLE: Unit: ADC_UNIT_1, Channel: 5, Value: b40
I (466) EXAMPLE: Unit: ADC_UNIT_1, Channel: 4, Value: fff
I (466) EXAMPLE: Unit: ADC_UNIT_1, Channel: 7, Value: fff
I (476) EXAMPLE: Unit: ADC_UNIT_1, Channel: 6, Value: 561
I (476) EXAMPLE: Unit: ADC_UNIT_1, Channel: 5, Value: c3f
I (486) EXAMPLE: Unit: ADC_UNIT_1, Channel: 4, Value: fff
[... 249 lines cut ...]
I (1266) EXAMPLE: Unit: ADC_UNIT_1, Channel: 5, Value: fff
I (1276) EXAMPLE: Unit: ADC_UNIT_1, Channel: 4, Value: fff
I (1276) EXAMPLE: Unit: ADC_UNIT_1, Channel: 5, Value: fff
I (1286) EXAMPLE: Unit: ADC_UNIT_1, Channel: 4, Value: fff
I (1286) EXAMPLE: Unit: ADC_UNIT_1, Channel: 7, Value: fff
I (1296) EXAMPLE: Unit: ADC_UNIT_1, Channel: 6, Value: fff
I (1306) EXAMPLE: Unit: ADC_UNIT_1, Channel: 5, Value: fff
I (1306) EXAMPLE: Unit: ADC_UNIT_1, Channel: 4, Value: fff
I (1316) EXAMPLE: Unit: ADC_UNIT_1, Channel: 7, Value: fff
I (1326) EXAMPLE: Unit: ADC_UNIT_1, Channel: 6, Value: fff
I (1326) EXAMPLE: Unit: ADC_UNIT_1, Channel: 7, Value: fff
I (1336) EXAMPLE: Unit: ADC_UNIT_1, Channel: 4, Value: fff
I (1336) EXAMPLE: Unit: ADC_UNIT_1, Channel: 5, Value: fff
I (1346) EXAMPLE: Unit: ADC_UNIT_1, Channel: 4, Value: fff
I (1356) EXAMPLE: Unit: ADC_UNIT_1, Channel: 7, Value: fff
I (1356) EXAMPLE: Unit: ADC_UNIT_1, Channel: 6, Value: fff
I (1366) EXAMPLE: Unit: ADC_UNIT_1, Channel: 5, Value: fff
I (1366) EXAMPLE: Unit: ADC_UNIT_1, Channel: 4, Value: fff
I (1376) EXAMPLE: Unit: ADC_UNIT_1, Channel: 7, Value: fff
```

### More Information.

The program runs on a development board with all the inputs left open, so I don't know if the values displayed are correct.
Anyway, the values should appear in exactly the configured order.

Icarus113, 2023-02-23:
@Kurs270  Hi, thanks for your report, I can reproduce the scenario. From my immediate idea, this might because the pattern table of the ESP32 is different with other chips. 

We'll check this and let you know the updates.

Icarus113, 2023-02-23:
@Kurs270  I'm very sorry that the ESP32 ADC DMA mode isn't easy to use.

The ESP32 ADC uses I2S0 DMA to finish its data tranaction (from ADC -> DMA -> memory). However, the ESP32 I2S is not easy to use..

[image]

See this picture, as ESP32 ADC result is 16bit, so even:
- set I2S 16 bits
- set I2S stereo
You will still get wrong data, as under stereo mode, the ADC result will be doubled, this means you'll get two same data...
See my test result
```
I (590) TASK: ret is 0, ret_num is 256
I (600) EXAMPLE: Unit: ADC_UNIT_1, Channel: 1, Value: 70
I (600) EXAMPLE: Unit: ADC_UNIT_1, Channel: 2, Value: 190
I (610) EXAMPLE: Unit: ADC_UNIT_1, Channel: 2, Value: 190
I (610) EXAMPLE: Unit: ADC_UNIT_1, Channel: 3, Value: f8d
I (620) EXAMPLE: Unit: ADC_UNIT_1, Channel: 3, Value: f8d
I (630) EXAMPLE: Unit: ADC_UNIT_1, Channel: 4, Value: fff
I (630) EXAMPLE: Unit: ADC_UNIT_1, Channel: 4, Value: fff
I (640) EXAMPLE: Unit: ADC_UNIT_1, Channel: 5, Value: fff
I (640) EXAMPLE: Unit: ADC_UNIT_1, Channel: 5, Value: fff
I (650) EXAMPLE: Unit: ADC_UNIT_1, Channel: 6, Value: fff
I (660) EXAMPLE: Unit: ADC_UNIT_1, Channel: 6, Value: fff
I (660) EXAMPLE: Unit: ADC_UNIT_1, Channel: 7, Value: fff
I (670) EXAMPLE: Unit: ADC_UNIT_1, Channel: 7, Value: fff
I (680) EXAMPLE: Unit: ADC_UNIT_1, Channel: 0, Value: fff
I (680) EXAMPLE: Unit: ADC_UNIT_1, Channel: 0, Value: fff
I (690) EXAMPLE: Unit: ADC_UNIT_1, Channel: 1, Value: 505
I (690) EXAMPLE: Unit: ADC_UNIT_1, Channel: 1, Value: 505
I (700) EXAMPLE: Unit: ADC_UNIT_1, Channel: 2, Value: 250
I (710) EXAMPLE: Unit: ADC_UNIT_1, Channel: 4, Value: fff
I (710) EXAMPLE: Unit: ADC_UNIT_1, Channel: 5, Value: fff
I (720) EXAMPLE: Unit: ADC_UNIT_1, Channel: 5, Value: fff
I (720) EXAMPLE: Unit: ADC_UNIT_1, Channel: 6, Value: fff
I (730) EXAMPLE: Unit: ADC_UNIT_1, Channel: 6, Value: fff
I (740) EXAMPLE: Unit: ADC_UNIT_1, Channel: 7, Value: fff
I (740) EXAMPLE: Unit: ADC_UNIT_1, Channel: 7, Value: fff
I (750) EXAMPLE: Unit: ADC_UNIT_1, Channel: 0, Value: fff
I (750) EXAMPLE: Unit: ADC_UNIT_1, Channel: 0, Value: fff
```
(ignore the data, those pins just floating, see the `Channel`)

I'm very sorry as I can't make it out, the ESP32 is been a long time... I've recorded this and will try the best to push this process.

What I can suggest now is:
- If you can use other chips, for example ESP32S3, turn to use that one
- If you stick to ESP32, only use 1 pin, or at most 2 pin at the same time. 

I will also update the programming guide for this issue, at least you guys should have a known issue note.

hephaisto, 2023-05-26:
I have the same problem. Did I understand correctly that there will be no fix for this?

baechipower, 2024-07-02:
I just noticed a similar problem on the ESP32S3 with 8 channels used as in Espressif's Arduino example. It seems that the sampling frequency plays a role. However the channel order stays the same with a given configuration and seems not to change non-deterministically.