espressif/esp-idf #10638: Cannot debug at all. Very discouraged: please help.

ChrisJones79, 2023-01-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using IDFv5.0 in vscode on Win10x64. The device I'm using is the esp32-s3 n32r8.

Just to try and get debugging, I opened the cdc_acm_vcp example and built it. I checked the boxes in config so that both the bootloader and application had th -Og debug flag.

After flashing, I went to the vscode sidebar and clicked launch arrow
[image]

All that is in launch.json:

    {
      "version": "0.2.0",
      "configurations": [
        {
          "type": "espidf",
          "name": "Launch",
          "request": "launch"
        }
      ]
    }

I currently have both USB ports connected to the desktop, as seen here:
[image]
I put a copy of the esp32s3 svd file in my project directory, where the main folder is.
I've reset in to the bootloader and saw different messages, they are below. 

I really don't know what to do. I've always used printf messages and never really had very large programs. So just using them worked enough. But I have convinced my employer to use ESP32 for upcoming projects and my first project is pretty big, many source files and printf isn't enough to make steady progress. 

I really need help getting debugging working. Thank you in advance for any help!

I will copy everything that was printed during the build and debug start attempts here:

```
    Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
    Executing action: fullclean
    Build directory 'c:\users\cjone\code\esp-idf\simpledebug\build' not found. Nothing to clean.
    Executing action: set-target
    Set Target to: esp32s3, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
    Running cmake in directory c:\users\cjone\code\esp-idf\simpledebug\build
    Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=0 c:\users\cjone\code\esp-idf\simpledebug"...
    -- Found Git: C:/Users/cjone/.espressif/tools/idf-git/2.30.1/cmd/git.exe (found version "2.30.1.windows.1") 
    -- The C compiler identification is GNU 11.2.0
    -- The CXX compiler identification is GNU 11.2.0
    -- The ASM compiler identification is GNU
    -- Found assembler: C:/Users/cjone/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe
    -- Detecting C compiler ABI info
    -- Detecting C compiler ABI info - done
    -- Check for working C compiler: C:/Users/cjone/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
    -- Detecting C compile features
    -- Detecting C compile features - done
    -- Detecting CXX compiler ABI info
    -- Detecting CXX compiler ABI info - done
    -- Check for working CXX compiler: C:/Users/cjone/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
    -- Detecting CXX compile features
    -- Detecting CXX compile features - done
    -- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
    -- Building ESP-IDF components for target esp32s3
    Solving dependencies requirements
    
    .......Updating lock file at C:/Users/cjone/code/esp-idf/SimpleDebug\dependencies.lock
    
    Processing 2 dependencies:
    
[... 1077 lines cut ...]
      File "threading.py", line 932, in _bootstrap_inner
    Exception in thread Thread-2:
    Traceback (most recent call last):
      File "threading.py", line 932, in _bootstrap_inner
      File "threading.py", line 870, in run
      File "threading.py", line 870, in run
      File "subprocess.py", line 1366, in _readerthread
      File "subprocess.py", line 1366, in _readerthread
    OSError: [Errno 22] Invalid argument
    [Debug Adapter]
    DEBUG_ADAPTER_STOPPED
    2023-01-28 14:10:55,472 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED
    [Debug Adapter]
    OSError: [Errno 22] Invalid argument
    [OpenOCD]
    Info : [esp32s3.cpu0] Target halted, PC=0x40041A76, debug_r[OpenOCD]
    eason=00000000
    [OpenOCD]
    Info : Set GDB target to 'esp32s3.cpu0'
    [OpenOCD]
    Info : [esp32s3.cpu1] Target halted, PC=0x40043A40, debug_reason=00000000
    [Stopped] : ESP-IDF Debug Adapter
    [OpenOCD]
Info : dropped 'gdb' connection
```

ChrisJones79, 2023-01-28:
Here is the doctor command output:

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\Microsoft\jdk-11.0.12.7-hotspot\bin;C:\Program Files\Eclipse Adoptium\jdk-17.0.5.8-hotspot\bin;C:\Program Files\Microsoft MPI\Bin\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files\TortoiseGit\bin;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\ProgramData\chocolatey\bin;C:\Program Files\nodejs\;C:\Program Files\dotnet\;C:<HOMEPATH>\Programs\Python\Python311\Scripts\;C:<HOMEPATH>\Programs\Python\Python311\;C:\Program Files\ngsolve-v6.2.2204\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:\Program Files\Graphviz\bin;C:\Program Files\7-Zip;E:\AVRDUDE;C:<HOMEPATH>\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\AppData\Local\Programs\MiKTeX\miktex\bin\x64\;C:\Program Files (x86)\GnuWin32\bin;C:\Xilinx\14.7_VM;C:\Program Files\Git\bin;C:\Program Files\Elmer 9.0-Release\bin;C:\Program Files\ParaView 5.11.0\bin;C:\MinGW\bin;C:<HOMEPATH>\Programs\imagej\ij153t-src\source\apache-ant-1.10.12\bin;C:<HOMEPATH>\AppData\Roaming\npm;C:\Program Files\CMake\bin;C:<HOMEPATH>\Programs\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706\openocd-esp32\bin; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.74.3 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.5.1 
Workspace folder c:<HOMEPATH>\code\esp-idf\SimpleDebug 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\espV5\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.10.2;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\cjone\\.espressif\\tools\\openocd-esp32\\v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM35
OpenOCD Configs (idf.openOcdConfigs) board/esp32s3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin: true
Access to C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706\openocd-esp32\bin: true
Access to C:<HOMEPATH>\.espressif\tools\ninja\1.10.2: true
Access to C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3: true
Access to C:<HOMEPATH>\.espressif\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64: true
Access to C:<HOMEPATH>\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
[... cut ...]
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            }
        }
    ],
    "version": 4
}
 
--------------------------------------------------------------------------------------------------------------------------------------------

```

chipweinberger, 2023-01-29:
look here:

https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md

brianignacio5, 2023-01-30:
Have you check the log output, there seems to be an error in your application regarding `uxTopUsedPriority` ?

```
❌ Error: FreeRTOS: uxTopUsedPriority is not defined, consu[OpenOCD]
    lt the OpenOCD manual for a work-around!
    [Debug Adapter]
    Exception in thread Thread-1:
    Traceback (most recent call last):
      File "threading.py", line 932, in _bootstrap_inner
    Exception in thread Thread-2:
    Traceback (most recent call last):
      File "threading.py", line 932, in _bootstrap_inner
      File "threading.py", line 870, in run
      File "threading.py", line 870, in run
      File "subprocess.py", line 1366, in _readerthread
      File "subprocess.py", line 1366, in _readerthread
```

ChrisJones79, 2023-01-30:
Thank you Brian, there is a file you can include which defines that. Or, you can kconfig it out by making the debugger not thread aware, or similar. What would help would be a complete tasks.json and launch.json along with the correct driver to use for the two interfaces(JTAG serial) available from the USB port.

You have a working configuration you can share?

ChrisJones79, 2023-01-30:
> look here:
> 
> https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md

Thanks, can you tell me what the difference is between using the espidf debug settings and the C/C++ settings? In terms of what is available and how to interface with the program? Also, how can I get the svd info to get used? Shouldn't it show up in the ESP peripheral view?

ChrisJones79, 2023-01-31:
@brianignacio5 I think I may have other problems. When following the instructions [for running GDB from the command line](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/jtag-debugging/using-debugger.html#command-line), this happened:

```
Microsoft Windows [Version 10.0.19045.2130]
(c) Microsoft Corporation. All rights reserved.

C:\Users\cjones\moreStaticVars>xtensa-esp32s3-elf-gdb -x gdbinit build/moreStaticVars.elf
GNU gdb (esp-gdb) 11.2_20220823
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-w64-mingw32 --target=xtensa-esp-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/moreStaticVars.elf...
gdbinit:1: Error in sourced command file:
:3333: Connection timed out.
(gdb)
```

First thing I noticed was that the target was incorrect, and I also mis-interpreted the `Error in sourced command` file line. I thought at first it meant my code, i.e. what I was trying to debug. Now I think it means the gdbinit file (from the prompt) although I did copy without altering from the instructions I linked to the ESP webpage.

Anyhow, I ran the `show configuration` command to see this:
```
gdbinit:1: Error in sourced command file:
:3333: Connection timed out.
(gdb)
(gdb) show configuration
This GDB was configured as follows:
   configure --host=x86_64-w64-mingw32 --target=xtensa-esp-elf
             --with-auto-load-dir=$debugdir:$datadir/auto-load
             --with-auto-load-safe-path=$debugdir:$datadir/auto-load
             --with-expat
             --with-gdb-datadir=//share/gdb (relocatable)
             --with-jit-reader-dir=//lib/gdb (relocatable)
             --without-libunwind-ia64
             --without-lzma
             --without-babeltrace
             --without-intel-pt
             --without-mpfr
             --without-xxhash
             --without-python
             --without-python-libdir
             --without-debuginfod
             --without-guile
             --disable-source-highlight
             --with-separate-debug-dir=//lib/debug (relocatable)

("Relocatable" means the directory can be moved with the GDB installation
tree, and GDB will still find it.)
(gdb) configure --target=xtensa-esp32s3-elf
Undefined command: "configure".  Try "help".
```

I tried changing the target but this is the first time I've used gdb so... Next I tried this:
```
(gdb) show auto-load
auto-load gdb-scripts:  Auto-loading of canned sequences of commands scripts is on.
auto-load local-gdbinit:  Auto-loading of .gdbinit script from current directory is on.
auto-load safe-path:  List of directories from which it is safe to auto-load files is $debugdir:$datadir/auto-load.
auto-load scripts-directory:  List of directories from which to load auto-loaded scripts is $debugdir:$datadir/auto-load.
(gdb) show data-directory
[... 32 lines cut ...]
GNU gdb (crosstool-NG esp-2022r1) 9.2.90.20200913-git
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-host_w64-mingw32 --target=xtensa-esp32s3-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/moreStaticVars.elf...
C:\Users\cjones\Programs\Espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\share\gdb\gdbinit:6: Error in sourced command file:
Scripting in the "Python" language is not supported in this copy of GDB.
(gdb) quit
```
The target is correct now, using the full path, and I can see that it's not even the same gdb now! Now it's `GNU gdb (crosstool-NG esp-2022r1) 9.2.90.20200913-git` where before it was `GNU gdb (esp-gdb) 11.2_20220823`

I feel like it's probably something obvious that experienced individuals such as yourself and @chipweinberger so please tell me if I'm wasting my time and hopefully what to do/where to go next.

Thanks

ChrisJones79, 2023-01-31:
I got something working for a brief moment but there was weird output form the openocd output. It said `hola juan` a bunch of times...

[image]

ChrisJones79, 2023-02-02:
@chipweinberger @brianignacio5 can either of you provide any further assistance?

Specifically these things:

1. Which debugger should I use for a devkitC esp32s3? (Like in the tasks/launch.json in vscode)
2. Which drivers should I from Zadig?
* There are two interfaces provided by the single USB port on the devkit
* * Drivers can be set somewhat independently for each.
* I know we need one to be a serial device (with a COM port), which driver?
* The other needs to be ? WinUSB? libusb0? libusbK?
* Does it matter if itf_0 is the serial and itf2 is the JTAG debugger? Does it have to be reversed?
3. Is this easier in the Esp IDE? The eclipse-based one?

Thanks for the help you have already given.

brianignacio5, 2023-02-03:
When things are not working I suggest going to basics.

First make sure you can debug a very basic esp-idf example (such as the blink example). This can help you separate the hardware and extension configuration issues from software issues.

Second, I think using the esp32s3 builtin usb config from OpenOCD should work. You can try just starting the openOCD server and see if it works.

Third, you can use any debugger you prefer, the `espidf` should work. Maybe changing the `logLevel` to 4 in launch.json can give us a better detail about the error in the debug session.

About the drivers: Check the [jtag configuration for esp32 s3](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/jtag-debugging/index.html#configure-and-connect-jtag-interface).

About the `hola Juan` output, apologies, was a log left out on development. It is fixed in master release. You could try with this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/workflows/ci/master/esp-idf-extension.vsix.zip)

closed 2023-02-09 by ChrisJones79

ChrisJones79, 2023-04-27:
> @ChrisJones79 <https://github.com/ChrisJones79> did you manage to get rid
> of the uxTopUsedPriority error by including that file? I'm new to ESP32 &
> FreeRTOS, how exactly should I include this file in my project?
> Thanks in advance!

@ThomasDQS

It was a while ago, I think the thing that worked for me was changing
something in menu config and possibly #defining that ux value.

Sorry I can't be of more help. I eventually just switched over to Linux and
was able to debug just fine.