espressif/esp-idf #10641: I2S_DATA_BIT_WIDTH_20BIT

plasmaphase, 2023-01-29:
### Is your feature request related to a problem?

I'm trying to get an ESP32-S3 device work with the ADAU7002 Stereo PDM-to-I2S device (https://www.analog.com/media/en/technical-documentation/data-sheets/ADAU7002.pdf).  Currently I see the following bit width support:

typedef enum {
    I2S_DATA_BIT_WIDTH_8BIT    = 8,            /*!< I2S channel data bit-width: 8 */
    I2S_DATA_BIT_WIDTH_16BIT   = 16,           /*!< I2S channel data bit-width: 16 */
    I2S_DATA_BIT_WIDTH_24BIT   = 24,           /*!< I2S channel data bit-width: 24 */
    I2S_DATA_BIT_WIDTH_32BIT   = 32,           /*!< I2S channel data bit-width: 32 */
} i2s_data_bit_width_t;

It's unclear to me whether just using the integer 20 would work or not.  I've been struggling to configure this part to work in stereo and it's difficult to know whether it's the part or the I2S configuration.

More details on my debug/progress and some code snippets here:
https://www.esp32.com/viewtopic.php?f=13&t=31748&p=108985#p108985

### Describe the solution you'd like.

Formal support for 20-bit data width I2S.

### Describe alternatives you've considered.

Just simply using 20 as the value for .data_bit_width:

i2s_std_slot_config_t slot_cfg = {
        .data_bit_width = 20,
...

### Additional context.

https://www.esp32.com/viewtopic.php?f=13&t=31748&p=108985#p108985

L-KAYA, 2023-01-30:
Unfortunately, PDM output only support 16-bit width (mentioned in the [pdm example](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/i2s/i2s_basic/i2s_pdm/main/i2s_pdm_tx.c#L58))

plasmaphase, 2023-01-30:
This would be standard I2S via the ADAU7002 part.  Are you saying even using the ESP32-S3 I2s_std is limited to 16-bits?  If I were interfacing via PDM to the IM69D130 part I could understand the 16-bit limitation, but the ADAU7002 is a standard i2s interface with 20-bit resolution.

L-KAYA, 2023-01-30:
Oh, sorry, if you just want output 20-bit width in standard format, set `i2s_data_bit_width_t ` to `20` should work.

plasmaphase, 2023-01-30:
Ok, so then this would be a feature request to formally add that.

plasmaphase, 2023-02-01:
So with a 20-bit data_bit_width, and a 32-bit ws_width, I'm not having much success.  My config is as follows:

`   i2s_chan_config_t rx_chan_cfg = I2S_CHANNEL_DEFAULT_CONFIG(I2S_NUM_AUTO, I2S_ROLE_MASTER);
    ESP_ERROR_CHECK(i2s_new_channel(&rx_chan_cfg, NULL, &rx_chan));

    i2s_std_clk_config_t clk_cfg = {
        .clk_src = I2S_CLK_SRC_DEFAULT,
        .sample_rate_hz = SAMPLE_RATE,
        .mclk_multiple = I2S_MCLK_MULTIPLE_384};

    i2s_std_slot_config_t slot_cfg = {
        .data_bit_width = BIT_WIDTH,
        .slot_bit_width = I2S_SLOT_BIT_WIDTH_AUTO,
        .slot_mode = I2S_SLOT_MODE_STEREO,
        .slot_mask = I2S_STD_SLOT_BOTH,
        .ws_width = I2S_DATA_BIT_WIDTH_32BIT,
        .ws_pol = false,
        .bit_shift = true,
        .left_align = false,
        .big_endian = false,
        .bit_order_lsb = false};

    i2s_std_config_t rx_std_cfg = {
        .clk_cfg = clk_cfg,
        .slot_cfg = slot_cfg,
        .gpio_cfg = {
            .mclk = I2S_GPIO_UNUSED,
            .bclk = I2S_BCLK,
            .ws = I2S_WS,
            .dout = I2S_GPIO_UNUSED,
            .din = I2S_DIN,
            .invert_flags = {
                .mclk_inv = false,
                .bclk_inv = false,
                .ws_inv = false,
            },
        },
    };

    rx_std_cfg.slot_cfg.slot_mask = I2S_STD_SLOT_BOTH;
    ESP_ERROR_CHECK(i2s_channel_init_std_mode(rx_chan, &rx_std_cfg));`

Sample rate is 44100, bit width is 20.

Has this even been tested?  Are you sure this is possible?

superware, 2024-06-18:
Hello @plasmaphase,

I'm also trying the `ADAU7002` with ESP32-S3, and saw your [post](https://www.esp32.com/viewtopic.php?t=31748#p108985). I still can't manage to extract the channels (left/right) data from the `i2s_channel_read` buffer. Can you please post the complete code as an example?

Thank you!