espressif/esp-idf #10643: Latest docker build (47ace812ee50) crashes macOS when installed through dev container

parksj10, 2023-01-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### OS

MacOS

### Operating System version

macOS 12.6

### Visual Studio Code version

1.74.3

### Docker Desktop

4.13.1 (90346)

When trying to install the latest docker build on macOS M1Max processor, through the VSCode dev container, it crashes the entire OS. The same procedure with 5.0 (e1dffee42b2f) installs fine

peterdragun, 2023-01-30:
Hi, can you please try a different version of Docker Desktop? Seems like there was a bug in 4.13.0 and 4.13.1: https://github.com/docker/for-mac/issues/6530
As per the comments, it seems that the newer version 4.16 doesn't have this issue, so I would suggest upgrading docker to 4.16 or if not possible then downgrading docker to 4.12. Let me know if that helped.

parksj10, 2023-01-31:
@peterdragun, upgraded to 4.16 and it still crashed. however, I docker pulled the image from the command line (`docker pull espressif/idf`), then opened from container when prompted by vscode and it worked... still a bug here with the latest release, but workaround makes it non-blocking for me

peterdragun, 2023-02-06:
I am glad that you were at least able to find a workaround for this issue.
Interesting that CLI worked without any issues, they should be doing the same thing. Could you please share your configuration for dev containers? That would be helpful in reproducing the issue locally.

parksj10, 2023-02-06:
@peterdragun, the dev container was boilerplate 

```json
// For format details, see https://aka.ms/devcontainer.json. For config options, see the README at:
// https://github.com/microsoft/vscode-dev-containers/tree/v0.183.0/containers/ubuntu
{
	"name": "ESP-IDF QEMU",
	"build": {
		"dockerfile": "Dockerfile"
	},
	// Add the IDs of extensions you want installed when the container is created.
	"extensions": [
		"ms-vscode.cpptools",
		"espressif.esp-idf-extension"
	],
	"workspaceMount": "source=${localWorkspaceFolder},target=/workspaces,type=bind",
	/* the path of workspace folder to be opened after container is running
	 */
	"workspaceFolder": "/workspaces",
	"mounts": [
		"source=extensionCache,target=/root/.vscode-server/extensions,type=volume"
	],
	"settings": {
		"terminal.integrated.defaultProfile.linux": "bash",
		"idf.espIdfPath": "/opt/esp/idf",
		"idf.customExtraPaths": "",
		"idf.pythonBinPath": "/opt/esp/python_env/idf5.1_py3.8_env/bin/python",
		"idf.toolsPath": "/opt/esp",
		"idf.gitPath": "/usr/bin/git"
	},
	"runArgs": ["--privileged"]
}
```

Also, just pulled directly from docker desktop with no issue

peterdragun, 2023-02-07:
Thanks.
Unfortunately, I am not sure if we can do something about that as you can pull and run the container using both CLI and Desktop Docker. This seems to be issue somewhere in dev containers then, but I am not sure as I was not able to reproduce this issue locally. Thanks for also providing the config, I can see that it is basically the default one. I was using the same one, but everything worked as expected for me. The only difference is that I am running macOS 13.1

closed 2023-02-13 by dobairoland