espressif/esp-idf #10644: IDF v5.0 regression: mqtt not providing `user_context` field

jmattsson, 2023-01-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I'm working on porting the NodeMCU mqtt module from IDFv4 to IDFv5, but I've run into what seems to be a big issue.

Previously*, it was possible to set a "user context" pointer when initialising the client, and that pointer would then be handed back in each event. This followed the typical pattern for event driven libraries in C, where the caller is free to associate whatever context with a session and have it accessible at the time of handling the event.

In IDFv5**, I cannot find any support for this. Is it really intentional that such a fundamental aspect of the library has been changed/removed? I'm looking at a significant rewrite of the module code here if that's the case. By the looks of it I would have to establish and maintain an external `esp_mqtt_client_handle_t` to user context lookup structure in order to achieve what the old `user_context` field did. Obviously, I'd prefer to not have to do that.

Have I overlooked something here? I can't find any mention of this change in the API documentation, release notes or migration notes.

I originally posted this on the [forum](https://esp32.com/viewtopic.php?f=13&t=31786) but got no response there. This seems like quite a regression to me.

Thanks in advance.

*) See [https://docs.espressif.com/projects/esp ... parameters](https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32/api-reference/protocols/mqtt.html#other-configuration-parameters)
**) See [https://docs.espressif.com/projects/esp ... figuration](https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32/api-reference/protocols/mqtt.html#configuration)

david-cermak, 2023-01-30:
Hi @jmattsson 

Yes, the field `user_context` was removed intentionally, since mqtt client uses IDF standard `esp-event`, which allows adding its own user context. It's briefly described in the examples:

https://github.com/espressif/esp-idf/blob/236fa5e6698a262aed534d2a46c78711204b903a/examples/protocols/mqtt/ssl/main/app_main.c#L135-L136

(note that `user_context` was used before migration to `esp-event` when notifications were posted as callbacks, and was kept for compatibility up to `v5.x`)

jmattsson, 2023-01-31:
Oh! Okay, that explains a lot. Thanks @david-cermak !
I hadn't seen any deprecation messages about the callbacks, so I was oblivious to the fact it'd been changed. I went back and checked all the release notes, and there is no mention of that anywhere that I can find. May I suggest adding a note to at least the 4->5 migration doc about this change before someone else gets similarly confused? :)
On my end I only saw errors about missing `user_context` and the changed `esp_mqtt_client_config_t` structure, which made me stare at the wrong things and miss the bigger picture.

Thanks again for the quick response. Feel free to close the issue (unless you want it to hang around for doc update reasons).

AxelLin, 2023-01-31:
migration-guides should mention this change.

closed 2023-02-03 by espressif-bot