espressif/esp-idf #10645: esp32c6 esp_zigbee_rcp example build error

lbuque, 2023-01-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-2283-gc92f2394f2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Trying to compile esp_zigbee_rcp for esp32-c6, running `idf.py --preview set-target esp32c6` cannot set the target normally.

### What is the actual behavior?

[image]

### Steps to reproduce.

1. cd examples/zigbee/esp_zigbee_rcp
2. idf.py --preview set-target esp32c6

### Build or installation Logs.

```plain
$ idf.py --preview set-target esp32c6
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/lewin/esp/esp-idf-v5.0/examples/zigbee/esp_zigbee_rcp/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32c6, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /home/lewin/esp/esp-idf-v5.0/examples/zigbee/esp_zigbee_rcp/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/lewin/.espressif/python_env/idf5.1_py3.8_env/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=esp32c6 -DIDF_TARGET=esp32h4 -DCCACHE_ENABLE=0 /home/lewin/esp/esp-idf-v5.0/examples/zigbee/esp_zigbee_rcp"...
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- Component directory /home/lewin/esp/esp-idf-v5.0/components/asio does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/lewin/esp/esp-idf-v5.0/components/cbor does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/lewin/esp/esp-idf-v5.0/components/coap does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/lewin/esp/esp-idf-v5.0/components/esp32 does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/lewin/esp/esp-idf-v5.0/components/expat does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/lewin/esp/esp-idf-v5.0/components/libsodium does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/lewin/esp/esp-idf-v5.0/components/nghttp does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/lewin/esp/esp-idf-v5.0/components/tinyusb does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/lewin/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc
-- Check for working C compiler: /home/lewin/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc
-- Check for working C compiler: /home/lewin/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/lewin/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/riscv32-esp-elf-g++
-- Check for working CXX compiler: /home/lewin/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/riscv32-esp-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
[... 10 lines cut ...]
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: /home/lewin/.espressif/python_env/idf5.1_py3.8_env/bin/python (found version "3.8.10") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "esp_zigbee_rcp" version: v5.1-dev-2283-gc92f2394f2
-- Adding linker script /home/lewin/esp/esp-idf-v5.0/examples/zigbee/esp_zigbee_rcp/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/lewin/esp/esp-idf-v5.0/components/esp_system/ld/esp32h4/sections.ld.in
-- Adding linker script /home/lewin/esp/esp-idf-v5.0/components/esp_rom/esp32h4/ld/rev2/esp32h4.rom.ld
-- Adding linker script /home/lewin/esp/esp-idf-v5.0/components/esp_rom/esp32h4/ld/rev2/esp32h4.rom.api.ld
-- Adding linker script /home/lewin/esp/esp-idf-v5.0/components/esp_rom/esp32h4/ld/rev2/esp32h4.rom.libgcc.ld
-- Adding linker script /home/lewin/esp/esp-idf-v5.0/components/esp_rom/esp32h4/ld/rev2/esp32h4.rom.newlib.ld
-- Adding linker script /home/lewin/esp/esp-idf-v5.0/components/esp_rom/esp32h4/ld/rev2/esp32h4.rom.version.ld
-- Adding linker script /home/lewin/esp/esp-idf-v5.0/components/soc/esp32h4/ld/esp32h4.peripherals.ld
zigbee-c92f2394f2-c92f2394f2-0.2.0-2023-01-30 06:52:28 UTC
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_netif esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__esp-zboss-lib espressif__esp-zigbee-lib esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table protobuf-c protocomm pthread riscv sdmmc soc spi_flash spiffs tcp_transport ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant
-- Component paths: /home/lewin/esp/esp-idf-v5.0/components/app_trace /home/lewin/esp/esp-idf-v5.0/components/app_update /home/lewin/esp/esp-idf-v5.0/components/bootloader /home/lewin/esp/esp-idf-v5.0/components/bootloader_support /home/lewin/esp/esp-idf-v5.0/components/bt /home/lewin/esp/esp-idf-v5.0/components/cmock /home/lewin/esp/esp-idf-v5.0/components/console /home/lewin/esp/esp-idf-v5.0/components/cxx /home/lewin/esp/esp-idf-v5.0/components/driver /home/lewin/esp/esp-idf-v5.0/components/efuse /home/lewin/esp/esp-idf-v5.0/components/esp-tls /home/lewin/esp/esp-idf-v5.0/components/esp_adc /home/lewin/esp/esp-idf-v5.0/components/esp_app_format /home/lewin/esp/esp-idf-v5.0/components/esp_common /home/lewin/esp/esp-idf-v5.0/components/esp_eth /home/lewin/esp/esp-idf-v5.0/components/esp_event /home/lewin/esp/esp-idf-v5.0/components/esp_gdbstub /home/lewin/esp/esp-idf-v5.0/components/esp_hid /home/lewin/esp/esp-idf-v5.0/components/esp_http_client /home/lewin/esp/esp-idf-v5.0/components/esp_http_server /home/lewin/esp/esp-idf-v5.0/components/esp_https_ota /home/lewin/esp/esp-idf-v5.0/components/esp_https_server /home/lewin/esp/esp-idf-v5.0/components/esp_hw_support /home/lewin/esp/esp-idf-v5.0/components/esp_lcd /home/lewin/esp/esp-idf-v5.0/components/esp_local_ctrl /home/lewin/esp/esp-idf-v5.0/components/esp_netif /home/lewin/esp/esp-idf-v5.0/components/esp_partition /home/lewin/esp/esp-idf-v5.0/components/esp_phy /home/lewin/esp/esp-idf-v5.0/components/esp_pm /home/lewin/esp/esp-idf-v5.0/components/esp_psram /home/lewin/esp/esp-idf-v5.0/components/esp_ringbuf /home/lewin/esp/esp-idf-v5.0/components/esp_rom /home/lewin/esp/esp-idf-v5.0/components/esp_system /home/lewin/esp/esp-idf-v5.0/components/esp_timer /home/lewin/esp/esp-idf-v5.0/components/esp_wifi /home/lewin/esp/esp-idf-v5.0/components/espcoredump /home/lewin/esp/esp-idf-v5.0/examples/zigbee/esp_zigbee_rcp/managed_components/espressif__esp-zboss-lib /home/lewin/esp/esp-idf-v5.0/examples/zigbee/esp_zigbee_rcp/managed_components/espressif__esp-zigbee-lib /home/lewin/esp/esp-idf-v5.0/components/esptool_py /home/lewin/esp/esp-idf-v5.0/components/fatfs /home/lewin/esp/esp-idf-v5.0/components/freertos /home/lewin/esp/esp-idf-v5.0/components/hal /home/lewin/esp/esp-idf-v5.0/components/heap /home/lewin/esp/esp-idf-v5.0/components/http_parser /home/lewin/esp/esp-idf-v5.0/components/idf_test /home/lewin/esp/esp [... cut ...] ransport /home/lewin/esp/esp-idf-v5.0/components/ulp /home/lewin/esp/esp-idf-v5.0/components/unity /home/lewin/esp/esp-idf-v5.0/components/usb /home/lewin/esp/esp-idf-v5.0/components/vfs /home/lewin/esp/esp-idf-v5.0/components/wear_levelling /home/lewin/esp/esp-idf-v5.0/components/wifi_provisioning /home/lewin/esp/esp-idf-v5.0/components/wpa_supplicant
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lewin/esp/esp-idf-v5.0/examples/zigbee/esp_zigbee_rcp/build
```

likunqiao097304, 2023-01-30:
@lbuque Hi. It might be your idf version is slightly old, our zigbee example support for C6 is ready only after this commit:
[zigbee c6 support MR](https://github.com/espressif/esp-idf/commit/83d1bedb25350cfbbbfeb4e05fa5ae7cd555a85a). You can try with this commit

lbuque, 2023-01-30:
@likunqiao097304 Your workaround works, it compiles successfully! ! ! I think this issue can be closed.

Alvin1Zhang, 2023-01-31:
Thanks for sharing the updates.

closed 2023-01-31 by Alvin1Zhang