espressif/esp-idf #10646: Wifi : "wifi:m f null" losing connection.

Coffeeye, 2023-01-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4-dev-3978-gf0628ea2db

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

Iam connecting to STA and the connection drops it self after couple of minutes.
Last wifi logs are :
```
D (08:12:07.579) event: running post WIFI_EVENT:4 with handler 0x4014b6bc and context 0x3ffe2518 on loop 0x3ffdcd98
0x4014b6bc: wifi_default_action_sta_connected at C:\Users\../esp-idf/components/esp_wifi/src/wifi_default.c:81

D (08:12:07.590) esp_netif_handlers: esp_netif action connected with netif0x3ffe260c from event_id=4
D (08:12:07.599) esp_netif_lwip: check: remote, if=0x3ffe260c fn=0x401486ec
0x401486ec: esp_netif_up_api at C:\Users\../esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1256

I (132293) wifi:AP's beacon interval = 102400 us, DTIM period = 1
D (08:12:07.606) esp_netif_lwip: esp_netif_up_api esp_netif:0x3ffe260c

D (132311) wifi:set max rate: from <rate=130, phy=3, sig=0> to <rate=144, phy=3 sig=0>
D (08:12:07.620) esp_netif_lwip: check: local, if=0x3ffe260c fn=0x40147c28
0x40147c28: esp_netif_update_default_netif_lwip at C:\Users\../esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:188

D (132318) wifi:sig_b=0, sig_g=0, sig_n=0, max_b=22, max_g=108, max_n=144
D (08:12:07.633) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffe260c
D (132331) wifi:update trc
D (08:12:07.647) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (08:12:07.656) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffe260c
D (08:12:07.663) esp_netif_lwip: esp_netif_get_old_ip_info esp_netif:0x3ffe260c
D (08:12:07.670) esp_netif_lwip: check: remote, if=0x3ffe260c fn=0x401484a8
0x401484a8: esp_netif_set_ip_old_info_api at C:\Users\../esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1380

D (08:12:07.677) esp_netif_lwip: esp_netif_set_ip_old_info_api esp_netif:0x3ffe260c
D (08:12:07.685) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (08:12:07.691) esp_netif_handlers: static ip: ip changed=0
D (08:12:07.697) event: running post IP_EVENT:0 with handler 0x400ddd58 and context 0x3ffde1c0 on loop 0x3ffdcd98
0x400ddd58: core_event at C:\Users/../main/system/core.c:80

D (08:12:07.704) httpd: httpd_server: processing listen socket 48
D (08:12:07.716) httpd: httpd_accept_conn: newfd = 51
D (08:12:07.724) httpd: httpd_accept_conn: complete
D (08:12:07.734) httpd: httpd_server: doing select maxfd+1 = 52
D (08:12:07.739) event: running post IP_EVENT:0 with handler 0x4014b664 and context 0x3ffe25ec on loop 0x3ffdcd98
0x4014b664: wifi_default_action_sta_got_ip at C:\Users\./esp-idf/components/esp_wifi/src/wifi_default.c:123
[... 137 lines cut ...]
        |  |    esp_netif_set_ip_info |        |  CONNECT STA  -----------   esp_wifi_connect      |              |                                                                  
        |  |                          |        +-------|-------+         +-------------------------+              |                                                                  
        |  |   esp_netif_set_dns_info |                |                                           +--------------|-------------+                                                    
        |  |                          |                |                                           |   +--------------------+   |                                                    
        |  |   esp_netif_dhcps_option |                +--------------------------------------------   |  RESTART WEBSERVER |   |                                                    
        |  |                          |                                                            |   +--------------------+   |                                                    
        |  |    esp_netif_dhcps_start |                                                            +----------------------------+                                                    
        |  |                          |       +-----------------+                                                                                                                    
        |  |           esp_wifi_start |       | +-------------+ |                                                                                                                    
        |  +----------------------------------- | DNS SERVER  | |                                                                                                                    
        |   +------------+                    | +-------------+ |                                                                                                                    
        +---- WIFI SCAN  |------------+       +-----------------+                                                                                                                    
            +------------+            |                                                                                                                                              
           +--------------------------|                                                                                                                                              
           |   esp_wifi_set_mode STA  |                                                                                                                                              
           |                          |                                                                                                                                              
           |          esp_wifi_start  |                                                                                                                                              
           |                          |                                                                                                                                              
           |     esp_wifi_start_scan  |                                                                                                                                              
           +--------------------------+                                                                                                                                              
           ```

### Steps to reproduce.

.

closed 2023-01-30 by Coffeeye

LeBellier, 2023-02-06:
I still have this problem. Can you share the commit containing the fix?

Coffeeye, 2023-02-07:
> I still have this problem. Can you share the commit containing the fix?

It all strictly reference to actual debug prints of your wifi module. I had no other prints than this one, nothing about memory allocation or whatsoever. In fact I still didnt manage to completly avoid the issue.