espressif/esp-idf #10647: [v4.4.3][ESP32-S3] "Timed out waiting for completion of AES Interrupt"

chipweinberger, 2023-01-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3 Dev Kit C

### Power Supply used.

USB

### What is the expected behavior?

AES CBC should complete.

### What is the actual behavior?

Hits AES timeout.

```
/* Wait for AES hardware block operation to complete */
static void esp_aes_dma_wait_complete(bool use_intr, lldesc_t *output_desc)
{
#if defined (CONFIG_MBEDTLS_AES_USE_INTERRUPT)
    if (use_intr) {
        if (!xSemaphoreTake(op_complete_sem, 2000 / portTICK_PERIOD_MS)) {
            /* indicates a fundamental problem with driver */
            ESP_LOGE("AES", "Timed out waiting for completion of AES Interrupt");
            abort();
        }
#ifdef CONFIG_PM_ENABLE
        esp_pm_lock_release(s_pm_cpu_lock);
        esp_pm_lock_release(s_pm_sleep_lock);
#endif  // CONFIG_PM_ENABLE
    }
#endif
```

### Steps to reproduce.

I've never hit this before. Might be tempermental.

### Debug Logs.

```plain
0x40376487: panic_abort at /esp-idf/components/esp_system/panic.c:429

0x40383581: esp_system_abort at /esp-idf/components/esp_system/esp_system.c:128

0x4038daab: abort at /esp-idf/components/newlib/abort.c:46

0x4211057e: esp_aes_dma_wait_complete at /esp-idf/components/mbedtls/port/aes/dma/esp_aes.c:215
 (inlined by) esp_aes_process_dma at /esp-idf/components/mbedtls/port/aes/dma/esp_aes.c:435

0x421109a3: esp_aes_crypt_cbc at /esp-idf/components/mbedtls/port/aes/dma/esp_aes.c:729

0x4207f297: pd_ota_session_write(unsigned long long, pd_buf_t, pd_error_t*) at /jamcorder-firmware/jamcorder_common/jcommon/ota/pd_ota_session.cpp:467
 (inlined by) pd_ota_session_write(unsigned long long, pd_buf_t, pd_error_t*) at /jamcorder-firmware/jamcorder_common/jcommon/ota/pd_ota_session.cpp:672
```

chipweinberger, 2023-01-30:
This is really bad behavior. Crashing is not acceptable. 

Can we not return an error? Should we increase this timeout? is 2 seconds really enough?

chipweinberger, 2023-01-30:
PR to do something more friendly: https://github.com/espressif/esp-idf/pull/10648/files

ESP-Marius, 2023-02-08:
Hmm, I dont really see how this could happen. It would help a lot if you were able to find a way to reproduce it.

Adding an error return instead of abort seems reasonable, but I would still like to get to the bottom of how this happened.

ESP-Marius, 2023-02-08:
@chipweinberger are you by any chance using the Digital signature peripheral as part of your OTA process? (or at all in your app)

chipweinberger, 2023-02-08:
No I don't use DS peripheral for OTA.

Wifi + aes.

Yes it would be great to repro. I've only hit it once.

Thanks for merging the band-aid fix.

chipweinberger, 2023-02-08:
Oh one thing! At the time I hit this I was debugging a big memory leak in my OTA code! I was allocating a mutex in internal ram ~20 times a second.

My OTA would fail due to out of memory half way into the process.

Maybe ota + aes + memory exhaustion.

I am also using PSRAM, not sure if related.

mahavirj, 2023-03-20:
@chipweinberger 

Is it possible to create a small reproducible use-case for this issue? So far, we have not been able to recreate this scenario on our side (tried different configurations including SPIRAM).

chipweinberger, 2023-03-20:
I appreciate your effort!

I remember reproducing it a couple times at least, but i'm imagine a minimal example will take awhile for me to create as well.

Did you try creating a new mutex every time new ota data arrived?

I was doing the update over wifi.

mahavirj, 2023-03-20:
> Did you try creating a new mutex every time new ota data arrived?

No, I didn't. Do you mean OTA with low internal memory could run into this situation?

dsiganos, 2023-03-28:
I am hitting this problem every time. Increasing the main stack size to 6000 bytes makes it work. I do not know what is eating up the stack memory yet. However this likely means that the problem happens when the task performing the AES operations runs out of stack.

mahavirj, 2023-09-11:
Update:

We have added few fixes in the AES DMA port layer recently, following are pointers for v4.4 branch

- https://github.com/espressif/esp-idf/commit/6afa8e4347a8dc425131be040efd718808ab79e6 
- https://github.com/espressif/esp-idf/commit/da65a5c0db4e751cb8238fc601ee570e3d07d582

First one is the fix for heap corruption issue that may occur with specific AES dma lengths and second is specific to incorrect DMA descriptor setup. Please have a look at them, they are relevant to this issue. If you could not recreate "AES timeout" issue on latest `release/v4.4` branch then we can close this tracker.

Thanks.

closed 2023-10-03 by mahavirj

chipweinberger, 2023-11-28:
im still seeing this error in v5.1.2

as a workaround im using CONFIG_MBEDTLS_AES_USE_INTERRUPT=n

kel30a, 2024-01-19:
> im still seeing this error in v5.1.2
> 
> as a workaround im using CONFIG_MBEDTLS_AES_USE_INTERRUPT=n

Same. It was happening on 5.0.2 for me and still on 5.1.2 after upgrading.

Maldus512, 2024-01-19:
I'm facing the same issue while generating  the private key for a Rainmaker application. In order to make it work I have to increase `ESP_RMAKER_CLAIM_TASK_STACK_SIZE` to at least `22 * 1024` bytes (up from `10 * 1024`).