espressif/esp-idf #10652: Customizing COMPONENT_DIRS... Instructions or guidelines?

ChrisJones79, 2023-01-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am making a minimal project and tried a few things intending to trim down the included components in the build.

Using this guidance:
[image]

I have a project level CMakeLists.txt which includes the mandatory elements:
[image]

I tried editing the project level CMakeLists.txt file but I can't seem to get it to work.

So far I've tried

- set(COMPONENT_DIRS "")
- set(COMPONENT_DIRS ".")
- set(COMPONENT_DIRS "./main")
- set(COMPONENT "")
- set(COMPONENT ".")
- set(COMPONENT "./main")

> Any paths in these variables can be absolute paths, or set relative to the project directory.

> To set these variables, use the [cmake set command](https://cmake.org/cmake/help/v3.16/command/set.html) ie set(VARIABLE "VALUE"). The set() commands should be placed **after the cmake_minimum(...) line but before the include(...) line**.

I want to only bring in what is REQUIRED or PRIV_REQUIRED.

igrr, 2023-01-30:
Hi @ChrisJones79, you can set the COMPONENTS variable to the list of component names (not paths) before including project.cmake, as it is done in the following app:

https://github.com/espressif/esp-idf/blob/236fa5e6698a262aed534d2a46c78711204b903a/tools/test_apps/system/panic/CMakeLists.txt#L5

Note that your `main` component has to pass REQUIRES or PRIV_REQUIRES arguments to idf_component_register for this to reduce the number of components built.

ChrisJones79, 2023-01-30:
Thank you @igrr. 

By main component, this means that CMakeLists in my main folder? 

How do I know what to call the components that I do want to include? For example, if I want to use esp-ADC and a managed component like esp_tinyusb (how would I know it's 'adc' and not 'esp-adc' or something else?). My project cmaketxt would have this:

`set(COMPONENTS main espcoredump esp_gdbstub adc)`

before the `include($ENV{IDF_PATH}/tools/cmake/project.cmake)`

and also an idf_component.yml in main with
```
dependencies:
  espressif/esp_tinyusb: "^1.0.3"
  idf:
    version: ">=4.1.0"
```

Would the component manager still work and is the above a sufficient pattern for esp components and managed components?

Thanks!

igrr, 2023-02-01:
> By main component, this means that CMakeLists in my main folder?

Yes, `main/CMakeLists.txt`.

> How do I know what to call the components that I do want to include?

Unfortunately the only way to answer this question, for now, is to find the header file you want to use in esp-idf and see in which component it is located. The component name will be equal to the component directory name. So in the case of the ADC related functions that would be esp_adc.

The REQUIRES or PRIV_REQUIRES list passed to idf_component_register should contain the names of components (== component directories), not file names. So that would be `REQUIRES esp_adc` in your case.

> set(COMPONENTS main espcoredump esp_gdbstub adc)

This was just an example, sorry if I wasn't clear. You can change it to suit your application.

ChrisJones79, 2023-02-01:
Last question, are any components required?

closed 2023-02-09 by ChrisJones79