espressif/esp-idf #10654: RTC wdt not active in deep sleep, even with RTC_CNTL_WDT_PAUSE_IN_SLP cleared

zevv, 2023-01-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-2668-g4c98bee8a4

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

-

### Power Supply used.

USB

### What is the expected behavior?

I am using the RTC watchdog and want it to trigger in deep sleep mode. I noticed that `rtc_wdt_enable()` disables the RTC WDT in deep sleep with

```
SET_PERI_REG_MASK(RTC_CNTL_WDTCONFIG0_REG, RTC_CNTL_WDT_EN | RTC_CNTL_WDT_PAUSE_IN_SLP);
```

Instead, I now replaced `rtc_wdt_enable()` with this in my code:

```
    SET_PERI_REG_MASK(RTC_CNTL_WDTCONFIG0_REG, RTC_CNTL_WDT_EN);
    CLEAR_PERI_REG_MASK(RTC_CNTL_WDTCONFIG0_REG, RTC_CNTL_WDT_PAUSE_IN_SLP);
```
I have verified that the `RTC_CNTL_WDT_PAUSE_IN_SLP` is really cleared in `RTC_CNTL_WDTCONFIG0_REG`

### What is the actual behavior?

When entering deep sleep, the RTC WDT will not wake the CPU. Instead, it will be stuck in deep sleep state indefinitely.

### Steps to reproduce.

Compile and run the below program. When replacing `esp_deep_sleep_start()` with `for();`, the watchdog will properly trigger and the ESP will report `RTCWDT_RTC_RESET` as boot reason.

```
#include <rtc_wdt.h>    
#include <stdio.h>
#include <esp_sleep.h>

void app_main(void)
{
        rtc_wdt_protect_off();
        rtc_wdt_disable();
        rtc_wdt_set_stage(RTC_WDT_STAGE0, RTC_WDT_STAGE_ACTION_RESET_SYSTEM);
        rtc_wdt_set_time(RTC_WDT_STAGE0, 3000);
        SET_PERI_REG_MASK(RTC_CNTL_WDTCONFIG0_REG, RTC_CNTL_WDT_EN);
        CLEAR_PERI_REG_MASK(RTC_CNTL_WDTCONFIG0_REG, RTC_CNTL_WDT_PAUSE_IN_SLP);  
        rtc_wdt_protect_on();
                                                                                
        printf("entering deep sleep\n");
        esp_deep_sleep_start();
        printf("should never get here\n");
}
```

ESP-Marius, 2023-01-31:
The system, i.e. CPU and peripherals are all shutdown during sleep, so doing `RTC_WDT_STAGE_ACTION_RESET_SYSTEM` probably ends up doing nothing.

Using `RTC_WDT_STAGE_ACTION_RESET_RTC` which will also reset the RTC systems seems to work, you can try this.

zevv, 2023-01-31:
Great, solves my problem. Thank you!

ESP-Marius, 2023-01-31:
Good! I'll mark this issue as resolved then.

closed 2023-01-31 by ESP-Marius