espressif/esp-idf #10656: Can't Run any code to ESP32

pyrotechnik, 2023-01-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.0-dirty

### Operating System used.

macOS

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-D0WD-V3 (revision 3)

### Power Supply used.

USB

### What is the expected behavior?

I have just installed according to the instructions of ESP IDF in Eclipse Version: 2022-09 (4.25.0). The build will go through without a problem. But it won't upload even with the basic flashing code.

### What is the actual behavior?

Not Work.

### Steps to reproduce.

Can't find any solution.

### Debug Logs.

```plain
Executing action: dfu-flash
Running ninja in directory /Users/davidjezek/eclipse-workspace/Gateway/build
Executing "ninja dfu-flash"...
[0/1] cd /Users/davidjezek/eclipse-workspace/Gateway/build && /Users/davidjezek/.espressif/tools/cmake/3.24.0/CMake.app/Contents/bin/cmake -D ESP_DFU_BIN="/Users/davidjezek/eclipse-workspace/Gateway/build/dfu.bin" -D ESP_DFU_PID="9" -P /Users/davidjezek/esp/esp-idf/tools/cmake/run_dfu_util.cmake
Command list: dfu-util;-d;303a:9;-D;/Users/davidjezek/eclipse-workspace/Gateway/build/dfu.bin
Done
```

dobairoland, 2023-01-30:
If you have an ESP32 then `dfu-flash` won't work. It requires native USB and it doesn't have it.

`idf.py flash` is the way to go for you.

pyrotechnik, 2023-01-31:
Thanks for the reply, unfortunately I can't find a place to change it. In the launch settings I have in the arguments: /Users/davidjezek/esp/esp-idf/tools/idf.py -p ${serial_port} flash

When I select esp32 as IDF target (instead of esp32s3) I get a message:
The selected target does not support the device firmware upgrade (DFU) through USB. Supported targets are esp32s2 and esp32s3.

AND

ninja: error: unknown target 'dfu-flash', did you mean 'app-flash'?
command "dfu-flash" is not known to idf.py and is not a Ninja target
Executing action: dfu-flash

dobairoland, 2023-01-31:
The error is expected because as I explained above, ESP32 doesn't have DFU capabilities.

Please follow the guide here: https://github.com/espressif/idf-eclipse-plugin#configuring-launch-target. Eclipse related issues can be reported in that repository. But please before opening a new issue, follow those steps carefully.

I'm closing this as we cannot handle this on the ESP-IDF side.

closed 2023-01-31 by dobairoland

igrr, 2023-02-01:
@pyrotechnik The issue you are facing is probably due to the DFU toggle button in the toolbar, please check this section of the docs: https://github.com/espressif/idf-eclipse-plugin#device-firmware-upgrade-dfu-through-usb. You need to click that button again to disable DFU flashing and then try to flash again.