espressif/esp-idf #10659: Docs and technical reference manual inconsistent for clock stretching timeout on ESP32-C3

TheButlah, 2023-01-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

In https://github.com/esp-rs/esp-hal/issues/352 I have been working on delving into esp-idf's implementation of clock stretching. 

* [`i2c_set_timeout()`](https://github.com/espressif/esp-idf/blob/8a7f6af6256f4c954053e08e819f55b93ddc1ff1/components/driver/include/driver/i2c.h#L587) specifies its delay in APB clock cycles (not exponential).
* It [calls]() `i2c_ll_set_tout()` which [states](https://github.com/espressif/esp-idf/blob/236fa5e6698a262aed534d2a46c78711204b903a/components/hal/esp32c3/include/hal/i2c_ll.h#L258-L269) that the timeout is `2**tout` APB clock cycles. This controls the `I2C_TIME_OUT_VALUE` parameter of the `I2C_TO_REG` register (0x0C). [modifies](https://github.com/espressif/esp-idf/blob/236fa5e6698a262aed534d2a46c78711204b903a/components/hal/esp32c3/include/hal/i2c_ll.h#L258-L269)
* The technical reference for esp32c3 states that `I2C_TIME_OUT_VALUE` describes the number of APB clock cycles (not exponential). Says this several places.

I am requesting clarification here:
1. `I2C_TIME_OUT_VALUE` linear or exponential? Is the technical reference correct that its linear?
2. If it is exponential, the docs for `i2c_set_timeout()` should be changed, and if its linear, the docs for `i2c_ll_set_tout()` should be changed.

TheButlah, 2023-06-13:
This issue has been marked as done - what was the resolution? @mythbuster5

mythbuster5, 2023-11-22:
Yeah.. the docs of `i2c_ll_set_tout` should be changed. It's linear.

TheButlah, 2023-12-11:
We still have contradicting evidence. According to [this]( https://github.com/esp-rs/esp-hal/issues/352#issuecomment-1703095118) experiment,  `i2c_ll_set_tout` and `I2C_TIME_OUT_VALUE` is exponential, and the TRM is wrong.

krzychb, 2025-02-03:
The description of `I2C_TIME_OUT_VALUE` in the ESP32-C3 TRM was updated in version 1.1 on January 19, 2024. 

Thank you @TheButlah!

closed 2025-02-03 by krzychb