espressif/esp-idf #10660: ESP32-D0WD-V3 doesn't send back SCAN_RESP packets.

zapta, 2023-01-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

Espressif IDF v1.5.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 WROOM  boards

### Power Supply used.

USB

### What is the expected behavior?

Device responds to BLE SCAN_REQ requests, when using the Espressif gatts_table_creat_demo.c example. 

### What is the actual behavior?

ESP32-D0WDQ6 (revision v1.0) works as expected.  ESP32-D0WD-V3 (revision v3.0) does not respond to BLE SCAN_REQ requests.  I am reporting here using your gatts_table_creat_demo.c example but experience the same issue with other BLE programs.

Attached are screen shots and packet captures files from wireshark.

[image]

[image]

[wireshark_ packets.zip](https://github.com/espressif/esp-idf/files/10539867/wireshark_.packets.zip)

### Steps to reproduce.

1. Build a run Espressif gatts_table_creat_demo.c example on a ESP32-D0WD-V3 device.
2. Use Nordic's nRF Connect app for android or iphone to scan for BLE devices. Notice that you see "ESP_GATTS_DEMO". Do not connect to it and leave the phone app scanning.
3. Run a BLE packet sniffer (I am using the Nordic sniffer with wireshark https://www.nordicsemi.com/Products/Development-tools/nrf-sniffer-for-bluetooth-le).
4. Observe in the sniffer traffic to/from the "ESP_GATTS_DEMO", identify the SCAN_REQ packets sent to it and observe if it sends back corresponding SCAN_RESP packets.

With the ESP32-D0WDQ6 (revision v1.0) device I do get the SCAN_REQ (good). With the ESP32-D0WD-V3 (revision v3.0) I don't. 

### Debug Logs.

```plain
This is the log when running the ESP32-D0WD-V3 (revision v3.0) which doesn't send back SCAN_RESP:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6992
load:0x40078000,len:15452
load:0x40080400,len:3840
0x40080400: _init at ??:?

entry 0x4008064c
I (27) boot: ESP-IDF v5.0 2nd stage bootloader
I (27) boot: compile time 10:43:07
I (27) boot: chip revision: v3.0
I (30) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 2MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (94) boot_comm: chip revision: 3, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=175a8h ( 95656) map
I (144) esp_image: segment 1: paddr=000275d0 vaddr=3ffbdb60 size=049a0h ( 18848) load
I (152) esp_image: segment 2: paddr=0002bf78 vaddr=40080000 size=040a0h ( 16544) load
I (159) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=5ea94h (387732) map 
I (299) esp_image: segment 4: paddr=0008eabc vaddr=400840a0 size=134f8h ( 79096) load
I (332) esp_image: segment 5: paddr=000a1fbc vaddr=50000000 size=00010h (    16) load
I (344) boot: Loaded app from partition at offset 0x10000
I (344) boot: Disabling RNG early entropy source...
I (356) cpu_start: Pro cpu up.
I (356) cpu_start: Starting app cpu, entry point is 0x400812f8
0x400812f8: call_start_cpu1 at C:/Users/user/esp/esp-idf/components/esp_system/port/cpu_start.c:142

I (0) cpu_start: App cpu up.
I (372) cpu_start: Pro cpu start user code
I (372) cpu_start: cpu freq: 160000000 Hz
I (372) cpu_start: Application information:
I (377) cpu_start: Project name:     esdidf_scan_resp
I (383) cpu_start: App version:      b5c7fbd-dirty
I (388) cpu_start: Compile time:     Jan 30 2023 12:26:07
I (394) cpu_start: ELF file SHA256:  3cf270b6c6e6f1e6...
I (400) cpu_start: ESP-IDF:          v5.0
I (405) heap_init: Initializing. RAM available for dynamic allocation:
I (412) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (418) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (424) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (430) heap_init: At 3FFC61D0 len 00019E30 (103 KiB): DRAM
I (437) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (443) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (450) heap_init: At 40097598 len 00008A68 (34 KiB): IRAM
I (457) spi_flash: detected chip: generic
I (460) spi_flash: flash io: dio
W (464) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.       
I (479) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (524) BTDM_INIT: BT controller compile version [8020d24]
I (524) system_api: Base MAC address is not set
I (524) system_api: read default base MAC address from EFUSE
I (534) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (934) GATTS_TABLE_DEMO: create attribute table successfully, the number handle = 8

I (944) GATTS_TABLE_DEMO: SERVICE_START_EVT, status 0, service_handle 40
I (954) GATTS_TABLE_DEMO: advertising start successfully
```

### More Information.

1. My test project is here https://github.com/zapta/espidf_scan_resp
2. This may be related to the reports here https://www.esp32.com/viewtopic.php?f=2&t=30483&p=109009#p109009

xiewenxiang, 2023-01-31:
ESP-IDF versions before V3.3 are already End of Life. could you please use the latest release branch to test?

https://github.com/espressif/esp-idf/blob/master/SUPPORT_POLICY.md

zapta, 2023-01-31:
@xiewenxiang, the bug report here is based on a fresh install of VSCode/ESPIDF and the gatts table create demo it offered me. Is there a newer version that a fresh install?  How do I upgrade to it in VSCode?

Here is a log of my build. It may give you information of the version I got from the fresh install.

```
 *  Executing task: cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B c:\projects\espidf_scan_resp\repo\build -S c:\projects\espidf_scan_resp\repo 

-- Found Git: C:/Users/user/.espressif/tools/idf-git/2.30.1/cmd/git.exe (found version "2.30.1.windows.1")
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/user/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/user/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/user/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features       
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file C:/projects/espidf_scan_resp/repo/sdkconfig
Loading defaults file C:/projects/espidf_scan_resp/repo/sdkconfig.defaults...
Loading defaults file C:/projects/espidf_scan_resp/repo/sdkconfig.defaults.esp32...
C:/projects/espidf_scan_resp/repo/sdkconfig.defaults.esp32:392 CONFIG_ESP32_PHY_CALIBRATION_AND_DATA_STORAGE was replaced with CONFIG_ESP_PHY_CALIBRATION_AND_DATA_STORAGE
C:/projects/espidf_scan_resp/repo/sdkconfig.defaults.esp32:393 CONFIG_ESP32_PHY_INIT_DATA_IN_PARTITION was replaced with CONFIG_ESP_PHY_INIT_DATA_IN_PARTITION       
C:/projects/espidf_scan_resp/repo/sdkconfig.defaults.esp32:394 CONFIG_ESP32_PHY_MAX_WIFI_TX_POWER was replaced with CONFIG_ESP_PHY_MAX_WIFI_TX_POWER
C:/projects/espidf_scan_resp/repo/sdkconfig.defaults.esp32:395 CONFIG_ESP32_PHY_MAX_TX_POWER was replaced with CONFIG_ESP_PHY_MAX_TX_POWER
C:/projects/espidf_scan_resp/repo/sdkconfig.defaults.esp32:1077 CONFIG_OTA_ALLOW_HTTP was replaced with CONFIG_ESP_HTTPS_OTA_ALLOW_HTTP
C:/projects/espidf_scan_resp/repo/sdkconfig.defaults.esp32:1100 CONFIG_ESP32_DEBUG_OCDAWARE was replaced with CONFIG_ESP_DEBUG_OCDAWARE
C:/projects/espidf_scan_resp/repo/sdkconfig.defaults.esp32:1101 CONFIG_ESP32_BROWNOUT_DET was replaced with CONFIG_ESP_BROWNOUT_DET
C:/projects/espidf_scan_resp/repo/sdkconfig.defaults.esp32:1102 CONFIG_ESP32_BROWNOUT_DET_LVL_SEL_0 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_0
C:/projects/espidf_scan_resp/repo/sdkconfig.defaults.esp32:1103 CONFIG_ESP32_BROWNOUT_DET_LVL_SEL_1 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_1
C:/projects/espidf_scan_resp/repo/sdkconfig.defaults.esp32:1104 CONFIG_ESP32_BROWNOUT_DET_LVL_SEL_2 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_2
C:/projects/espidf_scan_resp/repo/sdkconfig.defaults.esp32:1105 CONFIG_ESP32_BROWNOUT_DET_LVL_SEL_3 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_3
C:/projects/espidf_scan_resp/repo/sdkconfig.defaults.esp32:1106 CONFIG_ESP32_BROWNOUT_DET_LVL_SEL_4 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_4
C:/projects/espidf_scan_resp/repo/sdkconfig.defaults.esp32:1107 CONFIG_ESP32_BROWNOUT_DET_LVL_SEL_5 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_5
C:/projects/espidf_scan_resp/repo/sdkconfig.defaults.esp32:1108 CONFIG_ESP32_BROWNOUT_DET_LVL_SEL_6 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_6
C:/projects/espidf_scan_resp/repo/sdkconfig.defaults.esp32:1109 CONFIG_ESP32_BROWNOUT_DET_LVL_SEL_7 was replaced with CONFIG_ESP_BROWNOUT_DET_LVL_SEL_7
C:/projects/espidf_scan_resp/repo/sdkconfig.defaults.esp32:1110 CONFIG_ESP32_BROWNOUT_DET_LVL was replaced with CONFIG_ESP_BROWNOUT_DET_LVL
C:/projects/espidf_scan_resp/repo/sdkconfig.defaults.esp32:1111 CONFIG_ESP32_REDUCE_PHY_TX_POWER was replaced with CONFIG_ESP_PHY_REDUCE_TX_POWER
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: C:/Users/user/.espressif/python_env/idf5.0_py3.8_env/Scripts/python.exe (found version "3.8.7") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "esdidf_scan_resp" version: b5c7fbd-dirty
-- Adding linker script C:/projects/espidf_scan_resp/repo/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Users/user/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script C:/Users/user/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/user/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script C:/Users/user/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/user/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
[... cut ...] app partition is 0x100000 bytes. 0x6e010 bytes (43%) free.

 *  Executing task: C:\Users\user\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe C:\Users\user\esp\esp-idf\components\esptool_py\esptool\esptool.py -p COM10 -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x1000 bootloader/bootloader.bin 0x10000 esdidf_scan_resp.bin 0x8000 partition_table/partition-table.bin 

esptool.py v4.4
Serial port COM10
```

xiewenxiang, 2023-02-01:
@zapta It doesn't look like Espressif IDF v1.5.1. Can you provide the exact commit information for ESP-IDF? Or provide the app.bin file you compiled.

xiewenxiang, 2023-02-01:
@zapta 
I directly used the bin file you provided and made corresponding tests, and it seems that there is no problem.
[test_log_20230201.txt](https://github.com/espressif/esp-idf/files/10552778/test_log_20230201.txt)

xiewenxiang, 2023-02-01:
[espidf_scan_resp.zip](https://github.com/espressif/esp-idf/files/10552807/espidf_scan_resp.zip)

xiewenxiang, 2023-02-01:
[image]

The hardware information is the same

zapta, 2023-02-01:
Thanks @xiewenxiang, I will at your results and try to identify differences in the configurations.  Will also try to order ESP32 modules from other sources to compare.

Is there a free viewer to look at the capture file you sent?  I am using wireshark which is free but am not sure if it can open those files.

xiewenxiang, 2023-02-01:
@zapta 

https://teledynelecroy.com/support/softwaredownload/psgdocuments.aspx?standardid=2&mseries=671

zapta, 2023-02-04:
Thanks @xiewenxiang. I will try to debug it here. Can you give me a hint where the SCAN_REQ is handled in the ESP32 Bluddroid stack? E.g. where I can breakpoint and see if it arrived or not and where the decision to send back the SCAN_RSP is done.   I looked at the code and found events and function related to scan rsp but there are all for BLE 5.0 which is not enabled for the ESP32.

https://github.com/espressif/esp-idf/tree/master/components/bt/host/bluedroid

webmonkey, 2023-03-09:
@zapta I have been trying to work out a similar problem that was exhibited with a ESP32-D0WD-V3 but worked fine with ESP32-D0WDQ6 chip. Interestingly I am having problems with exactly the same type of "D1 Mini" style ESP32 board.

zapta, 2023-03-13:
> i have actualy the same problem and there is no clue i can find on
> internet ,D0WDQ6 work perfectly but D0WD-V3 wont work with ble
> i work with 2 similar uControler from the same order(
> https://www.amazon.fr/AZ-Delivery-NodeMCU-développement-dénergie-successeur/dp/B071P98VTG/ref=sxin_16_ci_mcx_mi_sr_m_il?__mk_fr_FR=ÅMÅŽÕÑ&content-id=amzn1.sym.21edd561-171c-4ac1-9264-688591dfe037%3Aamzn1.sym.21edd561-171c-4ac1-9264-688591dfe037&crid=1TDG8VFWHHCJM&cv_ct_cx=esp32+dev+c&keywords=esp32+dev+c&pd_rd_i=B071P98VTG&pd_rd_r=84561d0a-a525-47c2-9288-4ef63863e097&pd_rd_w=CWQt3&pd_rd_wg=8key4&pf_rd_p=21edd561-171c-4ac1-9264-688591dfe037&pf_rd_r=7C1Q090BKCAK0H03DXG7&qid=1678713442&sbo=EO%2BcdOgoisx3waG5xqcZQA%3D%3D&sprefix=esp32+dev+c%2Caps%2C81&sr=1-3-b3105280-ed1c-4fae-986f-6c4501b9525e
> <https://www.amazon.fr/AZ-Delivery-NodeMCU-d%C3%A9veloppement-d%C3%A9nergie-successeur/dp/B071P98VTG/ref=sxin_16_ci_mcx_mi_sr_m_il?__mk_fr_FR=%C3%85M%C3%85%C5%BD%C3%95%C3%91&content-id=amzn1.sym.21edd561-171c-4ac1-9264-688591dfe037%3Aamzn1.sym.21edd561-171c-4ac1-9264-688591dfe037&crid=1TDG8VFWHHCJM&cv_ct_cx=esp32+dev+c&keywords=esp32+dev+c&pd_rd_i=B071P98VTG&pd_rd_r=84561d0a-a525-47c2-9288-4ef63863e097&pd_rd_w=CWQt3&pd_rd_wg=8key4&pf_rd_p=21edd561-171c-4ac1-9264-688591dfe037&pf_rd_r=7C1Q090BKCAK0H03DXG7&qid=1678713442&sbo=EO%2BcdOgoisx3waG5xqcZQA%3D%3D&sprefix=esp32+dev+c%2Caps%2C81&sr=1-3-b3105280-ed1c-4fae-986f-6c4501b9525e>
> )

I can't say with high confidence that I don't have much information, but it
seems that when I use actual Espressif modules, I don't encounter the
problem. Again, I can't say it for sure and couldn't identify differences
in fuses or others between Espressif and non Espressif modules.

This is what I used for the PCB I designed:
https://jlcpcb.com/partdetail/EspressifSystems-ESP32_WROOM_32UN4/C328062

pablopeu, 2023-05-13:
Adding another case to the thread, ESP32-D0WD-V3 chip fails and ESP32-D0WDQ6(rev1) works as expected. Simple arduino app for emulating a PC keyboard using the https://github.com/T-vK/ESP32-BLE-Keyboard library.

soufiane707, 2024-04-21:
another case of ESP32-D0WD-V3 failing to connect with my ps5 dual sense controller via bluetooth but other esp32's worked

zhp0406, 2024-04-22:
@soufiane707 
This seems to be a new issue. Could you create a new GitHub issue and provide the complete log?