espressif/esp-idf #10661: Make Use Of "cmake-tools-kits.json" Or CMake Environment Variables In "tools/cmake/toolchain-<target>.cmake"

vindicatorr, 2023-01-31:
### Is your feature request related to a problem?

I don't know for sure, but it MAY be the case that `cmake-tools-kits.json` is a vscode cmake-extensions-only thing. It's sure looking like it, but the `CMAKE_*_COMPILER` values can be set manually if desired.

Instead of hardcoding `set(CMAKE_C_COMPILER riscv32-esp-elf-gcc)` in `tools/cmake/toolchain-esp32c3.cmake`, I'd like to be able to use the toolchain I select from `~/.local/share/CMakeTools/cmake-tools-kits.json`, or pass as variables to the `cmake` command or wherever else.

```
tools/cmake/project.cmake:329 (__target_set_toolchain()) ->
tools/cmake/targets.cmake:76 (set(toolchain_file_global ${idf_path}/tools/cmake/toolchain-${toolchain_type}${IDF_TARGET}.cmake)) ->
tools/cmake/toolchain-esp32c3.cmake:5 (
set(CMAKE_C_COMPILER riscv32-esp-elf-gcc)
set(CMAKE_CXX_COMPILER riscv32-esp-elf-g++)
set(CMAKE_ASM_COMPILER riscv32-esp-elf-gcc)
set(_CMAKE_TOOLCHAIN_PREFIX riscv32-esp-elf-)
)
```

### Describe the solution you'd like.

```
if (NOT CMAKE_C_COMPILER)
set(CMAKE_C_COMPILER riscv32-esp-elf-gcc)
endif()
if (NOT CMAKE_CXX_COMPILER)
set(CMAKE_CXX_COMPILER riscv32-esp-elf-g++)
endif()
if (NOT CMAKE_ASM_COMPILER)
set(CMAKE_ASM_COMPILER riscv32-esp-elf-gcc)
endif()
```

I don't know where `_CMAKE_TOOLCHAIN_PREFIX` is used, so not sure how to handle that, but I'd figure use whatever `cmake` offers to take a substring of `CMAKE_ASM_COMPILER`.

igrr, 2023-02-01:
@vindicatorr could you please describe the goal you would like to achieve by changing the compiler name?

vindicatorr, 2023-02-01:
One example is that I could roll my own toolchain. I know I could name it myself to use the `-esp-` identifier.
But let's say I'd want to use someone else's premade toolchain. Maybe Zephyr would use their own identifier `-zephyr-`.

igrr, 2023-02-01:
In general a toolchain built for Zephyr will not work for ESP-IDF, since we expect certain specific features in the toolchain:

- newlib should be configured with retargetable locking enabled
- GCC should be built with --thread-model=thread
- and a few others

You can definitely build the toolchain yourself. However in this case, same question: what are you trying to achieve? If it is to be in control of the process, you should still be able to use the same prefix as the one expected by IDF. If you intend to build a _different_ toolchain, then it's not something that we can guarantee will work in IDF, and hence not something we would like to actively support.

I'm sorry that this is probably not the answer you expect. If you explain the original problem you are trying to solve, I will try to give a more helpful answer.

vindicatorr, 2023-02-01:
Yeah, I was/am rolling my own toolchain for "reasons".
I've even rolled your toolchain using your version of `crosstool-NG`, and it was fine.

But rolling my own from native master upstreams (and encountering issues along the way, and dealing with them one-by-one, as I'm sure you all have had to do), I noticed this "restriction"(?), thus my mention, and the included solution.

But it's fine if you'd rather only allow toolchains that include the `-esp-` identifier, as it is your entire project, so I'll close this.

closed 2023-02-01 by vindicatorr

igrr, 2023-02-02:
> would be an all-encompassing toolchain for any riscv chip I'd come across

It's a nice goal, and I agree it would be best if everyone and every project didn't have to roll their own toolchain! However i am not sure how realistic this is, at least for environments which have non-trivial expectations about the C and C++ standard libraries. It's probably viable to have a fairly universal toolchain for bare metal projects — it would just be quite large due to the number of -march=... variants of libc (i.e. multilib). That's basically the toolchain available from the xPack project.

But as soon as you get into multithreaded environments, thread safety or reentrancy support typically requires libc and libstdc++ to be built in quite specific ways. Unfortunately, in GCC it's not possible (yet) to dispatch between single-threaded and multi-threaded versions of libc dynamically (by having yet another dimension in the multilib matrix). Add to that the fact that there are at least two mutually exclusive ways to configure newlib for reentrancy.

Overall I think that sadly, we are still stuck with having environment-specific toolchains for a while.

vindicatorr, 2023-02-02:
I'm sure you are MUCH more knowledgeable about this stuff than myself, so I can't speak from a grounded base, like I expect you ought to be considering you all are writing in these patches.

I had gotten the riscv toolchain from Arch Linux sometime last year I think and if I remember right, it failed due to something about not having rv32imc built-in.

Yet at the same time, I've been able to use the arm toolchain distributed by Arch Linux for 3 completely different boards (vendors: ST, Hardkernel, NordicSemi).

It may be the case I simply haven't encountered a limitation of the distribution release with any of the devices because my needs never reached whatever would be missing from the toolchain.