espressif/esp-idf #10662: ESP32-C6 Unable to join zigbee network

lbuque, 2023-01-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-3066-g236fa5e669

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

I have two boards, run HA_on_off_light and HA_on_off_switch respectively, and then I use zigbee sniffer to get their incoming packets, but I can't get anything.

### What is the actual behavior?

I can't get the Beacon Request message when I use Ubiqua.

[image]

### Steps to reproduce.

1. cd examples/zigbee/light_sample/HA_on_off_switch
2. idf.py --preview set-target esp32c6
3. idf.py build
4. idf.py flash
5. cd examples/zigbee/light_sample/HA_on_off_light
6. idf.py --preview set-target esp32c6
7. idf.py build
8. idf.py flash

### Debug Logs.

```plain
Executing action: monitor
Running idf_monitor in directory /home/lewin/esp/esp-idf-v5.0/examples/zigbee/light_sample/HA_on_off_switch
Executing "/home/lewin/.espressif/python_env/idf5.1_py3.8_env/bin/python /home/lewin/esp/esp-idf-v5.0/tools/idf_monitor.py -p /dev/ttyUSB1 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32c6 --decode-panic backtrace /home/lewin/esp/esp-idf-v5.0/examples/zigbee/light_sample/HA_on_off_switch/build/light_switch.elf -m '/home/lewin/.espressif/python_env/idf5.1_py3.8_env/bin/python' '/home/lewin/esp/esp-idf-v5.0/tools/idf.py' '-p' '/dev/ttyUSB1'"...
--- idf_monitor on /dev/ttyUSB1 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c610,len:0xc98
load:0x4086e610,len:0x2a0c
load:0x40875720,len:0x17b0
entry 0x4086c610
I (23) boot: ESP-IDF v5.1-dev-3066-g236fa5e669 2nd stage bootloader
I (24) boot: compile time Jan 31 2023 15:42:51
I (24) boot: chip revision: v0.0
I (28) boot.esp32c6: SPI Speed      : 40MHz
I (33) boot.esp32c6: SPI Mode       : DIO
I (38) boot.esp32c6: SPI Flash Size : 2MB
I (42) boot: Enabling RNG early entropy source...
W (48) bootloader_random: bootloader_random_enable() has not been implemented yet
I (56) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00089800
I (89) boot:  3 zb_storage       Unknown data     01 81 0009a000 00020000
I (97) boot:  4 zb_fct           Unknown data     01 81 000ba000 00000400
I (104) boot: End of partition table
I (108) esp_image: segment 0: paddr=00010020 vaddr=42060020 size=0a250h ( 41552) map
I (125) esp_image: segment 1: paddr=0001a278 vaddr=40800000 size=05da0h ( 23968) load
I (131) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=5cbb0h (379824) map
I (206) esp_image: segment 3: paddr=0007cbd8 vaddr=40805da0 size=06008h ( 24584) load
I (212) esp_image: segment 4: paddr=00082be8 vaddr=4080bdb0 size=01538h (  5432) load
I (217) boot: Loaded app from partition at offset 0x10000
I (218) boot: Disabling RNG early entropy source...
W (224) bootloader_random: bootloader_random_enable() has not been implemented yet
I (243) cpu_start: Pro cpu up.
W (252) clk: esp_perip_clk_init() has not been implemented yet
W (258) CACHE_ERR: esp_cache_err_int_init() has not been implemented yet
I (259) cpu_start: Pro cpu start user code
I (259) cpu_start: cpu freq: 160000000 Hz
I (264) cpu_start: Application information:
I (269) cpu_start: Project name:     light_switch
I (274) cpu_start: App version:      v5.1-dev-3066-g236fa5e669
I (281) cpu_start: Compile time:     Jan 31 2023 15:42:44
I (287) cpu_start: ELF file SHA256:  6a2fee3c1530ebe6...
I (293) cpu_start: ESP-IDF:          v5.1-dev-3066-g236fa5e669
I (299) cpu_start: Min chip rev:     v0.0
I (304) cpu_start: Max chip rev:     v0.99 
I (309) cpu_start: Chip rev:         v0.0
I (314) heap_init: Initializing. RAM available for dynamic allocation:
I (321) heap_init: At 40812E10 len 00069800 (422 KiB): D/IRAM
I (327) heap_init: At 4087C610 len 00002F54 (11 KiB): STACK/DRAM
I (334) heap_init: At 50000010 len 00003FF0 (15 KiB): RTCRAM
I (341) spi_flash: detected chip: gd
I (345) spi_flash: flash io: dio
W (348) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (362) app_start: Starting scheduler on CPU0
I (367) main_task: Started on CPU0
I (367) main_task: Calling app_main()
I (377) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (377) phy_init: phy_version 101,587adde,Dec 20 2022,14:21:21
I (457) main_task: Returned from app_main()
I (467) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 23, status: -1
I (467) ESP_ZB_ON_OFF_SWITCH: Zigbee stack initialized
I (497) ESP_ZB_ON_OFF_SWITCH: Start network formation
I (42757) ESP_ZB_ON_OFF_SWITCH: send 'on_off toggle' command
I (42757) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 60, status: 0
I (43297) ESP_ZB_ON_OFF_SWITCH: send 'on_off toggle' command
I (43297) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 60, status: 0
I (44187) ESP_ZB_ON_OFF_SWITCH: send 'on_off toggle' command
I (44187) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 60, status: 0
I (44847) ESP_ZB_ON_OFF_SWITCH: send 'on_off toggle' command
I (44847) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 60, status: 0
I (45167) ESP_ZB_ON_OFF_SWITCH: send 'on_off toggle' command
I (45167) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 60, status: 0
I (45417) ESP_ZB_ON_OFF_SWITCH: send 'on_off toggle' command
I (45417) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 60, status: 0
I (45717) ESP_ZB_ON_OFF_SWITCH: send 'on_off toggle' command
I (45717) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 60, status: 0
I (45977) ESP_ZB_ON_OFF_SWITCH: send 'on_off toggle' command
I (45977) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 60, status: 0
I (46247) ESP_ZB_ON_OFF_SWITCH: send 'on_off toggle' command
I (46247) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 60, status: 0
I (46507) ESP_ZB_ON_OFF_SWITCH: send 'on_off toggle' command
I (46507) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 60, status: 0
I (46827) ESP_ZB_ON_OFF_SWITCH: send 'on_off toggle' command
I (46827) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 60, status: 0
I (48077) ESP_ZB_ON_OFF_SWITCH: send 'on_off toggle' command
I (48077) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 60, status: 0
I (50417) ESP_ZB_ON_OFF_SWITCH: send 'on_off toggle' command
I (50417) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 60, status: 0
I (50677) ESP_ZB_ON_OFF_SWITCH: send 'on_off toggle' command
I (50677) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 60, status: 0
I (50937) ESP_ZB_ON_OFF_SWITCH: send 'on_off toggle' command
I (50937) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 60, status: 0
```

chshu, 2023-02-01:
@lbuque Could you try again with the latest IDF master, we just fixed a bug in commit: https://github.com/espressif/esp-idf/commit/897b9700c41fed603c61cadc205cc11f972ee7b4

lbuque, 2023-02-01:
@chshu Using the commit you suggested, I tried the HA_on_off_light example.
Still can't catch any messages in ubiqua.

Could it be that the Beacon Request is only initiated for certain channels?
If yes, which channels are used in the code?
I simply checked the code, but I didn't see the channel information.

likunqiao097304, 2023-02-01:
@lbuque the default channel is 13.  If you want to change to different one, please refer to the esp_zigbee_core.h:  esp_zb_set_network_channel(uint8_t channel);

lbuque, 2023-02-01:
@likunqiao097304 It works! ! !

Make two suggestions:
1. The light device should be used as a router (this should follow the device specification, it would be better)
2. When the device initiates network steering, it should initiate a Beacon Request to the primary channel and the secondary channel in sequence. (These should follow the bdb specification)

likunqiao097304, 2023-02-01:
@lbuque  zigbee light example is not a router is for the demonstration of zigbee-end device purpose. 
primary channel and secondary channel by default make them the same. So does your suggestion mean that on the monitor log side to show which channel we are sending beacon request to? both on primary channel and secondary channel?

lbuque, 2023-02-01:
@likunqiao097304 As a sub-device, it sends a Beacon Request to each channel during Network steering to find a suitable zigbee network to join as soon as possible.

Just like the Network steering procedure in the figure below:

[image]

I didn't find an api to set vScanChannels or bdbPrimaryChannelSet.

u236, 2023-02-01:
Same problem using `v5.1-dev-3066-g236fa5e669` but network successfully formed and joined with `v5.1-dev-3101-g1195b6cb2bz`, thanks for the fix!

likunqiao097304, 2023-02-01:
> bdbPrimaryChannelSet

@lbuque This api hasn't provided yet. We will add them in the next round release

momo1, 2023-02-06:
I am having a similar issue with the example. I flashed `HA_on_off_switch` and `HA_on_off_light` on two ESP32 C6s. I am using IDF `v5.1-dev-3213-gd29e53dc0c` on both and I also tried multiple older versions. I've erased NVRAM on both and clean built before each trial. The coordinator seems to work fine:

```
I (365) main_task: Started on CPU0
I (365) main_task: Calling app_main()
I (375) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (375) phy_init: phy_version 101,061a19e,Jan 11 2023,18:48:18
I (455) main_task: Returned from app_main()
I (465) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 23, status: -1
I (465) ESP_ZB_ON_OFF_SWITCH: Zigbee stack initialized
I (465) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 54, status: 0
I (465) ESP_ZB_ON_OFF_SWITCH: Start network formation
I (475) ESP_ZB_ON_OFF_SWITCH: Formed network successfully (Extended PAN ID: 00:00:94:21:f6:f9:55:60, PAN ID: 0x6d44)
I (945) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 54, status: 0
I (945) ESP_ZB_ON_OFF_SWITCH: Network steering started
I (1565) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 18, status: 0
I (2075) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 48, status: 0
I (3655) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 18, status: 0
I (7405) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 18, status: 0
I (7915) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 48, status: 0
I (9495) ESP_ZB_ON_OFF_SWITCH: ZDO signal: 18, status: 0
```

But the light end device fails to join the network:

```
I (345) main_task: Started on CPU0
I (345) main_task: Calling app_main()
I (355) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (355) phy_init: phy_version 101,061a19e,Jan 11 2023,18:48:18
W (365) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (455) main_task: Returned from app_main()
I (635) ESP_ZB_ON_OFF_LIGHT: ZDO signal: 23, status: -1
I (635) ESP_ZB_ON_OFF_LIGHT: Zigbee stack initialized
I (635) ESP_ZB_ON_OFF_LIGHT: Start network steering
I (3295) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: -1)
I (6965) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: -1)
I (10635) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: -1)
I (14305) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: -1)
I (17975) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: -1)
I (21645) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: -1)
I (25315) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: -1)
```

Thanks

lbuque, 2023-02-06:
@momo1 It is recommended that you buy a zigbee sniffer, which can capture zigbee data packets, making it easier to locate problems.

likunqiao097304, 2023-02-06:
> v5.1-dev-3213-gd29e53dc0c

I used the same commit of esp-idf. It works fine on both examples from my side.  From the log you provided, it did look like failed to join the network. Some suggestions you could try, firstly when you checkout to different esp-idf commit, do a `git submodule update --init --recursive`. Secondly, make sure you do erase-flash on the both examples.

Could you checkout what version of the esp-zboss-lib and esp-zigbee-lib? It is under each examples/management_components a file called idf_component.yml. They should be  0.3.1 and 0.5.0.

momo1, 2023-02-06:
@likunqiao097304 Thank you for the quick response! It worked!

I had already erased the flash and recursively updated all git modules before posting but the thing that I missed was that the library versions changed across commits in the `main/idf_component.yml`. Both zboss and zigbee libs were at a lower version. Once I corrected that it worked. Nice catch and thanks again.

likunqiao097304, 2023-02-06:
> @likunqiao097304 Thank you for the quick response! It worked!
>
> I had already erased the flash and recursively updated all git modules before posting but the thing that I missed was that the library [...]

Good to hear that. Again, the Zigbee sdk is still under 1.0.0 version, there are still some known / unknown issues need to fix from our side, so feel free to give us more feedback. Also you could check this repo for more example:[esp-zigbee-sdk](https://github.com/espressif/esp-zigbee-sdk)

bouvrais, 2023-02-09:
> > bdbPrimaryChannelSet
> 
> @lbuque This api hasn't provided yet. We will add them in the next round release

@likunqiao097304 therefore I suppose there is currently no way to connect an ESP32 End Device to a standard Coordinator network (i.e. a network that is not created by another ESP32)?
Is the a list of features yet to be implemented I could look at?

swkim01, 2023-02-09:
@likunqiao097304 Thank you!

After setting the channel number, I could make the device to be registered to home assistant via zigbee2mqtt addon with zigbee 3.0 usb dongle.

I used HA_on_off_switch example that was modified to include manufname and modelid attributes in basic cluster, refered by [this custom example](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_customized_devices/customized_server/main/esp_HA_customized_light.c).
```
...
char modelid[] = {13, 'E', 'S', 'P', '3', '2', 'C', '6', '.', 'L', 'i', 'g', 'h', 't'};
char manufname[] = {9, 'E', 's', 'p', 'r', 'e', 's', 's', 'i', 'f'};
...
    esp_zb_init(&zb_nwk_cfg);
    esp_zb_set_network_channel(23);
...
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID, &modelid[0]);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, &manufname[0]);
...
```

Then, I added the following switch converter to HA repository as to [this procedure](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#instructions).

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ESP32C6.Light'],
    model: 'ESP32C6.Light',
    vendor: 'Espressif',
    description: 'light_bulb',
    fromZigbee: [fz.on_off],
    toZigbee: [tz.on_off],
    exposes: [e.switch()],
    // The configure method below is needed to make the device reports on/off state changes
    // when the device is controlled manually through the button on it.
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
    },
};

module.exports = definition;
```
screenshot of z2m device list on HA.
[image]
screenshot of z2m dashboard on HA.
[image]

lbuque, 2023-02-10:
@swkim01 Can you show your code in detail?

According to your modification, I can't get the value of `ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID` using read attrib.

[image]

swkim01, 2023-02-10:
@lbuque My esp_zb_task function is as follows. That's all.
'''
static void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack with Zigbee end-device config */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
    esp_zb_init(&zb_nwk_cfg);
    esp_zb_set_network_channel(23);

    /* set the on-off light device config */
    uint8_t test_attr, test_attr2;
 
    test_attr = 0;
    test_attr2 = 4;
    /* basic cluster create with fully customized */
    esp_zb_attribute_list_t *esp_zb_basic_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_BASIC);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID, &test_attr);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_POWER_SOURCE_ID, &test_attr);
    esp_zb_cluster_update_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID, &test_attr2);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID, &modelid[0]);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, &manufname[0]);
    /* identify cluster create with fully customized */
    esp_zb_attribute_list_t *esp_zb_identify_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_IDENTIFY);
    esp_zb_identify_cluster_add_attr(esp_zb_identify_cluster, ESP_ZB_ZCL_ATTR_IDENTIFY_IDENTIFY_TIME_ID, &test_attr);
    /* group cluster create with fully customized */
    esp_zb_attribute_list_t *esp_zb_groups_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_GROUPS);
    esp_zb_groups_cluster_add_attr(esp_zb_groups_cluster, ESP_ZB_ZCL_ATTR_GROUPS_NAME_SUPPORT_ID, &test_attr);
    /* scenes cluster create with standard cluster + customized */
    esp_zb_attribute_list_t *esp_zb_scenes_cluster = esp_zb_scenes_cluster_create(NULL);
    esp_zb_cluster_update_attr(esp_zb_scenes_cluster, ESP_ZB_ZCL_ATTR_SCENES_NAME_SUPPORT_ID, &test_attr);
    /* on-off cluster create with standard cluster config*/
    esp_zb_on_off_cluster_cfg_t on_off_cfg;
    on_off_cfg.on_off = ESP_ZB_ZCL_ON_OFF_ON_OFF_DEFAULT_VALUE;
    esp_zb_attribute_list_t *esp_zb_on_off_cluster = esp_zb_on_off_cluster_create(&on_off_cfg);
    /* create cluster lists for this endpoint */
    esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
    esp_zb_cluster_list_add_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    /* update basic cluster in the existed cluster list */
    //esp_zb_cluster_list_update_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster_create(NULL), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_identify_cluster(esp_zb_cluster_list, esp_zb_identify_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_groups_cluster(esp_zb_cluster_list, esp_zb_groups_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_scenes_cluster(esp_zb_cluster_list, esp_zb_scenes_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_on_off_cluster(esp_zb_cluster_list, esp_zb_on_off_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

    esp_zb_ep_list_t *esp_zb_ep_list = esp_zb_ep_list_create();
    /* add created endpoint (cluster_list) to endpoint list */
    esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, HA_ESP_LIGHT_ENDPOINT, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_OUTPUT_DEVICE_ID);
    esp_zb_device_register(esp_zb_ep_list);
    esp_zb_device_add_set_attr_value_cb(attr_cb);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}
'''

likunqiao097304, 2023-02-10:
> @likunqiao097304 therefore I suppose there is currently no way to connect an ESP32 End Device to a standard Coordinator network (i.e. a network that is not created by another ESP32)? Is the a list [...]

I think you can, by calling `esp_zb_set_network_channel`, you could switch to other channel that your Zigbee coordinator formed. Just a note here, we will provide `esp_zb_set_primary_network_channel_set` and `esp_zb_set_secondary_network_channel_set` API pretty soon.

bouvrais, 2023-02-10:
Thank you @likunqiao097304, I can confirm that forcing the channel to the one used by the coordinator does indeed allow joining the network.
Also @swkim01's example works sufficiently well on my end by changing the channel.

likunqiao097304, 2023-02-13:
> > bdbPrimaryChannelSet
> 
> @lbuque This api hasn't provided yet. We will add them in the next round release

@lbuque The API for primary/secondary channel set has been provided in the latest version of lib file. Please take a look and close the issue if fixed it

lbuque, 2023-02-13:
@likunqiao097304 thank you for your support! ! ! It works.

closed 2023-02-13 by lbuque

pcurt, 2023-11-30:
Does someone test this following code with the last esp-idf version (ESP-IDF) 5.1.2 ? 

```
...
char modelid[] = {13, 'E', 'S', 'P', '3', '2', 'C', '6', '.', 'L', 'i', 'g', 'h', 't'};
char manufname[] = {9, 'E', 's', 'p', 'r', 'e', 's', 's', 'i', 'f'};
...
    esp_zb_init(&zb_nwk_cfg);
    esp_zb_set_network_channel(23);
...
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID, &modelid[0]);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, &manufname[0]);
... 
```

It uses the following versions for zigbee and zboss libraries
```
  esp-zigbee-lib: "0.9.0"
  esp-zboss-lib: "0.7.0"
```

With 5.1.2 esp-idf, Zigbee model and manufacturer name are set to 'undefined'
It was working with previous version  5.1.1 (esp-zigbee-lib: 0.7.0,  espressif/esp-zboss-lib: 0.5.0)
So I don't know who to set model and manufacturer with last esp-idf.

Tokn59, 2023-12-16:
> char modelid[] = {13, 'E', 'S', 'P', '3', '2', 'C', '6', '.', 'L', 'i', 'g', 'h', 't'};
> char manufname[] = {9, 'E', 's', 'p', 'r', 'e', 's', 's', 'i', 'f'};

Could you please share your full example or indicate clearly where you defined and added these in the source code ? Thank you !!!!

swkim01, 2023-12-17:
@Tokn59 
Within [esp_zb_light.c](https://github.com/espressif/esp-zigbee-sdk/blob/main/examples/esp_zigbee_HA_sample/HA_on_off_light/main/esp_zb_light.c) of HA_on_off_light example, replace esp_zb_task function with the following code, that's all.
```
char modelid[] = {13, 'E', 'S', 'P', '3', '2', 'C', '6', '.', 'L', 'i', 'g', 'h', 't'};
char manufname[] = {9, 'E', 's', 'p', 'r', 'e', 's', 's', 'i', 'f'};

static void esp_zb_task(void *pvParameters)
{
    /* initialize Zigbee stack with Zigbee end-device config */
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
    esp_zb_init(&zb_nwk_cfg);
    esp_zb_set_primary_network_channel_set(ESP_ZB_PRIMARY_CHANNEL_MASK);

    /* set the on-off light device config */
    uint8_t test_attr, test_attr2;
 
    test_attr = 0;
    test_attr2 = 4;
    /* basic cluster create with fully customized */
    esp_zb_attribute_list_t *esp_zb_basic_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_BASIC);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID, &test_attr);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_POWER_SOURCE_ID, &test_attr2);
    esp_zb_cluster_update_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_ZCL_VERSION_ID, &test_attr2);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MODEL_IDENTIFIER_ID, &modelid[0]);
    esp_zb_basic_cluster_add_attr(esp_zb_basic_cluster, ESP_ZB_ZCL_ATTR_BASIC_MANUFACTURER_NAME_ID, &manufname[0]);
    /* identify cluster create with fully customized */
    esp_zb_attribute_list_t *esp_zb_identify_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_IDENTIFY);
    esp_zb_identify_cluster_add_attr(esp_zb_identify_cluster, ESP_ZB_ZCL_ATTR_IDENTIFY_IDENTIFY_TIME_ID, &test_attr);
    /* group cluster create with fully customized */
    esp_zb_attribute_list_t *esp_zb_groups_cluster = esp_zb_zcl_attr_list_create(ESP_ZB_ZCL_CLUSTER_ID_GROUPS);
    esp_zb_groups_cluster_add_attr(esp_zb_groups_cluster, ESP_ZB_ZCL_ATTR_GROUPS_NAME_SUPPORT_ID, &test_attr);
    /* scenes cluster create with standard cluster + customized */
    esp_zb_attribute_list_t *esp_zb_scenes_cluster = esp_zb_scenes_cluster_create(NULL);
    esp_zb_cluster_update_attr(esp_zb_scenes_cluster, ESP_ZB_ZCL_ATTR_SCENES_NAME_SUPPORT_ID, &test_attr);
    /* on-off cluster create with standard cluster config*/
    esp_zb_on_off_cluster_cfg_t on_off_cfg;
    on_off_cfg.on_off = ESP_ZB_ZCL_ON_OFF_ON_OFF_DEFAULT_VALUE;
    esp_zb_attribute_list_t *esp_zb_on_off_cluster = esp_zb_on_off_cluster_create(&on_off_cfg);
    /* create cluster lists for this endpoint */
    esp_zb_cluster_list_t *esp_zb_cluster_list = esp_zb_zcl_cluster_list_create();
    esp_zb_cluster_list_add_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    /* update basic cluster in the existed cluster list */
    //esp_zb_cluster_list_update_basic_cluster(esp_zb_cluster_list, esp_zb_basic_cluster_create(NULL), ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_identify_cluster(esp_zb_cluster_list, esp_zb_identify_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_groups_cluster(esp_zb_cluster_list, esp_zb_groups_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_scenes_cluster(esp_zb_cluster_list, esp_zb_scenes_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);
    esp_zb_cluster_list_add_on_off_cluster(esp_zb_cluster_list, esp_zb_on_off_cluster, ESP_ZB_ZCL_CLUSTER_SERVER_ROLE);

    esp_zb_ep_list_t *esp_zb_ep_list = esp_zb_ep_list_create();
    /* add created endpoint (cluster_list) to endpoint list */
    esp_zb_ep_list_add_ep(esp_zb_ep_list, esp_zb_cluster_list, HA_ESP_LIGHT_ENDPOINT, ESP_ZB_AF_HA_PROFILE_ID, ESP_ZB_HA_ON_OFF_OUTPUT_DEVICE_ID);
    esp_zb_device_register(esp_zb_ep_list);
    esp_zb_core_action_handler_register(zb_action_handler);
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_main_loop_iteration();
}
```

Tokn59, 2023-12-17:
@swkim01 swkim01

I owe you a BIG thanks !!!!! Got it working after some fiddling around and creating the right converter file for Zigbee2MQTT. The switch works !!!!! Having the possibility now to build custom Zigbee devices is just great after years of Wifi only ! :-)

SCiunczyk, 2024-04-21:
Hi,

I'm trying to connect ESP32-H2/C6 Espressif HA Light on/off to my HA (Conbee III - distance to device 50cm) using @swkim01 example. I got error:
```
error 2024-04-21 11:00:36Interview failed for '0x744dbdfffe600a54 with error 'Error: Interview failed because can not get node descriptor ('0x744dbdfffe600a54')'
error 2024-04-21 11:00:36Failed to interview '0x744dbdfffe600a54', device has not successfully been paired
```
But with ZHA integration the board is connecting without any problems (successful screen below):
[image]

I have H2 and C6 boards - same problem. What steps should I perform to diagnose the bugs/misconfiguration. Any suggestions very appreciated.

chshu, 2024-04-22:
@SCiunczyk Could you please file a new issue here: https://github.com/espressif/esp-zigbee-sdk/issues.

> But with ZHA integration the board is connecting without any problems (successful screen below):

Could you also describe a bit more about this step in the new issue, what's the difference between 'ZHA integration' and `my HA (Conbee III - distance to device 50cm)`?

SCiunczyk, 2024-04-22:
@chshu sure, just give me few minutes.

czagaadam, 2025-02-01:
> [@SCiunczyk](https://github.com/SCiunczyk) Could you please file a new issue here: https://github.com/espressif/esp-zigbee-sdk/issues.
>
>
> Could you also describe a bit more about this step in the new [...]

Can you please support me?
I want to achive the similar with ESP32S3.
I'm using the HA_ON_OFF_LIGHT example.

I tried everything: 
-the original code
-using the pan_id, extended_id and network key
-different channel masks: 0x07FFF800U, 0x00000800U
-extended esp_zb_task from swkim01's comment in this thread [on Dec 17, 2023]

but I'm always getting the Network steering was not successful (status: ESP_FAIL) 

[image]

[image]

[image]

my coordinator is a Easyiot ZB-GW04 Revision v1.2 
Zigbee2MQTT seems working well on RPI
[image]

[image]

czagaadam, 2025-07-13:
Actually I was trying with H2 but maybe the same solution will be helpful for you also. I can send it on tuesday.

martino-giorgi, 2025-07-13:
> Actually I was trying with H2 but maybe the same solution will be helpful for you also. I can send it on tuesday.

Thanks, it would be great.

SCiunczyk, 2025-07-14:
I solved the problem by changing the coordinator from Conbee III to SLZB-MR1. My XIAO C6 was able to to connect without problems. On the other hand I'm playing now with seed studio - they provide framework for zigbee covering many zigbee difficulties with well organized classes.

czagaadam, 2025-07-16:
hi @martino-giorgi 
sorry for late answer, so this is my version:

I recommand firstly leave your code untouched just try these manufacturer and model definitions.
If it won't work place this esp_zb_task function also in your code (the switch parts are commented out, these needs a lot more extra lines...)

#define ESP_MANUFACTURER_NAME "\x09""HLQDQ01LM"
#define ESP_MODEL_IDENTIFIER "\x06""Xiaomi"

static void esp_zb_task(void *pvParameters)
{
    esp_zb_cfg_t zb_nwk_cfg = ESP_ZB_ZED_CONFIG();
    esp_zb_init(&zb_nwk_cfg);
    esp_zb_on_off_light_cfg_t light_cfg = ESP_ZB_DEFAULT_ON_OFF_LIGHT_CONFIG();
    zcl_basic_manufacturer_info_t info = {
        .manufacturer_name = ESP_MANUFACTURER_NAME,
        .model_identifier = ESP_MODEL_IDENTIFIER,
    };
    /*esp_zb_on_off_switch_cfg_t switch_cfg = ESP_ZB_DEFAULT_ON_OFF_SWITCH_CONFIG();
    esp_zb_ep_list_t *ep_list = esp_zb_ep_list_create();
    esp_zb_endpoint_config_t EPLight = {
        .endpoint=HA_ESP_LIGHT_ENDPOINT, 
        .app_profile_id=ESP_ZB_AF_HA_PROFILE_ID, 
        .app_device_id=ESP_ZB_HA_ON_OFF_LIGHT_DEVICE_ID, .app_device_version=1,
    };*/
    /*esp_zb_endpoint_config_t EPSwitch = {
        .endpoint=HA_ONOFF_SWITCH_ENDPOINT, 
        .app_profile_id=ESP_ZB_AF_HA_PROFILE_ID, 
        .app_device_id=ESP_ZB_HA_ON_OFF_SWITCH_DEVICE_ID, .app_device_version=1,
    };*/    
    esp_zb_ep_list_add_ep(ep_list, esp_zb_on_off_light_clusters_create(&light_cfg), EPLight); 
    //esp_zb_ep_list_add_ep(ep_list, esp_zb_on_off_switch_clusters_create(&switch_cfg), EPSwitch);
   
    ESP_ERROR_CHECK(esp_zcl_utility_add_ep_basic_manufacturer_info(ep_list, HA_ESP_LIGHT_ENDPOINT, &info));
    //ESP_ERROR_CHECK(esp_zcl_utility_add_ep_basic_manufacturer_info(ep_list, HA_ONOFF_SWITCH_ENDPOINT, &info));
	
    ESP_ERROR_CHECK(esp_zb_device_register(ep_list));
    esp_zb_core_action_handler_register(zb_action_handler);
    ESP_ERROR_CHECK(esp_zb_set_primary_network_channel_set(DEFAULT_CHANNEL_MASK));
    ESP_ERROR_CHECK(esp_zb_start(false));
    esp_zb_stack_main_loop();
}