espressif/esp-idf #10663: ESP32-S3 ADC digital controller no longer supports filtering

ves011, 2023-01-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

adc_digi_filter_...() functions implementation in ESP-IDF v4.4.3, are guarded by #if CONFIG_IDF_TARGET_ESP32C3, so cannot build for S3
Greping for these function in ESP-IDF 5.0 doesn't find anything

However this page https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32s3/api-reference/peripherals/adc.html has direct reference to these functions.

The question is if it is a miss, or the support for digital filtering is dropped and the page above is not updated?

Icarus113, 2023-01-31:
Hi @ves011 , sorry for the migration guide lacking for these filter related APIs.

The filter related APIs were added in previous IDF versions, but the API behaviours were not guaranteed.
This feature is planned as a IDF v5.1 feature. We will add the lacking migration guide later, and considering to bring it back to v5.0.

ves011, 2023-01-31:
Hi @Icarus113. Thx for your quick response!
Do you have a planned date for V5.1?

MiguelPond, 2023-01-31:
@Icarus113 Regarding the support for the ADC_line_fitting calibration scheme which is locked behind "#if CONFIG TARGET IS ESP32" , is that also scheduled for 5.1?

Icarus113, 2023-02-01:
@ves011  Should be in next few month, but I'm not very sure.

@MiguelPond Would you show me a code link? IIRC the line fitting scheme on v5.0 is for esp32 and s2. 

Besides, maybe you can tell more about what you want to be supported on v5.1. Because line fitting scheme is not planned to be supported on c3 / s3 (curve fitting scheme instead, on these chips).

ves011, 2023-02-20:
Thx @Icarus113 
Will filter by my own till then

closed 2023-02-20 by ves011