espressif/esp-idf #10664: C++ 20's std::jthread destructor causes assertion failure

jdoubleu, 2023-01-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0 

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-WROVER-B

### Power Supply used.

USB

### What is the expected behavior?

C++ 20's [`std::jthread`](https://en.cppreference.com/w/cpp/thread/jthread) should auto-join on destruction (i.e. the destructor calls `.join()`).

### What is the actual behavior?

Instead, it crashes the program, because an assertion inside the `pthread` library failed:
```
assert failed: pthread_self pthread.c:478 (false && "Failed to find current thread ID!")
```

### Steps to reproduce.

Run the following C++ code:
```
#include <thread>
#include <chrono>

#include <esp_log.h>

extern "C"
{

void app_main(void)
{
    ESP_LOGI("main", "Before starting thread.");

    {
        using namespace std::chrono_literals;

        std::jthread{[]{
            ESP_LOGI("main", "Running thread. Wating 3 seconds.");
            std::this_thread::sleep_for(3s);
            ESP_LOGI("main", "Finished waiting inside thread.");
        }};
    } // leaving scope, calls std::jthread destructor

    ESP_LOGI("main", "Thread joined, exiting app_main.");
}

} // extern "C"
```

I've also set `CONFIG_ESP_SYSTEM_PANIC_PRINT_HALT=y` in menuconfig to actually see the stack trace.

### Debug Logs.

```plain
assert failed: pthread_self pthread.c:478 (false && "Failed to find current thread ID!")

Backtrace: 0x40081c12:0x3ffb5e50 0x40085825:0x3ffb5e70 0x4008abc9:0x3ffb5e90 0x400d1369:0x3ffb5fb0 0x400d5fc1:0x3ffb5fd0 0x400d616b:0x3ffb6010 0x400e7a52:0x3ffb6040 0x400885b9:0x3ffb6070
0x40081c12: panic_abort at C:/esp/esp-idf/components/esp_system/panic.c:412

0x40085825: esp_system_abort at C:/esp/esp-idf/components/esp_system/esp_system.c:135

0x4008abc9: __assert_func at C:/esp/esp-idf/components/newlib/assert.c:78

0x400d1369: pthread_self at C:/esp/esp-idf/components/pthread/pthread.c:478

0x400d5fc1: std::stop_token::_Stop_state_t::_M_request_stop() at c:\esp\tools\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\11.2.0\xtensa-esp32-elf\no-rtti\bits/gthr-default.h:687
 (inlined by) std::this_thread::get_id() at c:\esp\tools\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\11.2.0\bits/std_thread.h:314
 (inlined by) std::stop_token::_Stop_state_t::_M_request_stop() at c:\esp\tools\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\11.2.0/stop_token:233

0x400d616b: std::stop_source::request_stop() const at c:\esp\tools\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\11.2.0/stop_token:535
 (inlined by) std::jthread::request_stop() at c:\esp\tools\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\11.2.0/thread:201
 (inlined by) std::jthread::~jthread() at c:\esp\tools\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\11.2.0/thread:129
 (inlined by) app_main at C:/Users/codef/Development/Workspace/espidf_jthread_issue/main/main.cpp:16
0x400e7a52: main_task at C:/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:131 (discriminator 2)

0x400885b9: vPortTaskWrapper at C:/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151
```
```

### More Information.

If I were to manually join the `std::jthread` instead, there's no crash:
```diff
         std::jthread{[]{
             ESP_LOGI("main", "Running thread. Wating 3 seconds.");
             std::this_thread::sleep_for(3s);
             ESP_LOGI("main", "Finished waiting inside thread.");
-        }};
+        }}.join();
     } // leaving scope, calls std::jthread destructor
 ```

I'm not sure where the issue originates, since libstdc++ uses the `pthread` implementation, as mentioned [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/pthread.html)

jdoubleu, 2023-01-31:
This looks more like an issue with the [`std::stop_token`](https://en.cppreference.com/w/cpp/thread/stop_token), which is used by `std::jthread`.

The following code causes a similar stack trace:
```cpp
{
        using namespace std::chrono_literals;

        std::stop_source stop{};

        std::thread t{[](std::stop_token token) {
            ESP_LOGI("main", "Running thread. Wating 3 seconds.");
            std::this_thread::sleep_for(3s);
            ESP_LOGI("main", "Finished waiting inside thread.");
        }, stop.get_token()};

        std::this_thread::sleep_for(1s);
        ESP_LOGI("main", "Requesting stop of thread.");
        stop.request_stop();

        t.join();
    }
```

jdoubleu, 2023-02-01:
As this is more of a limitation of the implementation and not really bug, I'm closing the issue for now. Feel free to re-open it any time, if you fell like this limitation should be overcome.

I only fear, that others might trap into the same problem, because it is not directly clear what the issue is here. Only after looking at the stack trace and knowing the pthread limitations, it becomes clear.

I also found a direct implementation of the `libstdc++` threading library with FreeRTOS, which would be really nice-to-have: https://www.codeproject.com/Articles/1278513/Cplusplus11-FreeRTOS-GCC. It looks like that layer can be implemented from "user" code, as opposed to the library, so maybe this is worth looking into for the [esp-idf-cxx](https://github.com/espressif/esp-idf-cxx) project.

closed 2023-02-01 by jdoubleu

CJCombrink, 2023-04-29:
@jdoubleu I also just ran into this issue with std::jthread.
If this is a known limitation can this please be made explicit in some documentation

0xjakob, 2023-05-04:
@jdoubleu We'll add this to the documentation. Sorry for the long turnaround.

5p4k, 2025-11-19:
I just hit this one as well, that's unfortunate. In fact, it's not just the destructor, it's `request_stop()` that cannot be called at all.

Would it be possible to add a public function (similarly to `pthread_find`) that checks if a thread is managed by pthread or not, so that applications and libraries can assert on that and provide a more meaningful runtime message? E.g. a class using pthreads could assert wiether it's being constructed on a pthread thread, and warn developers early.

I searched the documentation but could not find anything that would provide this.