espressif/esp-idf #10665: HTTP SERVER LATECNY doesn't go down below 30 ms.

Inverter-Jo, 2023-01-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

In tcp server, the average of communication latency is less than 7 ms.

I am using http server.
I call on the local network, but it doesn't go down below 30 ms.
What options do I need to see?

I checked ...
esp_wifi_set_ps(false);
CONFIG_ESP32_WIFI_AMPDU_TX_ENABLED = y -> false;
CONFIG_ESP32_WIFI_AMPDU_RX_ENABLED = y -> false;

Inverter-Jo, 2023-02-02:
checked...
CONFIG_HTTPD_ERR_RESP_NO_DELAY= false

hmalpani, 2023-02-13:
Hello @Inverter-Jo
This might help you improve the performance: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/speed.html#improving-network-speed

Closing this issue. Please feel free to reopen.

closed 2023-02-13 by hmalpani