espressif/esp-idf #10666: GPIO false interrupt when using multiple GPIO interrupt pins

ermacv, 2023-01-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-3101-g1195b6cb2b

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C3-DevKitM-1

### Power Supply used.

USB

### What is the expected behavior?

I expect that when using multiple GPIO input interrupts configured as falling edge interrupt type each interrupt calls its own ISR processing function.

### What is the actual behavior?

When multiple GPIO are configured as input with enabled falling edge interrupt type sometimes incorrect ISR function is called.

### Steps to reproduce.

For example pin 2 and 3 are configured as inputs with interrupts (with internal pull-ups). I send pulse sequence (e.g. 16 pulses) to pin 3 and do nothing with pin 2 (pulled up) from external device. And when ISR is called in RX side I see some of the ISR calls were made from pin 2 ISR function (it is something around 1-4 false ISR calls for 16 pulses on different pin). 

### Debug Logs.

```plain
ESP32 is TX. ESP32C3 is RX. No fixes applied:
TX:
test> tx  -n 25 -o 26 -d 1111111111111111
I (55659) cli_task: Sending 16 bytes
I (55669) cli_task: Data:1111111111111111
RX:
test> rx  -n 2 -o 3
I (22277) gpio: GPIO[2]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (22277) gpio: GPIO[3]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2
I (46787) cli_task: [CB]: Recieved 18 bits for 0 port
I (46787) cli_task: [CB]: Data:111111111101111011
```

### More Information.

I created and application for RX/TX side and I originally used ESP32 as TX and ESP32C3 as RX sides.
The application is CLI based and to configure MCU as  RX you need to call the command:
```
rx  -n 2 -o 3

```
Where 2 is D0 and 3 is D1. After calling this command the pins will be configured to handle interrupts and callback print function is registered. So you need to call this command only once.

For  TX you need to:
```
tx -n 25 -o 26 -d 1111111111111111
```
Where 25 is D0 and 26 is D1 and `1111111111111111` is binary string (there could be 0 and 1 only) that should be send to RX side using provided TX pins. When 1 is set then short pulse generated on D1 pin, and when 0 is set the short pulse generated on D0 pin. The command could be called multiple times.

RX/TX should be different devices because on TX side critical section is used when data are transmitted (so it won't be possible to receive data correctly.)

My colleague used ESP32S3 instead of ESP32C3 as RX part he observes the same issue that I do.

I also found the issue https://github.com/espressif/esp-idf/issues/9746 for ESP32 and I thought that they are similar but after I applied the fix from this issue (that is also mentioned in the ESP32 errata doc) nothing changed. Then I found out that ESP32C3/ESP32S3 don't have any GPIO/Interrupt issues described in the errata docs.

I also checked the `gpio_hal_get_intr_status` function when common GPIO ISR is called and when I send pulses only for one RX pin (D1 e.g.) then I see that the register bit is set for D0 pin but it should not. So it seems that GPIO component behavior is correct and it is related to HW.

The interesting part that I swapped TX and RX sides so ESP32 became RX and ESP32C3 became TX and I applied the fix from the issue I mentioned above for ESP32 and the RX side behavior changed to expected. So it seems for me that GPIO IP blocks are different for ESP32 and ESP32C3/ESP32S3 and something is changed after ESP32 release.

We also found out that we can apply another fix to workaround our problem. When ISR is called we manually check the pin state and if it is 0 then we can process further, otherwise ISR call should be dropped. Using this technique helps us.

To emulate ESP32 interrupt fix please build the code with `kENABLE_ESP32_GPIO_INT_WORKAROUND` defined. If you want to build the application with our fix enabled please use `kENABLE_POST_ESP32_GPIO_INT_WORKAROUND` defined. You can find them in the source file commented at lines 20 and 21.

[main.zip](https://github.com/espressif/esp-idf/files/10545785/main.zip)

ermacv, 2023-02-15:
Hello. Any updates on this issue?

ermacv, 2023-03-03:
Hello @igrr !
Sorry for mentioning you but could you please give any feedback about that issue?

karandwivedi42, 2023-04-20:
Hi @ermacv ! We are also facing the same issue. Did you find a workaround for this?

ermacv, 2023-04-20:
> Hi @ermacv ! We are also facing the same issue. Did you find a workaround for this?

Hello! No, there are no more updates from my side, still waiting for the maintainers response.
For ESP32C3/ESP32S3 I use an extra state check (gpio read) in gpio isr function to verify that isr is really happened. For more information you can check the code I attached to the issue description.

ermacv, 2024-06-03:
Hello!
I checked the issue again and it seems that it was fixed there:
https://github.com/espressif/esp-idf/commit/61282cc5dd14d2f74868384067820a55e1deddd0

The issue could be closed.

closed 2024-06-03 by ermacv

KaeLL, 2024-06-03:
@songruo is this the workaround for the 3.14 issue in the errata?