espressif/esp-idf #10667: Openthread Border Router with ESP32-C6 rcp tx timeout

OriKomemi, 2023-01-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-3101-g1195b6cb2b

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32 && esp32c6

### Power Supply used.

USB

### What is the expected behavior?

Expected the border router to be able to work with a C6 RCP

### What is the actual behavior?

After initializing the wifi connection, the border router is constantly trying to send Thread packets, but fails with following errors:
`W(29231) OPENTHREAD:[W] Platform------: radio tx timeout
W(29231) OPENTHREAD:[W] Platform------: RCP failure detected
W(29241) OPENTHREAD:[W] Platform------: Trying to recover (1/1)
I(29381) OPENTHREAD:[N] Platform------: RCP recovery is done
I(29391) OPENTHREAD:[N] MeshForwarder-: Failed to send IPv6 UDP msg, len:84, chksum:2fe9, ecn:no, to:0xffff, sec:no, error:Abort, prio:net
`

### Steps to reproduce.

1. Flash ot_br example on ESP32
2. Flash ot_rcp example on ESP32-C6
3. Connect the boards via UART
4. Start the border router

### Debug Logs.

```plain
I(8011) OPENTHREAD:[N] Mle-----------: Attach attempt 1, AnyPartition reattaching with Active Dataset
W(11341) OPENTHREAD:[W] Platform------: radio tx timeout
W(11341) OPENTHREAD:[W] Platform------: RCP failure detected
W(11341) OPENTHREAD:[W] Platform------: Trying to recover (1/1)
I(11491) OPENTHREAD:[N] Platform------: RCP recovery is done
I(12081) OPENTHREAD:[N] MeshForwarder-: Failed to send IPv6 UDP msg, len:82, chksum:2c3a, ecn:no, to:0xffff, sec:no, error:Abort, prio:net
I(12081) OPENTHREAD:[N] MeshForwarder-:     src:[fe80:0:0:0:2c6b:1af1:7f25:eb0c]:19788
I(12101) OPENTHREAD:[N] MeshForwarder-:     dst:[ff02:0:0:0:0:0:0:2]:19788
I(14621) OPENTHREAD:[N] RouterTable---: Allocate router id 50
I(14631) OPENTHREAD:[N] Mle-----------: RLOC16 fffe -> c800
I(14641) OPENTHREAD:[N] Mle-----------: Role detached -> leader
I(14641) OPENTHREAD:[N] Mle-----------: Leader partition id 0x33bc27
I (14661) OPENTHREAD: Platform UDP bound to port 49154
I (15661) OPENTHREAD: Received PIO
I (15661) OPENTHREAD: Received PIO
I (15661) OPENTHREAD: Received PIO
[... 124 lines cut ...]
I (18681) OPENTHREAD: Received PIO
I (18681) OPENTHREAD: Received PIO
I (21701) OPENTHREAD: Received PIO
I (21701) OPENTHREAD: Received PIO
I (21701) OPENTHREAD: Received PIO
W(23461) OPENTHREAD:[W] Platform------: radio tx timeout
W(23461) OPENTHREAD:[W] Platform------: RCP failure detected
W(23461) OPENTHREAD:[W] Platform------: Trying to recover (1/1)
I(23611) OPENTHREAD:[N] Platform------: RCP recovery is done
I(23631) OPENTHREAD:[N] MeshForwarder-: Failed to send IPv6 UDP msg, len:84, chksum:17e3, ecn:no, to:0xffff, sec:no, error:Abort, prio:net
I(23631) OPENTHREAD:[N] MeshForwarder-:     src:[fe80:0:0:0:2c6b:1af1:7f25:eb0c]:19788
I(23641) OPENTHREAD:[N] MeshForwarder-:     dst:[ff02:0:0:0:0:0:0:2]:19788
I (24741) OPENTHREAD: Received PIO
I (24741) OPENTHREAD: Received PIO
I (24741) OPENTHREAD: Received PIO
I (29211) OPENTHREAD: Received PIO
I (29221) OPENTHREAD: Received PIO
I (29221) OPENTHREAD: Received PIO
I (29221) OPENTHREAD: Received PIO
W(29231) OPENTHREAD:[W] Platform------: radio tx timeout
W(29231) OPENTHREAD:[W] Platform------: RCP failure detected
W(29241) OPENTHREAD:[W] Platform------: Trying to recover (1/1)
I(29381) OPENTHREAD:[N] Platform------: RCP recovery is done
I(29391) OPENTHREAD:[N] MeshForwarder-: Failed to send IPv6 UDP msg, len:84, chksum:2fe9, ecn:no, to:0xffff, sec:no, error:Abort, prio:net
```

xieqinan, 2023-02-01:
I try to re-produce the BUG by your provided description, but the Border Router runs normally in my environment.
So I want to know the conntection between ESP32 and ESP32C6.
The connecttion type can reference: https://github.com/espressif/esp-idf/blob/master/examples/openthread/ot_br/README.md
My LOG likes below:
```
I (27) boot: ESP-IDF v5.1-dev-3101-g1195b6cb2b 2nd stage bootloader
I (27) boot: compile time Feb  1 2023 10:44:42
I (29) boot: chip revision: v3.0                                                                                                         
I (33) boot.esp32: SPI Speed      : 40MHz                                                                                                
I (38) boot.esp32: SPI Mode       : DIO                                                                                                  
I (42) boot.esp32: SPI Flash Size : 2MB                           
I (47) boot: Enabling RNG early entropy source...                                                                                        
I (52) boot: Partition Table:                                                                                                            
I (56) boot: ## Label            Usage          Type ST Offset   Length        
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00177000
I (85) boot:  3 ot_storage       Unknown data     01 3a 00187000 00002000
I (93) boot: End of partition table                         
I (97) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=3aa14h (240148) map
I (192) esp_image: segment 1: paddr=0004aa3c vaddr=3ffb0000 size=039f8h ( 14840) load
I (198) esp_image: segment 2: paddr=0004e43c vaddr=40080000 size=01bdch (  7132) load
I (202) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=1067c8h (1075144) map
I (595) esp_image: segment 4: paddr=001567f0 vaddr=40081bdc size=14be0h ( 84960) load
I (641) boot: Loaded app from partition at offset 0x10000
I (642) boot: Disabling RNG early entropy source...
I (653) cpu_start: Pro cpu up.           
I (653) cpu_start: Starting app cpu, entry point is 0x40081448
0x40081448: call_start_cpu1 at /home/xieqinan/ESP/esp-idf/components/esp_system/port/cpu_start.c:150
                                                                                                                                         
I (0) cpu_start: App cpu up.                                                                                                             
I (670) cpu_start: Pro cpu start user code                                                                                               
I (670) cpu_start: cpu freq: 160000000 Hz                                                                                                                                                                                                                                          
I (670) cpu_start: Application information:  
I (674) cpu_start: Project name:     esp_ot_br
I (680) cpu_start: App version:      v5.1-dev-3101-g1195b6cb2b
I (686) cpu_start: Compile time:     Feb  1 2023 10:44:36       
I (692) cpu_start: ELF file SHA256:  34fc1b2b39b695e8...
I (698) cpu_start: ESP-IDF:          v5.1-dev-3101-g1195b6cb2b                                                                           
I (705) cpu_start: Min chip rev:     v0.0                     
I (709) cpu_start: Max chip rev:     v3.99                    
I (714) cpu_start: Chip rev:         v3.0                     
I (719) heap_init: Initializing. RAM available for dynamic allocation:                                             
I (726) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM                                                                                
I (732) heap_init: At 3FFC2B70 len 0001D490 (117 KiB): DRAM
I (738) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (745) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM                                                                            
I (751) heap_init: At 400967BC len 00009844 (38 KiB): IRAM                                                                               
I (759) spi_flash: detected chip: generic                                                                                                
I (762) spi_flash: flash io: dio                                                                                                         
W (766) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (780) app_start: Starting scheduler on CPU0
I (784) app_start: Starting scheduler on CPU1
I (784) main_task: Started on CPU0
I (794) main_task: Calling app_main()
I (814) main_task: Returned from app_main() 
I(814) OPENTHREAD:[I] Platform------: RCP reset: RESET_POWER_ON
I(1044) OPENTHREAD:[I] Settings------: Read NetworkInfo {rloc:0xe400, extaddr:527e323ca6c73e5a, role:leader, mode:0x0f, version:4, keyseq:0x0, ...
I(1054) OPENTHREAD:[I] Settings------: ... pid:0x517f9f7a, mlecntr:0x139e, maccntr:0x138e, mliid:8de0b6d9ab8501f9}
I (1074) OPENTHREAD: OpenThread attached to netif
> ifconfigup                                
Error 35: InvalidCommand                    
> ifconfig up                               
[... cut ...]
I (149324) OPENTHREAD: Received PIO                               
I (155514) OPENTHREAD: Multicast listener add: FF05::2            
I (156134) OPENTHREAD: Multicast listener add: FF05::2            
I (156734) OPENTHREAD: Received PIO                               
I (164094) OPENTHREAD: Received PIO                               
I (168524) OPENTHREAD: Received PIO
I (172364) OPENTHREAD: Received PIO                               
I (175384) OPENTHREAD: Received PIO
I (191404) OPENTHREAD: Received PIO
```

OriKomemi, 2023-02-01:
Problem solved after replacing to another esp32c6.

closed 2023-02-01 by OriKomemi