espressif/esp-idf #10668: esp_wifi_80211_tx sends RTS, not arbitrary frames

faragher, 2023-01-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-CAM

### Power Supply used.

USB

### What is the expected behavior?

Broadcast message 08 02 00 00 DE AD BE EF 42 72 DE AD BE EF 42 72 DE AD BE EF 42 72 00 00 08 06 07 05 03 00 09  + CRC32

(Data packet from AP [From AP mode tests] with sequence 0x00 0x00 [Appears to be overwritten; acceptable] and three DE:AD:BE:EF:42:72 dummy MACs, followed by test payload [8675309] - CRC32 can be created in program or hardware)

### What is the actual behavior?

Broadcasts a burst (~24) of frames with payload:  B4 00 56 04 (3x dummy MAC) + CRC32

(Request to send)

### Steps to reproduce.

1. Set up WiFi
2. Send packet with esp_wifi_80211_tx
3. Packet sniffed with promiscuous mode adapter using Wireshark.

### Debug Logs.

```plain
(Typical)
Building Frame
8 2 0 0 DE AD BE EF 42 72 DE AD BE EF 42 72 DE AD BE EF 42 72 0 0 8 6 7 5 3 0 9 
63 44 E0 7E 
8 2 0 0 DE AD BE EF 42 72 DE AD BE EF 42 72 DE AD BE EF 42 72 0 0 8 6 7 5 3 0 9 63 44 E0 7E 

(Lowest two lines confirm arrays contain information as desired)
```

### More Information.

Day 4 of attempting to broadcast packets for unidirectional data flow. Issues persisted in previous versions, updated to v4.4.2 via Arduino board manager (via 2.0.5, as 2.0.6 has a 404 and cannot install through the board manager). System functions as intended except esp_wifi_80211_tx does not transmit packet. Function broadcasts multiple RTS frames instead of the frame desired. As it is not connected to an access point (or is an access point) these fail and the desired frame is never sent.

Please advise if backtrace or linked debugger is required, as well as how to perform these actions with the Arduino IDE.

Code follows: (note: may contain remnants of camera control code)
```
#include "esp_camera.h"

#include "freertos/FreeRTOS.h"

#include "esp_event.h"
#include "esp_system.h"
#include "esp_event.h"
#include "esp_wifi.h"

#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_log.h"
#include "nvs_flash.h"

#define CAMERA_MODEL_AI_THINKER

#include "camera_pins.h"
#include "esp_system.h"

esp_err_t esp_wifi_80211_tx(wifi_interface_t ifx, const void *buffer, int len, bool en_sys_seq);

uint8_t BadMAC[] = {
  0xde, 0xad, 0xbe, 0xef, 0x42
};

uint8_t DataFrameHeader[] ={
  0x08, 0x02, 0x00, 0x00,
  0xde, 0xad, 0xbe, 0xef, 0x42, 0x72,
  0xde, 0xad, 0xbe, 0xef, 0x42, 0x72,
  0xde, 0xad, 0xbe, 0xef, 0x42, 0x72
};

uint8_t Sequence[] = {
  0x00, 0x00
};

uint8_t Payload[1024];

uint8_t FCS[4];

uint8_t TestInfo[] = {
  0xc4, 0x00, 0xe8, 0x07, 0x9c,
  0x8e, 0xcd, 0x25, 0xa3, 0x28
};

size_t TestLength = sizeof(TestInfo);

uint32_t crc32_formula_reflect( size_t len, const void *data, const uint32_t POLY = 0xEDB88320 )
    {
        const unsigned char *buffer = (const unsigned char*) data;
        uint32_t crc = -1;

        while( len-- )
        {
            crc = crc ^ *buffer++;
            for( int bit = 0; bit < 8; bit++ )
            {
                if( crc & 1 ) crc = (crc >> 1) ^ POLY;
                else          crc = (crc >> 1);
            }
        }
        return ~crc;
    }

void BuildFrame(){
  Serial.println("Building Frame");
  uint8_t LoadPay[] = {0x08, 0x06, 0x07, 0x05, 0x03, 0x00, 0x09};
  int TempLen = sizeof(DataFrameHeader) + sizeof(Sequence) + sizeof(LoadPay);
  uint8_t Temp[TempLen];
  memcpy( &Temp[0], &DataFrameHeader[0], 22 * sizeof( uint8_t ) );
  memcpy( &Temp[22], &Sequence[0], 2 * sizeof( uint8_t ) );
  memcpy( &Temp[24], &LoadPay[0], sizeof(LoadPay) );
  for(int i = 0; i<(sizeof(Temp)/sizeof(Temp[0]));i++){
    Serial.print(Temp[i],HEX);
    Serial.print(" ");
  }
  Serial.println("");
  uint32_t crcW = crc32_formula_reflect(TempLen,Temp);
  byte* crcB = (byte*)&crcW;
    for(int i = 0; i<4;i++){
    Serial.print(crcB[i],HEX);
    Serial.print(" ");
  }
    Serial.println("");

  uint8_t Temp2[TempLen+4];
  memcpy( &Temp2[0], &Temp[0], TempLen);
  memcpy( &Temp2[TempLen],&crcB[0],4);

  for(int i = 0; i<TempLen+4;i++){
    Serial.print(Temp2[i],HEX);
    Serial.print(" ");
  }
    Serial.println("");
    if (ESP_OK != esp_wifi_80211_tx(WIFI_IF_STA, Temp, TempLen, false)){
    ESP_LOGE("raw tx", "tx failed");;
  }
  
}

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println();

  wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
  ESP_ERROR_CHECK(esp_wifi_init(&cfg));
  ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));
  ESP_ERROR_CHECK(esp_netif_init());
  ESP_ERROR_CHECK(esp_event_loop_create_default());
  esp_netif_create_default_wifi_sta();
  ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));

  wifi_config_t wifi_config = { };
  wifi_config.sta.channel = 1;
 
  ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &wifi_config));
  ESP_ERROR_CHECK(esp_wifi_start());
  ESP_ERROR_CHECK(esp_wifi_set_ps(WIFI_PS_NONE));

  Serial.print("IDF Version: ");
  Serial.println(esp_get_idf_version());
  BuildFrame();
  

}

void loop() {
  delay(10000);
  BuildFrame();
}
```

Syphontwo, 2024-03-05:
I would like to add that I have observed this issue, with a few extra points.
- I am sending a null data frame (0x48) in order to keep the message as short as possible.
- Sometimes the arbitrary frame goes through exactly as desired, other times, it tries to use RTS/CTS first.
- My device never receives CTS, even from devices which are correctly sending CTS to others (such as my AP)
- The RTS seems to spam out pretty extreme.  A single call to esp_wifi_80211_tx() can result in a dozen RTS calls.

As a user, this is the behavior I would want-
- if esp_wifi_80211_tx() fails to actually send due to no CTS, I would like a callback function to inform me of this failure once the device gives up.
- I would like to be able to tell it not to use RTS/CTS, with a specified window of time to look for an opening to send without it.  Something like "listen for 50 milliseconds and attempt to send", again, I would like a callback or even potentially a blocking function.
- I would expect a single call to esp_wifi_80211_tx() to result in a configurable number of RTS retries (down to 1).

In general, I would like a little more control over either what the esp_wifi_80211_tx() does regarding RTS/CTS, or callback functions that can deliver the results of the attempted transmission.  I will submit the callback as a feature request.

faragher, 2024-03-05:
For context: I did some research: this is a deliberate. Espressif removed this capability with a version change, I forget which one. You can no longer send arbitrary frames without rolling back to an older version of IDF.

Syphontwo, 2024-03-05:
> For context: I did some research: this is a deliberate. Espressif removed this capability with a version change, I forget which one. You can no longer send arbitrary frames without rolling back to an older version of IDF.

What version was this changed in?  I'll downgrade no problem.

zhangyanjiaoesp, 2024-03-26:
@Syphontwo @faragher we will consider the  feature request.

Sherry616, 2024-03-26:
This feature request and [https://github.com/espressif/esp-idf/issues/13319](url) are repeated. We will close this ticket first and please move to [https://github.com/espressif/esp-idf/issues/13319](url) for more updates, thanks.

closed 2024-03-26 by Sherry616

Rokachy, 2025-01-07:
Please look at below coding instruction to disable RTS
------------------------------------------------------------

//define
typedef struct 
{ 
 uint16_t threshold;                 // RTS packet length threshold 
 uint8_t  mpdu_retry_before_rts;     // Number of retries before sending RTS 
 uint8_t  long_ampdu_retry_before_rts; 
 uint8_t  short_ampdu_retry_before_rts; 
} wifi_internal_rts_cfg_t;
static wifi_internal_rts_cfg_t rts_cfg={0};

//Apis prototypes
extern int esp_wifi_internal_set_rts(wifi_internal_rts_cfg_t* rts);
extern int esp_wifi_internal_set_retry_counter(int src, int lrc);

//at wifi init config for NO RTS 
  wifi_internal_rts_cfg_t rts_cfg={0};
  rts_cfg.threshold =1600;
  rts_cfg.mpdu_retry_before_rts=32;

//after wifi start call this api to avoid RTS 
//         ---------
          esp_wifi_internal_set_rts(&rts_cfg);

//just before sending the message call this api to avoid retry 
        esp_wifi_internal_set_retry_counter(0, 0);
        esp_wifi_80211_tx(WIFI_IF_STA, buf, len, false);

enjoy :)

Syphontwo, 2025-01-07:
I definitely do enjoy this, but is this method of turning off RTS something that may be supported in the future? Don't get me wrong, I love a good hack, just want to know how well I can rely on it.

zhangyanjiaoesp, 2025-01-08:
@Syphontwo 
You can call these two functions using the method described above. We will consider officially making them available for customer use in the future.

roger-, 2025-01-17:
@Rokachy thanks for that!

@zhangyanjiaoesp please do consider making those functions official. I’m also worried about them going away. Can you provide some more documentation on them in the mean time?