espressif/esp-idf #10670: Unable to execute parttool.py

william-ferguson-au, 2023-02-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

Show the various usage options for parttool (first step before using it to retrieve the partitions info from a device).

### What is the actual behavior?

```
C:\Dev\espressif\esp-idf\components\partition_table>parttool.py --help
Traceback (most recent call last):
  File "C:\Dev\espressif\esp-idf\components\partition_table\parttool.py", line 28, in <module>
    import gen_esp32part as gen
  File "C:\Dev\espressif\esp-idf\components\partition_table\gen_esp32part.py", line 254
    def verify_size_fits(self, flash_size_bytes: int) -> None:
                                               ^
SyntaxError: invalid syntax
```

### Steps to reproduce.

1. Opened command window
2. cd $IDF_PATH
3. parttoo.l.py --help

### More Information.

C:\Dev\espressif\esp-idf\components\partition_table>python --version

Python 3.8.7

dobairoland, 2023-02-01:
Hi @william-ferguson-au. Are you in an ESP-IDF CMD or Power shell environment (shortcuts are installed by the Windows tool installer)? `parttool.py` should properly run there (mainly because of https://github.com/espressif/esp-idf/blob/release/v4.4/export.bat#L51).

The issue probably here is that `*.py` files could be associated on Windows with any arbitrary Python. I'm guessing from the error that it is Python 2.7. It is not guaranteed that it is the same `python` you've invoked with `python --version`. You could try `python parttool.py --help` instead. But the best option is my first suggestion: to run the tool in an ESP-IDF console environment.

william-ferguson-au, 2023-02-01:
Ah thanks @dobairoland. That's that I was missing. I needed to be in the ESP-IDF CMD (I was but must have jumped out).

closed 2023-02-01 by william-ferguson-au