espressif/esp-idf #10674: assert failed: spi_bus_deinit_lock

Massimiliano-solutiontech, 2023-02-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Operating System used.

macOS

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-s3-devkitc-1

### Power Supply used.

USB

### What is the expected behavior?

I expect to deinit the SPI correctly

### What is the actual behavior?

The firmware crashes

### Steps to reproduce.

When I call `spi_bus_free((spi_host_device_t)_host.slot);` the firmware crashes, that always occurs when in the log I also see `spi_master: spi_master_deinit_driver(272): not all CSses freed` (I don't know if the two errors are correlated.

In the `spi_common_internal.h` there are some instruction on how to deinit the spi but those are not public api that I can call
```
* * Deinitialization (optional):
* 1. Call `spi_bus_lock_unregister_dev` for each device when they are no longer needed.
* 2. Call `spi_bus_deinit_lock` to spirelease the resources occupied by the lock.
 ```

I also tried to call the `spi_bus_free` only when the SPI is initialized using this function `bool SDManager::isSpiInit() {  return spi_bus_get_attr((spi_host_device_t)_host.slot) != NULL; }`

The function `SDManager::_reattachSD(void*)` is a ISR that is triggered when ever the SD insert pin changes and creates a task `xTaskCreatePinnedToCore(_reattachSD, "_reattachSD", 3 * 1024, NULL, 5, NULL, 0);` that checks if the SD is inserted or not if is inserted is called `_initSD()`  otherwise `deinitSpi()`

```
void SDManager::_initSD()
{
    isSDInit = false;

    spi_bus_config_t bus_cfg = {
        .mosi_io_num = SD_MOSI,
        .miso_io_num = SD_MISO,
        .sclk_io_num = SD_CLK,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
        .data4_io_num = -1,
        .data5_io_num = -1,
        .data6_io_num = -1,
        .data7_io_num = -1,
        .max_transfer_sz = 4092,
        .flags = SPICOMMON_BUSFLAG_MASTER | SPICOMMON_BUSFLAG_SCLK | SPICOMMON_BUSFLAG_MOSI | SPICOMMON_BUSFLAG_MISO,
        .intr_flags = ESP_INTR_FLAG_IRAM};

    deinitSpi(); ** --> crash here (line 88)<-- **
    auto spiInitErr = spi_bus_initialize((spi_host_device_t)_host.slot, &bus_cfg, SPI_DMA_CH_AUTO);
    if (spiInitErr != ESP_OK)
    {
        ESP_LOGE(__func__, "spi_bus_initialize failed with error 0x%x", spiInitErr);
    }

    esp_vfs_fat_sdmmc_mount_config_t mount_config = {
        .format_if_mount_failed = true,
        .max_files = 5,
        .allocation_unit_size = 16 * 1024};

    sdspi_device_config_t slot_config = {
        .host_id = (spi_host_device_t)_host.slot,
        .gpio_cs = (gpio_num_t)SD_CS,
        .gpio_cd = (gpio_num_t)SD_INSERT,
        .gpio_wp = SDSPI_SLOT_NO_WP,
        .gpio_int = SDSPI_SLOT_NO_INT};
}
```

### Debug Logs.

```plain
E (22090) spi_master: spi_master_deinit_driver(272): not all CSses freed
I (22090) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (22096) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (22105) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 

assert failed: spi_bus_deinit_lock spi_bus_lock.c:599 (atomic_load(&lock->dev[i]) == (intptr_t)NULL)

Backtrace: 0x403784d2:0x3fcf4620 0x40380ca9:0x3fcf4640 0x40387ad9:0x3fcf4660 0x420374dd:0x3fcf4780 0x42036ded:0x3fcf47a0 0x420144e3:0x3fcf47c0 0x42014aa7:0x3fcf47e0 0x42014d56:0x3fcf48e0 0x42014ebf:0x3fcf4950 0x40383c7d:0x3fcf4970

  #0  0x403784d2:0x3fcf4620 in panic_abort at /Users/massimilianomontagni/.platformio/packages/framework-espidf/components/esp_system/panic.c:402
  #1  0x40380ca9:0x3fcf4640 in esp_system_abort at /Users/massimilianomontagni/.platformio/packages/framework-espidf/components/esp_system/esp_system.c:128
  #2  0x40387ad9:0x3fcf4660 in __assert_func at /Users/massimilianomontagni/.platformio/packages/framework-espidf/components/newlib/assert.c:85
  #3  0x420374dd:0x3fcf4780 in spi_bus_deinit_lock at /Users/massimilianomontagni/.platformio/packages/framework-espidf/components/driver/spi_bus_lock.c:599 (discriminator 1)
  #4  0x42036ded:0x3fcf47a0 in spi_bus_free at /Users/massimilianomontagni/.platformio/packages/framework-espidf/components/driver/spi_common.c:882
  #5  0x420144e3:0x3fcf47c0 in SDManager::deinitSpi() at src/SD/SDManager.cpp:213
  #6  0x42014aa7:0x3fcf47e0 in SDManager::_initSD() at src/SD/SDManager.cpp:88
  #7  0x42014d56:0x3fcf48e0 in SDManager::tmp() at src/SD/SDManager.cpp:168
  #8  0x42014ebf:0x3fcf4950 in SDManager::_reattachSD(void*) at src/SD/SDManager.cpp:143
  #9  0x40383c7d:0x3fcf4970 in vPortTaskWrapper at /Users/massimilianomontagni/.platformio/packages/framework-espidf/components/freertos/port/xtensa/port.c:131

ELF file SHA256: 4bef0ad3a541e461

Rebooting...
```

### More Information.

I use PlatformIO to build the project

wanckl, 2023-02-02:
Hello, seems it not a bug, the log `not all CSses freed` means you still have device not removed on spi_host, then driver invoke assert failure. spi_host can mount max 6 devices, so you need remove those devices manually before free bus, a normal struct may like:
```c
spi_bus_initialize(spi2);
spi_bus_add_device(dev1);
spi_bus_add_device(dev2);
spi_device_transmit(dev1);
spi_bus_remove_device(dev1);
spi_device_transmit(dev2);
spi_bus_remove_device(dev2);
spi_bus_free(spi2);
```
Please check your logic is somewhere forget to remove your devices, any device fail to remove then try to free bus will cause the crash like this.   

Best wishes!

Massimiliano-solutiontech, 2023-02-09:
Hi, I don't do any `spi_bus_add_device` so how can I call `spi_bus_remove_device(spi_device_handle_t handle)`?

```C++
sdmmc_host_t SDManager::_host = SDSPI_HOST_DEFAULT();
_host.flags |= SDMMC_HOST_FLAG_4BIT;
```

```C++
spi_bus_config_t bus_cfg = {
    .mosi_io_num = SD_MOSI,
    .miso_io_num = SD_MISO,
    .sclk_io_num = SD_CLK,
    .quadwp_io_num = -1,
    .quadhd_io_num = -1,
    .data4_io_num = -1,
    .data5_io_num = -1,
    .data6_io_num = -1,
    .data7_io_num = -1,
    .max_transfer_sz = 4092,
    .flags = SPICOMMON_BUSFLAG_MASTER | SPICOMMON_BUSFLAG_SCLK | SPICOMMON_BUSFLAG_MOSI | SPICOMMON_BUSFLAG_MISO,
    .intr_flags = ESP_INTR_FLAG_IRAM};

deinitSpi();
auto spiInitErr = spi_bus_initialize((spi_host_device_t)_host.slot, &bus_cfg, SPI_DMA_CH_AUTO);
if (spiInitErr != ESP_OK)
{
    ESP_LOGE(__func__, "spi_bus_initialize failed with error 0x%x", spiInitErr);
}

esp_vfs_fat_sdmmc_mount_config_t mount_config = {
    .format_if_mount_failed = true,
    .max_files = 5,
    .allocation_unit_size = 16 * 1024};

sdspi_device_config_t slot_config = {
    .host_id = (spi_host_device_t)_host.slot,
    .gpio_cs = (gpio_num_t)SD_CS,
    .gpio_cd = (gpio_num_t)SD_INSERT,
    .gpio_wp = SDSPI_SLOT_NO_WP,
    .gpio_int = SDSPI_SLOT_NO_INT};

auto sdMountErr = esp_vfs_fat_sdspi_mount(MOUNT_POINT, &_host, &slot_config, &mount_config, &_card);
if (sdMountErr != ESP_OK)
{
    return;
}

_host.set_card_clk((spi_host_device_t)_host.slot, SDMMC_FREQ_52M);
```

wanckl, 2023-02-10:
Hi @Massimiliano-solutiontech :
Both the error log of `not all CSses freed` and `(atomic_load(&lock->dev[i]) == (intptr_t)NULL)` indicate do is that you have spi device not freed.
So:
1. Is the first or second time you unpin the sd card then it crashed ?
2. You use a sd card with spi bus, in that case, the sdcard driver will call into spi driver to create the sdcard handle, please check if any errors or warnings report from sdcard driver then cause something like `sdcard_deinit` failed, or you forgot to free sdcard handle clearly.

Best wishes!

Massimiliano-solutiontech, 2023-02-10:
Hi, to be honest it’s quite random sometimes it happens at the 10th sd removal. If you mean compiling warnings I have nothing and at runtime I only have the `not all CSses freed` error and then the crash on the next sd insert

wanckl, 2023-02-10:
@Massimiliano-solutiontech 
If you use a SD insert pin interrupt, in some cases, you need to deal with glitch, which may lead to shoot your ISR repeatedly,  then it have chance to create task repeatedly. Or one `_initSD` interrupted by interrupt again. all those can cause `spi_bus_add_device` more than once, It is the most possible reason. 

another reason may be: how are your code to `deinitSpi`, is here some logic issue cause spi deinit failed. 

wish to have something helpful, please tell me if you have any update ~

Massimiliano-solutiontech, 2023-02-10:
At the beginning of the isr I detach the interrupt and then I create the task, maybe I could try to put the `xTaskCreatePinnedToCore` inside a `portENTER_CRITICAL` block?
Here the `deinitSpi`

```C++
bool SDManager::isSpiInit()
{
    return spi_bus_get_attr((spi_host_device_t)_host.slot) != NULL;
}

void SDManager::deinitSpi()
{
    if (!isSpiInit())
    {
        return;
    }

    spi_bus_free((spi_host_device_t)_host.slot);
}
```

wanckl, 2023-02-13:
No, I means if there is glitch noise, you may shoot one ISR several time during one insert or unpin action, I think you can,,,for example:
- Only notify another task in your ISR, than delay some time in that task to ensure sdcard had action done (e.g. 500ms), this can skip the noise, then read the level of your `SD_CS` pin to decide what to do.   
- Use flags to ensure `initSD` and `deinitSD` called only once within a time segment, e.g. 500ms.

Besides,,you use `spi_bus_initialize` and `esp_vfs_fat_sdspi_mount` in your init func, where is the `esp_vfs_fat_sdcard_unmount` in your deinit func??

Waiting for your reply!

MassimilianoMontagni, 2023-02-16:
I tried by checking if the task is already there but is still crashing

```C++
void IRAM_ATTR SDManager::_sdInsertChanged()
{
    if (_reattachSdTaskHandle != NULL)
    {
        auto taskState = eTaskGetState(_reattachSdTaskHandle);
        if (taskState != eTaskState::eDeleted)
        {
            return;
        }
    }

    xTaskCreatePinnedToCore(_reattachSD, "_reattachSD", 3.0 * 1024, NULL, 5, _reattachSdTaskHandle, 0);
}
```

I do the `esp_vfs_fat_sdcard_unmount` before the `deinitSpi()`

```C++
if (isSDInit)
{
    esp_vfs_fat_sdcard_unmount(MOUNT_POINT, _card);
}

deinitSpi();
```

wanckl, 2023-02-16:
Something strange, could you check return value of `esp_vfs_fat_sdcard_unmount` to see is any reason it unmount fialed, and some print to see is it unmount successfully each time. thank you!

Massimiliano-solutiontech, 2023-02-23:
Sorry for the late response. Below you can find the log of a remove and insert of the sd, as you can see unmount returns `ESP_OK` but the crash still happen

```C++
auto res = esp_vfs_fat_sdcard_unmount(MOUNT_POINT, _card);
if (res != ESP_OK)
{
    ESP_LOGE(__func__, "Failed to unmount SD card (%s)", esp_err_to_name(res));
}
else
{
    ESP_LOGI(__func__, "Unmounted SD card");
}
```

```
I (7852) tmp: Unmounted SD card
I (7855) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (7864) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (7874) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (9586) gpio: GPIO[10]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (9587) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (9635) sdspi_transaction: cmd=5, R1 response: command not supported
E (11896) spi_master: spi_master_deinit_driver(272): not all CSses freed
I (11896) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (11902) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (11911) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 

assert failed: spi_bus_deinit_lock spi_bus_lock.c:599 (atomic_load(&lock->dev[i]) == (intptr_t)NULL)
```

wanckl, 2023-03-23:
@MassimilianoMontagni 
,,,From your log, I don't know which time you insert the sd card again, there are some conditions I can think about,,,:
```
I (7852) tmp: Unmounted SD card                                 <-----Here seems the sd unmount success
I (7855) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0   <
I (7864) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0   <
I (7874) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0   <--  Here free the spi bus
I (9586) gpio: GPIO[10]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0   <
I (9587) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0   <--  Then what happen here, add or rm card ?
I (9635) sdspi_transaction: cmd=5, R1 response: command not supported            <---- if above is rm card, why here have a transaction failed; while if is add cared, why have a `deinit driver bellow?`
E (11896) spi_master: spi_master_deinit_driver(272): not all CSses freed
I (11896) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (11902) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (11911) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
assert failed: spi_bus_deinit_lock spi_bus_lock.c:599 (atomic_load(&lock->dev[i]) == (intptr_t)NULL)
```
---
In the second attempt, it is a insert behaviour without remove, but it mount failed right?
```
I (228860) gpio: GPIO[10]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (228860) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (228909) sdmmc_sd: sdmmc_init_spi_crc: sdmmc_send_cmd_crc_on_off returned 0x105
E (228909) vfs_fat_sdmmc: sdmmc_card_init failed (0x105).
I (228912) gpio: GPIO[10]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (228922) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (228931) _initSD: Failed to mount SD card (ESP_ERR_NOT_FOUND)              <---- This is your project log somewhere right?
I (229288) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0   <
I (229288) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0   <
I (229296) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0   <-- Then here seems free spi bus again!, is this happen automatically or by your code, because here have no `tmp: Unmounted SD card` logged befor.
I (229306) gpio: GPIO[10]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (229315) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (229325) sdspi_transaction: cmd=52, R1 response: command not supported
I (229372) sdspi_transaction: cmd=5, R1 response: command not supported         <----- Then why here transaction failed again
```

I think this maybe logic issue somewhere in your proj, or in `vfs` component of idf, and ESP32S3 has 2 cores, you need pay some attention to avoid possible competition or intercurrent problem, and you can also find who familiar with `vfs` for help.

isfandyarIOTMAV, 2024-02-27:
have you got any solution i have same issue

italocjs, 2024-07-14:
any fix on this?