espressif/esp-idf #10675: Allow obtaining preamble duration information

doragasu, 2023-02-01:
### Is your feature request related to a problem?

**TL;DR**: It would be nice to include a `preamble_length` field in `wifi_pkt_rx_ctrl_t` structure, with the preamble length in microseconds.

**Long story**: I'm trying to implement a WiFi channel load monitor, to compute the percentage of a channel that is used in real time, and be able to log that information (so I can know how the channel load evolves throughout the day). My approach uses promiscuous mode to receive all frames transmitted on a channel, and tries computing and aggregating the air time for each frame.

To compute the frame air time, I use the expression (maybe there's a better way, suggestions are welcome):

```
air_time = preamble + sig_len * 8 / bit_rate
```

I obtain the bit rate using the `rate` field in `wifi_pkt_rx_ctrl_t` structure for 802.11bg packets, and the `mcs`, `cwb` and `sgi` fields for 802.11n packets (with the help of [this table](https://docs.google.com/spreadsheets/d/e/2PACX-1vQXoEYLGWrR1aGyGaTXOOaDQSPLfeC4rv70KRFuRP6eZ5fL-Ku_YI6DgS6zZMNyIhQpQmnKQ1O7abij/pubhtml?gid=1367372895&single=true)).

The problem I have is I'm missing data to obtain the `preamble` time. For 802.11bg, `rate` field is decoded using `wifi_phy_rate_t` enum. When `rate` is between 0 and 7, this field shows if the preamble is long (192 us) or short (96 us). But for values greater than 7, how can I know which preamble length is used? The same happens for 802.11n, how can I know the preamble length?

### Describe the solution you'd like.

When receiving frames captured in promiscuous mode, the `wifi_pkt_rx_ctrl_t` structure should have a `preamble_length` with the duration of the preamble in microseconds, corresponding to the received frame.

### Describe alternatives you've considered.

As far as I know, other approaches are currently not possible because the required information is in the PHY layer, and we do not have access unless the WiFi driver exposes the information.

### Additional context.

Being able to obtain accurate channel load information is very useful to diagnose congested networks. Thus this feature can be very interesting.

zhangyanjiaoesp, 2024-04-01:
@doragasu 

> The problem I have is I'm missing data to obtain the preamble time. For 802.11bg, rate field is decoded using wifi_phy_rate_t enum. When rate is between 0 and 7, this field shows if the preamble is [...]

For values greater than 7, the rate is 802.11g/n/ac, the preamble can refer to the PPDU header structure. For more information, you can refer to chapter 19.3, 21.3 of the 802.11-2020 protocol.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

doragasu, 2024-04-02:
Thanks for feedback. As far as I know, the PPDU structure is not provided to user application, so how can I get this information?

zhangyanjiaoesp, 2024-04-03:
@doragasu As the images from the previous comment show, the 11g preamble is fixed at 20us, the 11n preamble is 8us+8us+4us+8us+4us+4*Nus, at least 36us, the N is related to MCS, you can take a look at the 802.11 protocol. The 11ac preamble is 8us+8us+4us+8us+4us+4*Nus+4us, the N is related to the mcs, cwb, stbc.

doragasu, 2024-04-05:
Oh I see, thanks!

closed 2024-06-07 by Sherry616