espressif/esp-idf #10676: Wrong library include

AndreaPici, 2023-02-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

Release/v4.4.4

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I'm trying to compile my code and I've got the missing reference to `prvTaskPriorityRaise` into `components/spi_flash/cache_utils.c` file.

I've opened this file and this is the include list:

```
#include <stdlib.h>
#include <assert.h>
#include <string.h>
#include <stdio.h>

#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <freertos/idf_additions.h>
#include <freertos/semphr.h>
#if CONFIG_IDF_TARGET_ESP32
...
```

If you go into [freertos](https://github.com/espressif/esp-idf/tree/v4.4.4/components/freertos/include/freertos) you can see that idf_additions.h is not present

### What is the actual behavior?

Cannot compile micropython project

### Steps to reproduce.

Just compile any project

igrr, 2023-02-01:
Could you please specify the commit ID of Micropython project which you are compiling, and the steps you are using to compile it? (I.e. the command you are running)

igrr, 2023-02-02:
Also, are you using https://github.com/micropython/micropython or https://github.com/pycom/pycom-micropython-sigfox?

AndreaPici, 2023-02-02:
Hi there. Today I've started from a new clean session and everything was ok. Maybe I had some problem with Git alignment.

I'm sorry

closed 2023-02-02 by AndreaPici